mu-editor/mu #351: Raspberry Pi - Debugger fails to start "Unable to connect"

martinohanlon, 2018-02-27:
Using mu 1.0.0.beta15 on a Raspberry Pi, when clicking debug, the debugger fails to start.

You are presented with:

```
Unable to connect to the Python debugger.

Connection timed out. Is your machine slow or busy? Free up some of the machine's resources and try again.

---------- FINISHED ----------
exit code: 1 status: -1
```

Details from the log file.

Starting Mu 1.0.0.beta.15
2018-02-27 22:11:54,945 - root:113(run) INFO: uname_result(system='Linux', node='raspberrypi', release='4.9.59-v7+', version='#1047 SMP Sun Oct 29 12:19:23 GMT 2017', machine='armv7l', processor='')
2018-02-27 22:11:54,945 - root:114(run) INFO: Python path: ['/home/pi/mu', '/usr/lib/python35.zip', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-arm-linux-gnueabihf', '/usr/lib/python3.5/lib-dynload', '/usr/local/lib/python3.5/dist-packages', '/usr/lib/python3/dist-packages', '/usr/local/lib/python3.5/dist-packages/IPython/extensions']
2018-02-27 22:11:55,490 - mu.logic:355(__init__) INFO: Setting up editor.
2018-02-27 22:11:55,491 - mu.logic:366(__init__) INFO: Settings path: /home/pi/.local/share/mu/settings.json
2018-02-27 22:11:55,491 - mu.logic:367(__init__) INFO: Log directory: /home/pi/.cache/mu/log
2018-02-27 22:11:55,492 - mu.logic:368(__init__) INFO: Data directory: /home/pi/.local/share/mu
2018-02-27 22:11:55,627 - mu.logic:376(setup) INFO: Available modes: adafruit, python, pygamezero, microbit, debugger
2018-02-27 22:11:55,826 - mu.logic:712(change_mode) INFO: Workspace directory: /home/pi/mu_code
2018-02-27 22:11:55,827 - mu.logic:415(restore_session) INFO: Restoring session from: /home/pi/.local/share/mu/settings.json
2018-02-27 22:11:55,828 - mu.logic:416(restore_session) DEBUG: {'workspace': '/home/pi/mu_code', 'mode': 'python', 'theme': 'day', 'paths': ['/home/pi/mu_code/hi.py'], 'microbit_runtime_hex': None}
2018-02-27 22:11:55,829 - mu.logic:470(_load) INFO: Loading script from: /home/pi/mu_code/hi.py
2018-02-27 22:11:55,830 - mu.logic:498(_load) DEBUG: print("hi")

print("bye")

print("welcome")

print("goodbye")

2018-02-27 22:11:56,485 - mu.logic:435(restore_session) INFO: Loaded files.
2018-02-27 22:11:56,562 - mu.logic:712(change_mode) INFO: Workspace directory: /home/pi/mu_code
2018-02-27 22:11:58,941 - mu.logic:668(show_logs) INFO: Showing logs from /home/pi/.cache/mu/log/mu.log
2018-02-27 22:14:10,305 - mu.modes.python3:162(run_script) DEBUG: print("hi")

print("bye")

print("welcome")

print("goodbye")

2018-02-27 22:14:10,321 - mu.interface.panes:602(start_process) INFO: Running script: /home/pi/mu_code/hi.py
2018-02-27 22:14:10,322 - mu.interface.panes:604(start_process) INFO: Running with interactive mode.
2018-02-27 22:14:10,323 - mu.interface.panes:607(start_process) INFO: Command args: []
2018-02-27 22:14:10,326 - mu.interface.panes:613(start_process) INFO: Working directory: /home/pi/mu_code
2018-02-27 22:14:10,327 - mu.interface.panes:618(start_process) INFO: Python path: ['/home/pi/mu', '/usr/lib/python35.zip', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-arm-linux-gnueabihf', '/usr/lib/python3.5/lib-dynload', '/usr/local/lib/python3.5/dist-packages', '/usr/lib/python3/dist-packages', '/usr/local/lib/python3.5/dist-packages/IPython/extensions']
2018-02-27 22:14:12,065 - mu.modes.python3:172(stop_script) DEBUG: Stopping script.
2018-02-27 22:14:12,576 - mu.modes.python3:183(debug) INFO: Starting debug mode.
2018-02-27 22:14:12,660 - mu.logic:712(change_mode) INFO: Workspace directory: /home/pi/mu_code
2018-02-27 22:14:12,660 - mu.modes.debugger:112(start) DEBUG: print("hi")

print("bye")

print("welcome")

print("goodbye")

2018-02-27 22:14:12,668 - mu.interface.panes:602(start_process) INFO: Running script: /home/pi/mu_code/hi.py
2018-02-27 22:14:12,668 - mu.interface.panes:607(start_process) INFO: Command args: []
2018-02-27 22:14:12,669 - mu.interface.panes:613(start_process) INFO: Working directory: /home/pi/mu_code
2018-02-27 22:14:12,669 - mu.interface.panes:618(start_process) INFO: Python path: ['/home/pi/mu', '/usr/lib/python35.zip', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-arm-linux-gnueabihf', '/usr/lib/python3.5/lib-dynload', '/usr/local/lib/python3.5/dist-packages', '/usr/lib/python3/dist-packages', '/usr/local/lib/python3.5/dist-packages/IPython/extensions']
2018-02-27 22:14:22,600 - mu.debugger.client:195(on_fail) ERROR: Connection timed out. Is your machine slow or busy? Free up some of the machine's resources and try again.
2018-02-27 22:14:23,858 - mu.logic:668(show_logs) INFO: Showing logs from /home/pi/.cache/mu/log/mu.log

ntoll, 2018-02-27:
How was this installed..? (It works for me.) When in the command prompt, does the command "mu-debug" autocomplete? If it doesn't, then Mu isn't installed properly. (My guess is that you're running from source? If you can confirm this, I'll email instructions for a fix tomorrow [just about to go to bed]).

martinohanlon, 2018-02-27:
Yes, running from source, built using:

```
sudo apt-get install python3-pyqt5 python3-pyqt5.qsci python3-pyqt5.qtserialport python3-pyqt5.qtsvg python3-dev

git clone https://github.com/mu-editor/mu

cd mu

sudo pip3 install -r requirements_pi.txt

python3 run.py
```

ntoll, 2018-02-27:
Aha... in your virtualenv and in the root of the Mu repos you should `pip install .`

The problem is there are two entry points into Mu: the `mu` command (which is the equivalent of `run.py`) and `mu-debug` which starts the debug runner process. If running from source, these commands are not immediately available since `pip` needs to run the setup.py code to make these entry points known to your OS.

This is a good spot and oversight on my part. These instructions should be in the developer documentation. Oops... sorry. :-/

martinohanlon, 2018-02-27:
How would I run the mu-debug from source?

ntoll, 2018-02-27:
I'm not sure I understand. *You* don't run `mu-debug`, Mu runs it in a child process, but for this to happen it needs to be a command that the OS knows about, hence the pip related instructions above (note - this only applies when running Mu from source). Does this help?

martinohanlon, 2018-02-28:
Sorry, if I wanted to startup My running the debugger from source how would I do it e.g. something like

```python
python3 run.py --debug
```

martinohanlon, 2018-02-28:
Ok, so looking at setup.py mu-debug uses

```
mu.app:run
```

So, I created a small program to launch mu in debug mode:

```
#!/usr/bin/env python3
from mu.app import debug

if __name__ == "__main__":
    debug()
```

It fails to start with the error:

```
Traceback (most recent call last):
  File "debug.py", line 6, in <module>
    debug()
  File "/home/pi/mu/mu/app.py", line 150, in debug
    filename = os.path.normcase(os.path.abspath(sys.argv[1]))
IndexError: list index out of range
```

martinohanlon, 2018-02-28:
Ok, so this is weird.

By editing the setup.py to remove the `install_requires` components I was able to run `sudo pip3 install .`
(if I didnt removed the `install_requires` pip3 fails due to missing packages)

Then if I run `python3 run.py` mu starts and the debugger works.

So what is pip doing during setup which enables the debugger?!

martinohanlon, 2018-02-28:
Ok, so in summary, it doesn't work into pip is run and pip installs the mu-debug entry point which the debugger needs.

For reference to install mu on a pi from source:

Run the following commands:

```
sudo apt-get install python3-pyqt5 python3-pyqt5.qsci python3-pyqt5.qtserialport python3-pyqt5.qtsvg python3-dev

git clone https://github.com/mu-editor/mu

cd mu

sudo pip3 install -r requirements_pi.txt
```

edit `mu/setup.py` and comment out the `install_requires` - they are already installed:

```
#    install_requires=['pycodestyle==2.3.1', 'pyflakes==1.6.0',
#                      'pyserial==3.4', 'pyqt5==5.10', 'qscintilla==2.10.2',
#                      'qtconsole==4.3.1', 'matplotlib==2.1.2',
#                      'pgzero==1.1', 'PyQtChart==5.10'],
```

Run pip to install mu:

```
sudo pip3 install .
```

Run mu:

```
python3 run.py
```

closed 2018-02-28 by martinohanlon

carlosperate, 2018-02-28:
>So what is pip doing during setup which enables the debugger?!

It registers `mu-debug` as a command that you can use in the terminal.

>By editing the setup.py to remove the install_requires components I was able to run sudo pip3 install .
>(if I didnt removed the install_requires pip3 fails due to missing packages)

I wonder if it would still fail when the required package version is already installed in the current environment.

If you have installed everything from apt-get and requirements_pi.txt, could you do a "pip freeze" to see the versions installed?
(@ntoll if versions are specified in setup.py, should they be in requirements.txt as well?)