mu-editor/mu #352: mu does not support python turtle programming

yonghuming, 2018-02-28:
* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 
i want to do some python turtle programming
i write a script 
```
from turtle import *
for x in range(4):
    forward(100)
    right(90)
```
if everything work i will get a square.,but actully i got an error
```
Traceback (most recent call last):
  File "c:\users\langxianmeng\mu_code\intro.py", line 1, in <module>
    from turtle import *
  File "turtle.py", line 107, in <module>
ModuleNotFoundError: No module named 'tkinter'
```

i use the  mu_2018-02-23_18_46_master_64db73d_32bit.exe on windows 10 64 bit system

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

martinohanlon, 2018-02-28:
In summary, Tkinter isnt installed on your computer, or Mu cant see it.

A couple of quick questions:
- are you using the windows installer? If so, have a look at https://github.com/mu-editor/mu/issues/340
- are you using linux distro? Often Tkinter isnt install by default. try `sudo apt install python3-tk` to install it.

ntoll, 2018-02-28:
Hi @yonghuming,

This is because tkinter isn't installed with the Windows installer. This is a known problem and it's on my todo list to fix. See #340 for the ticket that's tracking this.

In summary, it's a known problem and will be fixed soon..!

yonghuming, 2018-03-07:
@ntoll thank you i got it,

ntoll, 2018-03-21:
Fixed by #391, try a recent Windows installer to try it out (I've tested it on Windows 7 32bit and Windows 10 64bit).

closed 2018-03-21 by ntoll