mu-editor/mu #353: mu-debug fails to start - Linux, Windows, Raspberry Pi - is it still needed?

martinohanlon, 2018-02-28:
When you run `mu-debug` it fails to start raising the error:

```
C:\Users\Martin O'Hanlon-LT>mu-debug
Traceback (most recent call last):
  File "c:\users\martin o'hanlon-lt\appdata\local\programs\python\python36\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\martin o'hanlon-lt\appdata\local\programs\python\python36\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Martin O'Hanlon-LT\AppData\Local\Programs\Python\Python36\Scripts\mu-debug.exe\__main__.py", line 9, in <module>
  File "c:\users\martin o'hanlon-lt\appdata\local\programs\python\python36\lib\site-packages\mu\app.py", line 144, in debug
    filename = os.path.normcase(os.path.abspath(sys.argv[1]))
IndexError: list index out of range
```

[image]

This occurs on Windows, Linux and Raspberry Pi.

Is `mu-debug` still needed? It looks like it was created to launch the debugger around the beta8 stage, but the debugger runs as default now

carlosperate, 2018-02-28:
`mu-debug` is used internally to start the debugger process when the debugger view is opened, the app launches `mu-debug` with the script path and other flags.

This is probably related to https://github.com/mu-editor/mu/issues/346.

martinohanlon, 2018-02-28:
I understand now, that mu-debug is not a command to be run by the users, its an internal entry point.

closed 2018-02-28 by martinohanlon

carlosperate, 2018-02-28:
Btw, the error was a missing command line arguments, so not related to #_346.

@ntoll maybe there could be a simple print and exit when no arguments are given. Users could accidentally run the debugger instead of mu and this could be an intimidating error message.

ntoll, 2018-02-28:
Good call. I'll add a fix.

ntoll, 2018-02-28:
Fixed by 5c5d92c477346478a9df82065cfeb881ffd21dd2 ;-)