mu-editor/mu #357: Problem opening the REPL (Cannot connect to device Error)

yurikleb, 2018-03-01:
Hey, I'm having a problem with the REPL.
Installed on Ubuntu 16.04, opened MU, it seems like it found the board, and managed to load the code stored on it.
But when I try to open the REPL I get the : "Cannot connect to device on port /dev/ttyACM0" ERROR.
My user is in the "dialout" group.

I'm also getting serial data using "screen" in the terminal:
screen /dev/ttyACM0 115200

Also works well in sublime text.
Also in Windows everything works fine.

What else could be the issue?
Thanks!

ntoll, 2018-03-01:
Hi @yurikleb,

So Mu is written on Ubuntu 16.04 (and it works for me!). Ergo, it *should* work and we just (!) need to figure out what's the problem. 

It sounds to me that you have the correct permissions for reading and writing via USB serial since screen works. 

What version of Mu are you using? When you start Mu from the command line it'll tell you the location of its log file. Can you attach that to this ticket? Sometimes, it can take several seconds and up to a minute for the OS to make the connection detectable by Python, so this may just be a case of patience.

Also, having the device plugged in and pressing its restart button sometimes helps. Are you using a micro:bit or Adafruit board?

The more information I have the easier it will be for me to help.

Thanks!

yurikleb, 2018-03-06:
Hey @ntoll sorry for the slow response, it does work now although with some hiccups.
After I connect the board I do get the "a connection from a new device detected" notification. 
But if I try to open REPL I get the "Cannot connect to device on port /dev/ttyACM0" ERROR.
The REPL does open after a few attemts so I guess it needs some time to establish connection.

But then I get the following Traceback:
`Traceback (most recent call last):
  File "code.py", line 19, in <module>
KeyboardInterrupt: 

Adafruit CircuitPython 2.2.1 on 2018-01-23; Adafruit Feather M0 Adalogger with samd21g18
>>> T
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'T' is not defined
>>> AT
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'AT' is not defined
>>> AT
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'AT' is not defined
>>> ~x~~x~`

I then hit Ctrl+C and then Ctrl+D
Only then I get correct output from the board in the REPL.

I Did some more testing with the "screen" command in the terminal, and with [Sublime Text Serial Monitor Plugin](https://github.com/ThomasGerstenberg/serial_monitor) and they also don't connect right away sometimes (usually after a few seconds wait as you suggested), but when they do they do not cause "KeyboardInterrupt" like Mu does, I get to see the data flow from the board right away.

closed 2018-03-06 by yurikleb

ntoll, 2018-03-06:
Hey @yurikleb, thanks for the feedback.

OK... so what you're looking at is expected behaviour. When we looked into how people use the REPL with MicroPython based boards (especially learners), they expected to get a Python prompt rather than the potential "splurge" of data coming via the serial link. Ergo, when starting the REPL we send a KeyboardInterrupt (hence the messages you see). This forces the Python prompt. Of course, once in the prompt you can CTRL-D to restart the board (or press the on-board reset button).

However, this behaviour changes when you have the plotter enabled. Obviously, the plotter is expecting data via the serial link, and the REPL widget see's the plotter already waiting for data so won't issue the KeyboardInterrupt as it normally does.

Does this make sense..? ;-)

All feedback most welcome.

yurikleb, 2018-03-21:
Cool, Thanks!
How do I get the plotter to work, I can't see the button in the GUI.
I'm using version 1.0.0.beta.13

yurikleb, 2018-03-21:
Alright Just upgraded to 1.0.0.b.15 and can see it now.
Will keep experimenting!

ntoll, 2018-03-21:
:-) Expect a beta.16 sometime tomorrow or Friday (once I've cleared my PR / email backlog and done a bunch of testing).