mu-editor/mu #358: 1.0.0beta15  was creating folders sound/images - now crashes

jerryneedell, 2018-03-02:
mu seems to be creating folders on my CIRCUITPY drive:
sounds - with eep.wav
images - with alien.png and alien_hurt.png

I also find a copy of these on my host machine in the mu_code folder.

Is this a bug or a "feature"?

Tried a bit later and now getting:
```
Logging to /home/jerryneedell/.cache/mu/log/mu.log
Traceback (most recent call last):
  File "/usr/local/bin/mu", line 11, in <module>
    sys.exit(run())
  File "/usr/local/lib/python3.5/dist-packages/mu/app.py", line 121, in run
    editor.setup(setup_modes(editor, editor_window))
  File "/usr/local/lib/python3.5/dist-packages/mu/logic.py", line 387, in setup
    os.makedirs(images_path)
  File "/usr/lib/python3.5/os.py", line 241, in makedirs
    mkdir(name, mode)
OSError: [Errno 30] Read-only file system: '/media/jerryneedell/CIRCUITPY/images'
```

jerryneedell, 2018-03-02:
here is the mu.log
```
Starting Mu 1.0.0.beta.15
2018-03-01 22:51:10,050 - root:113(run) INFO: uname_result(system='Linux', node='Ubuntu-Macmini', release='4.13.0-36-generic', version='#40~16.04.1-Ubuntu SMP Fri Feb 16 23:25:58 UTC 2018', machine='x86_64', processor='x86_64')
2018-03-01 22:51:10,050 - root:114(run) INFO: Python path: ['/usr/local/bin', '/usr/local/lib/python3.5/dist-packages/click-6.7-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/adafruit_ampy-1.0.3-py3.5.egg', '/usr/lib/python35.zip', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-x86_64-linux-gnu', '/usr/lib/python3.5/lib-dynload', '/home/jerryneedell/.local/lib/python3.5/site-packages', '/usr/local/lib/python3.5/dist-packages', '/usr/lib/python3/dist-packages', '/home/jerryneedell/.local/lib/python3.5/site-packages/IPython/extensions']
2018-03-01 22:51:10,096 - mu.logic:355(__init__) INFO: Setting up editor.
2018-03-01 22:51:10,096 - mu.logic:366(__init__) INFO: Settings path: /home/jerryneedell/.local/share/mu/settings.json
2018-03-01 22:51:10,096 - mu.logic:367(__init__) INFO: Log directory: /home/jerryneedell/.cache/mu/log
2018-03-01 22:51:10,096 - mu.logic:368(__init__) INFO: Data directory: /home/jerryneedell/.local/share/mu
2018-03-01 22:51:10,115 - mu.logic:376(setup) INFO: Available modes: adafruit, pygamezero, debugger, python, microbit
2018-03-01 22:51:10,115 - mu.logic:386(setup) DEBUG: Creating directory: /media/jerryneedell/CIRCUITPY/images
2018-03-01 22:51:10,333 - mu.logic:393(setup) DEBUG: Creating directory: /media/jerryneedell/CIRCUITPY/sounds
2018-03-01 22:51:11,055 - mu.logic:712(change_mode) INFO: Workspace directory: /media/jerryneedell/CIRCUITPY
2018-03-01 22:51:11,055 - mu.logic:415(restore_session) INFO: Restoring session from: /home/jerryneedell/.local/share/mu/settings.json
2018-03-01 22:51:11,055 - mu.logic:416(restore_session) DEBUG: {'theme': 'day', 'mode': 'adafruit', 'paths': [], 'microbit_runtime_hex': None, 'workspace': '/media/jerryneedell/CIRCUITPY'}
2018-03-01 22:51:11,055 - mu.logic:435(restore_session) INFO: Loaded files.
2018-03-01 22:51:11,290 - mu.logic:444(restore_session) INFO: Starting with blank file.
2018-03-01 22:51:11,323 - mu.logic:712(change_mode) INFO: Workspace directory: /media/jerryneedell/CIRCUITPY
2018-03-01 22:51:11,510 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:12,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:13,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:14,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:15,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:16,428 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:17,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:18,425 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:19,425 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:20,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:20,664 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:20,698 - mu.modes.base:176(add_repl) INFO: Started REPL on port: /dev/ttyACM0
2018-03-01 22:51:20,699 - mu.modes.base:158(toggle_repl) INFO: Toggle REPL on.
2018-03-01 22:51:21,428 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:22,424 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:23,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:24,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:25,428 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:26,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:27,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:28,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:29,425 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:30,421 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:31,425 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:32,425 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:33,425 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:34,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:35,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:36,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:37,422 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:38,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:39,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:40,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:51:41,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
[... cut ...]
2018-03-01 22:52:01,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:52:02,425 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:52:03,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:52:04,425 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:52:05,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-01 22:52:06,349 - mu.logic:668(show_logs) INFO: Showing logs from /home/jerryneedell/.cache/mu/log/mu.log
```

ntoll, 2018-03-02:
Hi @jerryneedell 

It's a bug! Nice find and sorry 'bout that. It should be easy to fix and I'll make it a top priority for the next release (coming out very soon).

ntoll, 2018-03-02:
Hmmm... out of interest, in a Python3 REPL on your machine (and if required, in the same virtualenv that you use to run Mu) can you tell me what the output is of this:

```
import os.path
os.path.expanduser('~')
```
Thanks.

jerryneedell, 2018-03-02:
I am not using a python virtual environment - installed mu-editor via pip  - also tried runni  run.py in a clone of the repo with the same result.

```Python 3.5.2 (default, Nov 23 2017, 16:37:01) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import os.path
>>> os.path.expanduser('~')
'/home/jerryneedell'
>>> 
```

jerryneedell, 2018-03-02:
FYI - it does not "recreate" the folders if they are already there: Here is the log from most recent startup"
```
2018-03-02 07:17:58,582 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.0.beta.15
2018-03-02 07:17:58,582 - root:113(run) INFO: uname_result(system='Linux', node='Ubuntu-Macmini', release='4.13.0-36-generic', version='#40~16.04.1-Ubuntu SMP Fri Feb 16 23:25:58 UTC 2018', machine='x86_64', processor='x86_64')
2018-03-02 07:17:58,582 - root:114(run) INFO: Python path: ['/usr/local/bin', '/usr/local/lib/python3.5/dist-packages/click-6.7-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/adafruit_ampy-1.0.3-py3.5.egg', '/usr/lib/python35.zip', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-x86_64-linux-gnu', '/usr/lib/python3.5/lib-dynload', '/home/jerryneedell/.local/lib/python3.5/site-packages', '/usr/local/lib/python3.5/dist-packages', '/usr/lib/python3/dist-packages', '/home/jerryneedell/.local/lib/python3.5/site-packages/IPython/extensions']
2018-03-02 07:17:58,635 - mu.logic:355(__init__) INFO: Setting up editor.
2018-03-02 07:17:58,635 - mu.logic:366(__init__) INFO: Settings path: /home/jerryneedell/.local/share/mu/settings.json
2018-03-02 07:17:58,635 - mu.logic:367(__init__) INFO: Log directory: /home/jerryneedell/.cache/mu/log
2018-03-02 07:17:58,635 - mu.logic:368(__init__) INFO: Data directory: /home/jerryneedell/.local/share/mu
2018-03-02 07:17:58,657 - mu.logic:376(setup) INFO: Available modes: pygamezero, microbit, debugger, python, adafruit
2018-03-02 07:17:58,716 - mu.logic:712(change_mode) INFO: Workspace directory: /home/jerryneedell/mu_code
2018-03-02 07:17:58,716 - mu.logic:415(restore_session) INFO: Restoring session from: /home/jerryneedell/.local/share/mu/settings.json
2018-03-02 07:17:58,716 - mu.logic:416(restore_session) DEBUG: {'mode': 'adafruit', 'workspace': '/home/jerryneedell/mu_code', 'paths': [], 'microbit_runtime_hex': None, 'theme': 'day'}
2018-03-02 07:17:58,716 - mu.logic:435(restore_session) INFO: Loaded files.
2018-03-02 07:17:58,804 - mu.logic:444(restore_session) INFO: Starting with blank file.
2018-03-02 07:17:58,818 - mu.logic:712(change_mode) INFO: Workspace directory: /media/jerryneedell/CIRCUITPY
2018-03-02 07:17:59,631 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:00,582 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:01,530 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:02,483 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:02,775 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:02,783 - mu.modes.base:176(add_repl) INFO: Started REPL on port: /dev/ttyACM0
2018-03-02 07:18:02,783 - mu.modes.base:158(toggle_repl) INFO: Toggle REPL on.
2018-03-02 07:18:03,434 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:04,434 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:05,433 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:06,437 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:07,435 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:08,431 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:09,434 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:29,045 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:29,994 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:30,942 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:31,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:32,933 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:33,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:34,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:35,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:36,933 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:37,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:38,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:39,935 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:40,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:41,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:42,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:43,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:44,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:45,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:46,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:47,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:48,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
[... cut ...]
2018-03-02 07:18:51,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:52,928 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:53,963 - mu.interface.main:230(get_load_path) DEBUG: Getting load path: 
2018-03-02 07:18:53,991 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:54,934 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 07:18:55,365 - mu.logic:668(show_logs) INFO: Showing logs from /home/jerryneedell/.cache/mu/log/mu.log
```

jerryneedell, 2018-03-02:
Sorry for the misleading title for the issue. I thought I edited it. The "crashes" weer due to a corrupted  files system taht was mounting as read-only. When mu tried to create the souids/images folders, it could not and threw an erro and exited. After I repaired the file system, mu want back to working and creating the folders.

ntoll, 2018-03-02:
But is it still trying to recreate them on your Adafruit board..?

jerryneedell, 2018-03-02:
If I start mu on a board that already has the folders in place, the log does not show them be re-creadted. If I delete them from the board, it re-creates them.

ntoll, 2018-03-02:
Hmmm... that shouldn't happen. ;-)

I'm grabbing my Adafruit board and trying to recreate right now. Thanks for all the feedback. Here's hoping I can recreate it.

jerryneedell, 2018-03-02:
here are two logs: first I had deleted the folders and started mu - it re-creates them second, I just closed and restarted mu.

```

2018-03-02 08:51:03,144 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.0.beta.15
2018-03-02 08:51:03,144 - root:113(run) INFO: uname_result(system='Linux', node='Ubuntu-Macmini', release='4.13.0-36-generic', version='#40~16.04.1-Ubuntu SMP Fri Feb 16 23:25:58 UTC 2018', machine='x86_64', processor='x86_64')
2018-03-02 08:51:03,144 - root:114(run) INFO: Python path: ['/usr/local/bin', '/usr/local/lib/python3.5/dist-packages/click-6.7-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/adafruit_ampy-1.0.3-py3.5.egg', '/usr/lib/python35.zip', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-x86_64-linux-gnu', '/usr/lib/python3.5/lib-dynload', '/home/jerryneedell/.local/lib/python3.5/site-packages', '/usr/local/lib/python3.5/dist-packages', '/usr/lib/python3/dist-packages', '/home/jerryneedell/.local/lib/python3.5/site-packages/IPython/extensions']
2018-03-02 08:51:03,197 - mu.logic:355(__init__) INFO: Setting up editor.
2018-03-02 08:51:03,197 - mu.logic:366(__init__) INFO: Settings path: /home/jerryneedell/.local/share/mu/settings.json
2018-03-02 08:51:03,197 - mu.logic:367(__init__) INFO: Log directory: /home/jerryneedell/.cache/mu/log
2018-03-02 08:51:03,197 - mu.logic:368(__init__) INFO: Data directory: /home/jerryneedell/.local/share/mu
2018-03-02 08:51:03,216 - mu.logic:376(setup) INFO: Available modes: adafruit, python, microbit, debugger, pygamezero
2018-03-02 08:51:03,217 - mu.logic:386(setup) DEBUG: Creating directory: /media/jerryneedell/CIRCUITPY/images
2018-03-02 08:51:03,422 - mu.logic:393(setup) DEBUG: Creating directory: /media/jerryneedell/CIRCUITPY/sounds
2018-03-02 08:51:03,584 - mu.logic:712(change_mode) INFO: Workspace directory: /media/jerryneedell/CIRCUITPY
2018-03-02 08:51:03,585 - mu.logic:415(restore_session) INFO: Restoring session from: /home/jerryneedell/.local/share/mu/settings.json
2018-03-02 08:51:03,585 - mu.logic:416(restore_session) DEBUG: {'workspace': '/media/jerryneedell/CIRCUITPY', 'theme': 'day', 'microbit_runtime_hex': None, 'mode': 'adafruit', 'paths': []}
2018-03-02 08:51:03,585 - mu.logic:435(restore_session) INFO: Loaded files.
2018-03-02 08:51:03,670 - mu.logic:444(restore_session) INFO: Starting with blank file.
2018-03-02 08:51:03,685 - mu.logic:712(change_mode) INFO: Workspace directory: /media/jerryneedell/CIRCUITPY
2018-03-02 08:51:04,491 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:05,442 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:06,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:07,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:08,425 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:09,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:10,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:11,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:12,428 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:13,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:14,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:15,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:16,424 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:17,425 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:18,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:19,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:20,425 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:21,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:22,428 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:23,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:24,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:25,425 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:26,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:27,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:28,429 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:29,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:30,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:31,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:32,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:33,426 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:34,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:51:35,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
[... cut ...]
2018-03-02 08:53:02,530 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:53:02,536 - mu.modes.base:176(add_repl) INFO: Started REPL on port: /dev/ttyACM0
2018-03-02 08:53:02,537 - mu.modes.base:158(toggle_repl) INFO: Toggle REPL on.
2018-03-02 08:53:03,428 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:53:04,427 - mu.modes.base:127(find_device) INFO: Found device on port: ttyACM0
2018-03-02 08:53:05,361 - mu.logic:668(show_logs) INFO: Showing logs from /home/jerryneedell/.cache/mu/log/mu.log
```

ntoll, 2018-03-02:
As an aside... this also indicates that I need to fix device detection logging... we only need log when a device is *first* detected (not subsequent detections). You've inadvertently found two bugs! :-)

jerryneedell, 2018-03-02:
I'm on a role ;-)
FYI - I had been doing this on a Metro_M4_express_revb  but I get teh same result on a Metro_M0_express.  Both running Circuitpython 3.0 (latest master)

ntoll, 2018-03-02:
Hurrah... I've managed to recreate the problem. Now all I need to do is figure out why it's behaving strangely. Whatever the fix is, it'll be in the next version. I'll reference the fix in this issue so you have visibility on the progress.

jerryneedell, 2018-03-02:
Glad you are able to reproduce it! 
Thank you for all you are doing to provide this tool!

jerryneedell, 2018-03-02:
One update - since you can reproduce this you may have already found this:
Sometimes it does not create these folders on the first start-up of mu but it does if mu is close and reopened.
Fo example, I plugged in a clean (nothing on file system) itsybitsy_m0 (CP 2.2.3) and when I started mu there were no folders created. I closed mu and restarted and it and then created the sound/images folders.

ntoll, 2018-03-02:
So the problem is how Mu works out what your "workspace directory" (to use Mu parlance) is. This is the place where Mu should save your code. Given Mu is for beginner programmers, who may not know anything about file-systems, after quite a bit of feedback from teachers we decided to put all code in an arbitrary place (the workspace). When Mu starts it checks what's in the workspace and, if needed, copies over some assets used by some of the other modes (Pygame in this case).

However, when you have an Adafruit board Mu does the right thing and knows to save your code on the device. In effect, the workspace directory is the device's path (you can see where this is going...).

Here's the problem, when Mu starts up it reads a settings file in which the default workspace directory is defined. Unfortunately, when Mu shuts down it also stores its current workspace directory in the settings file (this is the bug). If you close down Mu while you have an Adafruit device plugged in and while Mu is in Adafruit mode, then the workspace dir is changed to the path for your device.

The fix should be easy to do. Finding the bug was a bit of an adventure though.

Thanks for contributing the bug report. Without your effort I'd be none-the-wiser. Thanks to you, it's getting fixed. :-)

jerryneedell, 2018-03-02:
Glad to help.  So that explains why it does not show up on the first connection.  Nice work tracking it down!

closed 2018-03-12 by ntoll

ntoll, 2018-03-12:
OK... a fix for this just landed in master and it'll be in the next release.

Thank you for taking the time to report this. :-)

jerryneedell, 2018-03-15:
Thanks for the quick turn-around! Looking forward to the update.

commit 2b5d8c4f (2018-03-12): Separate out session and settings configuration. Fixes #358 and gives more flexibility.