mu-editor/mu #360: 1.0.0.beta.15 Crash and delete content in file

smeand, 2018-03-02:
I was trying to correct something in my code. Suddenly the IDE crashes and shuts down quickly. I started again but the tab was empty and I tried to load the same file again but the file was empty (file existed with right <name.py>not content. I had the same issue yesterday with another file. I hope I got the right section from the logfile.
------------------------------------------------------------------------------------------
2018-03-02 13:26:21,911 - mu.logic:466(new) INFO: Added a new tab.
2018-03-02 13:26:25,852 - mu.modes.base:130(find_device) WARNING: Could not find device.
2018-03-02 13:26:25,852 - mu.modes.base:131(find_device) DEBUG: Available ports:
2018-03-02 13:26:25,852 - mu.modes.base:135(find_device) DEBUG: ['PID:0 VID:0 PORT:COM1', 'PID:15207 VID:32902 PORT:COM3']
2018-03-02 13:26:25,855 - mu.interface.main:650(show_message) DEBUG: Could not find an attached device.
2018-03-02 13:26:25,855 - mu.interface.main:651(show_message) DEBUG: Please make sure the device is plugged into this computer.

It must have a version of MicroPython (or CircuitPython) flashed onto it before the Plotter will work.

Finally, press the device's reset button and wait a few seconds before trying again.
2018-03-02 13:26:28,848 - mu.modes.base:205(toggle_plotter) INFO: Toggle plotter on.
2018-03-02 13:41:42,782 - mu.logic:732(autosave) INFO: Autosave has detected changes.
2018-03-02 13:41:42,783 - mu.logic:736(autosave) INFO: Saving script to: C:/Users/andber/mu_code/skicka alfabet.py
2018-03-02 13:41:42,897 - root:101(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Program Files\Mu\pkgs\mu\logic.py", line 738, in autosave
    write_and_flush(f, tab.text())
  File "C:\Program Files\Mu\pkgs\mu\logic.py", line 144, in write_and_flush
    fd.write(content)
  File "encodings\cp1252.py", line 19, in encode
UnicodeEncodeError: 'charmap' codec can't encode character '\ufffd' in position 989: character maps to <undefined>
2018-03-02 13:41:49,978 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.0.beta.15
2018-03-02 13:41:49,978 - root:113(run) INFO: uname_result(system='Windows', node='SMB3043', release='7', version='6.1.7601', machine='AMD64', processor='Intel64 Family 6 Model 37 Stepping 2, GenuineIntel')
2018-03-02 13:41:49,978 - root:114(run) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2018-03-02 13:41:50,706 - mu.logic:355(__init__) INFO: Setting up editor.
2018-03-02 13:41:50,707 - mu.logic:366(__init__) INFO: Settings path: C:\Users\andber\AppData\Local\python\mu\settings.json
2018-03-02 13:41:50,707 - mu.logic:367(__init__) INFO: Log directory: C:\Users\andber\AppData\Local\python\mu\Logs
2018-03-02 13:41:50,707 - mu.logic:368(__init__) INFO: Data directory: C:\Users\andber\AppData\Local\python\mu
2018-03-02 13:41:50,734 - mu.logic:376(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-03-02 13:42:01,690 - mu.logic:712(change_mode) INFO: Workspace directory: C:\Users\andber\mu_code
2018-03-02 13:42:01,691 - mu.logic:415(restore_session) INFO: Restoring session from: C:\Users\andber\AppData\Local\python\mu\settings.json
2018-03-02 13:42:01,691 - mu.logic:416(restore_session) DEBUG: {'theme': 'day', 'mode': 'microbit', 'paths': ['C:/Users/andber/mu_code/skicka alfabet.py', 'C:/Users/andber/mu_code/radio sänd text.py', 'C:/Users/andber/mu_code/testplot.py'], 'workspace': 'C:\\Users\\andber\\mu_code', 'microbit_runtime_hex': None}
2018-03-02 13:42:01,691 - mu.logic:470(_load) INFO: Loading script from: C:/Users/andber/mu_code/skicka alfabet.py
2018-03-02 13:42:01,691 - mu.logic:498(_load) DEBUG: 
2018-03-02 13:42:02,201 - mu.logic:470(_load) INFO: Loading script from: C:/Users/andber/mu_code/radio sänd text.py
2018-03-02 13:42:02,209 - mu.logic:498(_load) DEBUG: from microbit import *

tjguk, 2018-03-02:
Looks like the autosave functionality doesn't specify an output encoding. @ntoll should we be using "UTF-8" across the board? On Windows, by default you'l get cp1252

Also, the "write_and_flush" function appears to be expecting both a high-level Python file object and a a low-level file descriptor which it tries to sync.

tjguk, 2018-03-02:
From the docs (https://docs.python.org/3/library/os.html#os.fsync) that should probably be os.fsync(fd.fileno())

ntoll, 2018-03-02:
Hi,

Thanks for the report!

The problem appears to be indicated by this line in the logs:

"UnicodeEncodeError: 'charmap' codec can't encode character '\ufffd' in position 989: character maps to "

ntoll, 2018-03-02:
@tjguk you beat me to it! :-)

Yeah... so I was wondering if this was a Windows-ish thing. Happily we only use write_and_flush (or we should only be using it) so we only need change the code in one place.

ntoll, 2018-03-02:
@smeand also, apologies for the crashing... but thanks for the bug report! It's only because you reported this bug that we can fix it! :+1:

carlosperate, 2018-03-02:
This might be related: https://github.com/mu-editor/mu/issues/231 https://github.com/mu-editor/mu/issues/237

carlosperate, 2018-03-03:
Btw, losing the file during save is what the atomic_file module was aiming to mitigate in https://github.com/mu-editor/mu/pull/208.
I think that was removed at one point, was it because of the location of the temporary file causing problems in boards that save the python script in their file system?
The temp location could have been changed to the user data directory and have Mu detect if the original file was lost (original missing + temp file available) and recover.

ntoll, 2018-03-03:
The atomic file saving created all sorts of mess on CircuitPython boards which are programmed to restart when they detect something has been saved to their filesystem. :-(

carlosperate, 2018-03-03:
> The atomic file saving created all sorts of mess on CircuitPython boards which are programmed to restart when they detect something has been saved to their filesystem. :-(

But wasn't the problem just that temporary file was saved on the same directory than the opened python file? Because that temporary file could be save into the user files directory instead, so in that case it shoudn't touch the CircuitPython file system any more than the current implementation does.

ntoll, 2018-03-04:
Good point... so you're suggesting we put the temporary file somewhere safe but unseen by the developer? Python already has a comprehensive module for making temporary files in the correct location (e.g. `/tmp/mu`) for each OS (https://docs.python.org/3/library/tempfile.html).

carlosperate, 2018-03-04:
> Good point... so you're suggesting we put the temporary file somewhere safe but unseen by the developer?

Yes.

> Python already has a comprehensive module for making temporary files in the correct location (e.g. /tmp/mu) for each OS (https://docs.python.org/3/library/tempfile.html).

Yes that would work. If we want to be able to recover unsaved changes after a crash we could manage a temporary folder inside the directory we already create for the settings and log files, but maybe that's out of scope.

ntoll, 2018-03-21:
The root cause of this crash appears to be fixed by #399. Atomic saves (or not) that fell out of this discussion remain to be dealt with.

closed 2018-03-21 by ntoll