mu-editor/mu #361: Mu always adds .py to a file, even if the filename as a file extenstion.

martinohanlon, 2018-03-02:
When saving a file Mu will always add a .py extension to the end of the file. 

Super, unless of course you want to create a file which has a different file extension, e.g. a .txt file which your program loads in at runtime.

Using this model is really common when getting learners to work with files.

.e.g 
First create a file which has a list of names in it and called it names.txt. 
Open names.txt in your Python program.
Read in the names.
etc, etc.

martinohanlon, 2018-03-02:
I would recommend that if a filename is presented which doesnt have a file extension it adds .py, if a file extension is included it doesn't.

ntoll, 2018-03-02:
+1 on that suggested behaviour.

martinohanlon, 2018-03-02:
Submitted PR https://github.com/mu-editor/mu/pull/362

closed 2018-03-02 by ntoll