mu-editor/mu #363: Add a warning if the filename in mu shadows another module

tjguk, 2018-03-02:
cf https://gist.github.com/waveform80/c9d7c9f81d6b67028c90eac5f4080053

There is a common problem with beginners where they call their first attempt at using the turtle module turtle.py, or their picamera demo picamera.py. This instantly (but slightly obscurely) causes it not to work as the first thing it'll do is try to import itself and then find none of the expected names!

@waveform80 has put together an importlib hack to detect this situation and offer a warning.

We should consider something like this for Mu

bennuttall, 2018-06-09:
This is awesome! It's my biggest pet peeve with Python and this solution doesn't just avoid the problem but it also teaches users about the problem so they learn for future (when perhaps they're using another editor).

ntoll, 2018-06-20:
OK... this is cool. I'll take @waveform80's gist and add it to the functionality of the "Check" button.

ntoll, 2018-07-06:
Fixed in 0e6b89bc7d6631bdd9c8fa7158600c4f924b53bb

ntoll, 2018-07-06:
In the end I didn't go with the code you referenced @tjguk. Rather, I discovered a *far simpler* solution exists via the `pkgutils` module (i.e. `iter_modules`). I have to manually add `sys` and `builtin` to the set since they don't get picked up by this function. 

In any case, I hope many hours of head-scratching are avoided by this update. It'll be in the beta.17 release on Monday. :-)

Thanks for your suggestion and help.

closed 2018-07-06 by ntoll