mu-editor/mu #365: No clarity on how to install MU on current 'stretch' Raspbian

DavidMS51, 2018-03-03:
I may just be thick but I cannot find anywhere here or elsewhere a clear set of instructions (however messy) on how to get MU working on a Raspberry Pi running the latest Raspbian.  I can find lots of things that suggest it is a problematic.  I feel unless a clear set of worked instructions can be offered - which make no presumptions of peoples knowledge of operating systems or pip etc the 'claim' that this package works with the Raspberrypi should be removed until the issues can be resolved 

As of this morning I have installed everything listed in 'requirements_pi.txt'  - but even here how do run this as an automatic activity 'pip install requirements_pi.txt' returns the error 'could not find a version that matches the requirement ..... ?

Then I have tried 'sudo apt-get install mu_editor' as suggsted on Adafruit site - this tells me PyQtchart 5.1 is missing?

then I thought I would give 'sudo apt-get install mu' a try but that apparently does not exist ..

ZanderBrown, 2018-03-03:
Have you tried `sudo pip3 install mu-editor`

tjguk, 2018-03-03:
I think the story here is that, although https://codewith.mu mentions that the editor works with RPi, it doesn't say *how* to make it work. So even if (hopefully) we can help get @DavidMS51 working, we need to add something over there to say, eg, "sudo pip3 install mu-editor" (or whatever other incantation does the trick)

carlosperate, 2018-03-03:
The current instructions are located in: https://github.com/mu-editor/mu/blob/master/docs/setup.rst#raspberry-pi

This is not yet on the developer docs (https://mu.readthedocs.io/en/latest/setup.html#raspberry-pi) as it has been a update very recently.

The current version is still on beta, and things are bound to be a bit broken until the release is ready. Once Mu reaches that point, installing on raspberry pi should be a `apt-get install` single command.

> I think the story here is that, although https://codewith.mu mentions that the editor works with RPi, it doesn't say how to make it work. So even if (hopefully) we can help get @DavidMS51 working, [...]

Yes, the main website need to be updated as well. I assumed the old debian package was still on the raspberry pi repos, even though it's very very old (v0.1).

ntoll, 2018-03-03:
Hi Folks,

@DavidMS51, sorry to hear of your travails with this. As @carlosperate correctly points out, Mu is currently gearing up for a 1.0.0 release and so things will appear broken as old stuff is gradually replaced by new things. For example, I've been working a lot on the new version of the website just this past week, and liaising with Debian and Raspbian developers so Mu 1.0.0 should be just an "sudo apt-get" away.

Finally, Mu is entirely volunteer driven, so there are bound to be niggles. Having said that, we really appreciate you taking the time to report these pain points simply because they inform our decision on what to concentrate on next.

Thanks for your patience,

N.

DavidMS51, 2018-03-03:
Thanks for all the responses, I do fully understand that this is volunteer driven.  In this spirit once there is is a clear 'how to'  I am happy to put some effort into posting on Rpi forums which have threads of people either saying it does not work or have incorrect instructions - ie that became obsolete with the November Raspbian update. There are also a few blogs with out of date info so where I know the owners I will see if they will delete the out of date stuff.

In terms of the specific questions

- good to understand why the rpi instructions on 'readthedocs' are blank
- yes my starting point was  https://codewith.mu
- sudo pip3 install mu-editor   - does not work says it is missing qscintilla==2.10.2

I have run through the instructions in  carlosperate 's response above, I got to instruction 7 before things ground to a halt (sorry)

also

- virtualenv was missing - well at least for me and needed to be installed via apt-get
- instruction 6 says you need to be in mu-venv - should read 'mu-venv/mu' ?
- the comment out instruction I think is missing the 4th line of the install requires instruction ?
- instruction 7 is missing what you are actually installing, I am assuming this should be 'mu' but it does not work if I try that from either mu-venv or mu-venv/mu

tjguk, 2018-03-03:
Thanks @DavidMS51 that is very useful feedback. We can work through those things here to try to address everything that we reasonably can. And thanks for your volunteering to help clear things up downstream once we've pulled it together.

ntoll, 2018-03-04:
@DavidMS51 we're on this. The work on the next version of the website should fix many of the problems you've faced: simply, it'll be the single point of truth and kept up-to-date with the current version of Mu once we hit 1.0.

DavidMS51, 2018-03-04:
Thanks that makes good sense, I have gone back through the instructions carlosperate pointed to and seem now to have the beta running which is great
- ignore instruction 7
- run the modified setup.py with 'python setup.py develop'

running mu with python3 run.py  then certainly allows me to connect to a microbit and get messages scrolling on the screen

closed 2018-03-07 by DavidMS51