mu-editor/mu #367: Automate OSX installation target that isn't based on PyInstaller

ntoll, 2018-03-03:
PyInstaller creates something that doesn't work properly for Python3 mode (if it's like the resulting PyInstaller executable on Windows, the debugger and REPL don't work).

I've looked at some of the other Python packaging tools for OSX (to create a .app) and nothing appears to work (specifically py2app and briefcase).

Homebrew seems to be a possible solution to this problem. We should investigate the automation of creating the homebrew recipe needed to package/install Mu.

carlosperate, 2018-03-03:
Personally I'm not sure I like the idea of using brew. Being a beginners editor it feels like brew is too "developer" oriented. That being said, I don't really know if there is a better alternative.

I wonder what is better, asking to install brew and then mu, or asking to install Python 3.6 and the pip install mu-editor. Both option seem to be on the same "difficulty level" to me.

Just to refresh my mind, what was the exact issue with PyInstaller, that you cannot spun a new process with QProcess? Is that only for mu-debug or are there more? Was there any other reason?
The reason I ask is because you another option is to build "mu-debug" into it's own executable and call that from the mu-editor executable. We could also try to make another executable that "execs" (https://docs.python.org/3.3/library/functions.html#exec) a given file.

ntoll, 2018-03-03:
Hey @carlosperate, yes, it's the reasons you mention (which I encountered when trying to create the Windows installer -- happily Thomas's pynsist is much simpler and more flexible than PyInstaller).

I agree with you about homebrew... but I feel like we're being backed into a corner here... none of the other options produce anything that works! (And at least brew is, in some way, "native"). Looking at the layout of a .app I wonder how hard it would be to bespoke-bake one ourselves with a script but that possibility gives me the collywobbles for all sorts of NIH related reasons.

Regarding `exec` -- as you'll see by the debug runner (https://github.com/mu-editor/mu/blob/master/mu/debugger/runner.py#L468) this is already exactly what we're doing. Are you suggesting we do this in a new thread? This would create problems in terms of the program being executed inside the same process/context as Mu itself and we'd need to work out the implications of this (although I'm probably misunderstanding you). Having it in a separate process is very useful in that everything is nice and isolated IYSWIM.

If we can build mu-debug into its own executable (the script to launch such an executable already exists: https://github.com/mu-editor/mu/blob/master/mu/mu-debug.py). But I don't know enough about PyInstaller to understand how all this would fit together.

Ultimately, we need a beginner / teacher friendly "just download this" set of instructions. Homebrew doesn't quite meet that, but it's better than nothing, and using pip/Python3 might just be the only option we have unless we invest much more time in figuring out this rather infuriating jigsaw.

Does this make sense? As always, I'm open to suggestions and experiments that allow us to make as simple a solution as possible for the end user.

carlosperate, 2018-03-04:
>I agree with you about homebrew... but I feel like we're being backed into a corner here... none of the other options produce anything that works! (And at least brew is, in some way, "native").

Yeah, I definitely know what you mean, it doesn't feel like there is anything like pynsist on mac, and all the other options have worse drawbacks.
But is really more"native"? That's an honest question, because all the non-developer mac user I know haven't really heard of it, but my sample size here is tiny, barely a handful.

> Ultimately, we need a beginner / teacher friendly "just download this" set of instructions. Homebrew doesn't quite meet that, but it's better than nothing, and using pip/Python3 might just be the only option we have unless we invest much more time in figuring out this rather infuriating jigsaw.

So "install brew + brew install mu-editor" better than "install python + pip install mu-editor"? From my point of view second one might be a bit more "educational friendly"? (although it's hard to define what that means anyway)

> Regarding exec -- as you'll see by the debug runner (https://github.com/mu-editor/mu/blob/master/mu/debugger/runner.py#L468) this is already exactly what we're doing. Are you suggesting we do this [...]

I guess my main question was to know what exactly from the current implementation was failing in PyInstaller, just to do some digging/experimentation to find a possible walk around.
For instance, if the only issue was be able to execute `mu-debug` as done [here](https://github.com/mu-editor/mu/blob/4722c727df90fa79d0a1a3529ac25d0ef5c8574a/mu/interface/panes.py#L622) then we could create the executable in three phases:
1) First create a pyinstaller single file executable for mu-debug
2) Create an pyinstaller app bundle for mu-editor
3) Move the mu-debug executable inside the mu-editor app bundle

And then the panes.py file could have a bit of code to detect if the application was frozen to know exactly how to launch mu-debug.

Maybe the other issues with PyInstaller could be a bit more difficult to overcome.

paularmand, 2018-04-26:
Looking for a solution for freezing for os x. I see that the release page still contains a zip with a freezed app for os x, so I am assuming that the pyinstaller still works for the latest versions. Is this correct? I can not seem to get it to work, or maybe the latest releases were done manually or via py2app (the documentation alludes to the fact that this is a possible route for the future). We need the freezed app for an multiday event involving coding for kids and I would be okay with spending time for coding/testing  on it if some advice/guidance could be provided. Thanks.

carlosperate, 2018-04-26:
Hi @paularmand 

> I see that the release page still contains a zip with a freezed app for os x

Do you mean the download from http://codewith.mu ? That version is quite old and doesn't contain the incompatible changes with PyInstaller.

If you mean the files in th AWS S3 bucket, then those are automatically generated by the CI servers, but are tests builds and never guaranteed to work (which they currently don't for macOS and Linux).

> I am assuming that the pyinstaller still works for the latest versions. Is this correct?

Not really, the current implementation is incompatible with the way that PyInstaller, or most freezing options, would package the application into a single executable or an app bundle. This doesn't mean that it couldn’t be fixed to work with these, just that the current implementation would need to be updated to be able to do this.

> ...or maybe the latest releases were done manually or via py2app (the documentation alludes to the fact that this is a possible route for the future).

@ntoll, unless you've found a way to do this with py2app (which I don't know if it offers any advantages over PyInstaller, but that's not a criticism just ignorance from my part), then maybe that bit should be removed to avoid confusion? 

> We need the freezed app for an multiday event involving coding for kids and I would be okay with spending time for coding/testing on it if some advice/guidance could be provided. Thanks.

To better understand your situation and requirements, why do you need the app frozen? Could you install Python 3 and `pip install mu-editor`?

As far as what is needed to freeze the app with PyInstaller/py2app, I think the first steps is to understand what the original issue was, @ntoll could provide a bit more info there.
If all we need is a way to launch `mu-debug` as it's own process, then we could do what I mentioned in my last message, but I don't know if the piping or comms mechanism created by Qt would work that way, or if there are other areas in Mu that might need updating for a frozen app.

Apart from that there has been a significant change in how PyQt5 is packaged in PyInstaller, so that might be an area that could need some testing and possibly debugging as well.

paularmand, 2018-05-02:
Hi @carlosperate, thanks for your detailed answer. I understand a little better the history, thanks. We cannot install python3 nor pip nor mu-editor easily. The reason is that we are helping out [fri3dcamp](http://fri3d.be/english/ ), and cannot control the laptops of the participants. The mu-editor would allow the participants (large part kids, since it's a family oriented camp) to get started easily. As I understand, the build is broken since the introduction of modes. Is that correct? We would like a mu-editor build that supports modes, and can be frozen. Would there be a version that allows that (the 0.9.13 version available at codewithmu doesnt allow modes, right?). Thanks in advance. Paul

carlosperate, 2018-05-03:
Hi @paularmand,
v0.9.13 is pre-modes, so that one only has the micro:bit functionality. We are currently looking into alternative ways to package Mu for macOS, but we don't really know how viable or how much work it might be, as for some of us this is uncharted territory :)

> ... and cannot control the laptops of the participants

This is the difficult part, as there are a lot of variables and variations of OS environments, and we could really use a hand beta testing any form of Mu package we could come up with.  Do you have access to different mac computers? Ideally some with older versions of macOS/OS X, as our development machines are usually running with up-to-date software and that might not be the case for all users.

paularmand, 2018-05-04:
Hi @carlosperate, I can ask the members of our hackerspace to test it. Some have OS X, but I'm pretty confident that the SW will also be up-to-date. The organisers of Fri3dCamp however will probably be interested for the IDE in their open garage project, and we also have quite a number of other workshop organisers, I'll ask around, and we'll try to help with the testing yes. Do we only need to vary the os x version, or do you want to control/test other variables too?

ntoll, 2018-05-19:
OK folks, it sort-of-works! :-) Hurrah.

Check out the dmg image here:

http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=osx/

Which was automatically built on Travis.

Known bugs:

* Python 3 mode Debugger and REPL don't work... I suspect a path issue.
* PyGameZero mode needs checking, for the same reasons.
* Anything that uses `tkinter` won't work (yet)... e.g. turtle and GUIZero modules.
* Lack of license in DMG steps...
* Needs signing key so OSX doesn't get the collywobbles when you try to use Mu.

None of the above are insurmountable.

However, the most important step (automating the creation of an installer that is easy to use and the end result mostly works) is done. The work is on my `package_macos` branch if you're interested in how it's done.

darrell24015, 2018-05-20:
The drag-n-drop install with the dmg file worked fine.  Launching the mu-editor.app failed, attached the report.
I'm on a 2009 MacBook Pro w/Intel Core 2 Duo, El Capitan 10.11.6.
Will be glad to do further testing, if that is useful.  Thanks!
[mu-editor_failed_report_20180520.txt](https://github.com/mu-editor/mu/files/2020567/mu-editor_failed_report_20180520.txt)

carlosperate, 2018-05-20:
Could you try this one as well? It has been built differently, so it would interesting to see if this one also breaks:  
https://github.com/carlosperate/python-portable-macos/releases/download/v0.0.1/Mu-Installer.dmg

darrell24015, 2018-05-21:
Yes, the second one launches - Mu 1.0.0.beta.15.  I selected the Python 3 mode and was successful with a simple "Hello World" program.  

PS - would like to have a little more control where it saves the program instead of the default mu_code folder. But I'm probably jumping ahead here. 😺

ZanderBrown, 2018-05-21:
@darrell24015 workspace_dir in settings, master now has an option to set it in the admin area I believe but support has been in Mu for a while now (i added it :-) )

ntoll, 2018-05-21:
@carlosperate I guess this could be something to do with the compiled version of Python built into the stand-alone Mu package..? Apparently briefcase allows us to define Python runtimes to embed (although this isn't documented anywhere, I guess we should look at the code).

carlosperate, 2018-05-21:
Yes, we can definitely try that.

I think it's called "support pkg" in briefcase: https://github.com/pybee/briefcase/blob/1d8035a6c8f6486b1c1582341a63471b67ffde26/briefcase/app.py#L385-L416

However, we should also let @freakboy3742 know, was I think it's meant to support OS X 10.8+: https://twitter.com/PyBeeWare/status/998261263359664128

ntoll, 2018-05-21:
Sure, I guess we should first confirm that it *is* actually a problem with the embedded Python runtime (from my naive reading of the crash report generated by OSX, that's what seems to be the case).

carlosperate, 2018-05-21:
Everything else comes from PyPi, so it should be mostly the same for both outputs. But yes, we should confirm by creating a briefcase app bundle with this python build: https://github.com/carlosperate/python-portable-macos/releases/download/v0.0.1/python3-only-reduced.zip

ntoll, 2018-05-21:
@carlosperate does that build include tkinter..? The Briefcase based one does not (so if your build has tkinter then we win twice...). :-)

carlosperate, 2018-05-21:
I didn't do anything special to include it, but the tkinter module is included in the lib/python3.6 directory. If I run a tkinter example in the Mu packaged in https://github.com/carlosperate/python-portable-macos/releases/download/v0.0.1/Mu-Installer.dmg it seems to work fine.

ntoll, 2018-05-21:
Awesomeness on stilts... so I guess we just need to figure out how to include this with Briefcase, right..?

carlosperate, 2018-05-21:
`python setup.py macos --support-pkg=<path to tarball>`, but we'll likely need to zip the python build into a tarball instead of a zip.
I'll update the travis script in https://github.com/carlosperate/python-portable-macos to do that as well.

ntoll, 2018-05-21:
Sure... in the same way that we store tkinter based assets for Windows in a mu-editor sub-repository, we can do the same for OSX builds of Python 3 and travis can grab the latest tgz from GitHub as a released downloadable blob (if that makes sense).

carlosperate, 2018-05-21:
Yep, sounds great. I can transfer https://github.com/carlosperate/python-portable-macos  to the mu-editor org, or create a new one with just a simple build script. Did you have any name in mind for the repo?

carlosperate, 2018-05-21:
We can try this one: https://transfer.sh/QwWgh/python3-only-reduced.tar.gz

ntoll, 2018-05-21:
WRT name... how about mu_portable_python_macos..?

freakboy3742, 2018-05-21:
@ntoll Thanks for letting me know about the 10.8 problem; finding a way to reproduce the issue on a my 10.12 box will be the issue... However, if you want to use your own support package, you can pass that in at the command line; and if you can find the source of the compatibility issue, I'm happy to integrate that into the "standard" support package.

paularmand, 2018-05-23:
https://github.com/carlosperate/python-portable-macos/releases/download/v0.0.1/Mu-Installer.dmg  worked on os x high sierra 10.13.2 . Trying to figure out how to create these dmg for our fork of the mu-editor with our custom skin. Thanks for the great work !

paularmand, 2018-05-23:
@carlosperate in travis.yml. is the section on s3 not giving away to many details?

carlosperate, 2018-05-25:
The key is encrypted using this method, so it should be safe: https://docs.travis-ci.com/user/encryption-keys/

But in terms of security we should always double check, so thanks @paularmand for the concern!

darrell24015, 2018-05-30:
Ready for another test under 10.11.6?

ntoll, 2018-06-13:
OK... it works (as in a working automatically built target using Carlos's custom build of Python, is created). See: https://s3-eu-west-2.amazonaws.com/mu-builds/osx/mu-editor_2018-06-13_13_24_package_macos_49a9c2f.dmg

THERE ARE BUGS (e.g. PGZero doesn't run [but I know why]). 

However, I'm going to land my branch into master and close this issue. If anyone finds bugs in the automatically generated OSX build, please create a new issue here so we get a clean separation of issues rather than putting everything into a mega-ticket. ;-)

Thank you to everyone for all your help with this work. Getting this to work is a major step forwards. Well done folks!

closed 2018-06-13 by ntoll