mu-editor/mu #373: install the newest version on windows 10

yonghuming, 2018-03-07:
If you are reporting a bug, we would like to know:

i want to install mu_2018-02-23_18_43_master_64db73d_64bit on windows 10,but after i finished installing and run mu, i got an error 
[image]
then in instaled qt use  ```pip3 install pyqt5```, but i doest not work
i could not run mu on windows 10

ntoll, 2018-03-07:
Hi @yonghuming 

Hmm... can you tell me which version of Windows you're using, along with the locale you have it set to. For example, Windows 10 64bit, Chinese.

Also, can you tell me the precise steps you followed to install Mu? There may be something I've not seen before that could be affecting things.

Finally, is there anything about your setup that may be different to "normal" Windows. For example, perhaps you have a virus checker or your laptop is managed by your work. Sometimes these interfere with user installed programs.

If I can recreate the problem I'll be able to find a way to fix it! In order to recreate the problem I need to know *exactly* what your setup is. :-)

carlosperate, 2018-03-07:
This has been an issue with collecting and locating Qt dlls on PyInstaller as well, and there is a major redesign on the PyQt hook to help with that (Qt minor releases had a tendency to break pyinstaller packaging in one way or another).

For easy search in this issue tracker, this is the displayed error message:
```
This application failed to start because it could not find or load the Qt platform plugin "windows"
```

ntoll, 2018-03-07:
Hi @yonghuming,

It appears a number of package version mismatches happened when creating the Windows installer. I've fixed these problems and tested both the 32bit (on Windows 7) and 64bit (on Windows 10) installers. They work for me!

Please try the latest build and let me know how you get on: http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=windows/

Also, there was a bug in the way Mu set up translation strings. I've just spent a fun evening working with a Chinese version of Windows 10 to make sure it works properly with Mu. As far as I can tell (and, remember, it's all in Chinese, which I don't understand) it works for me. You'll notice that some of the UI isn't translated. I believe you'll need to re-create the `messages.pot` file with the command `make translate`. I've attached it to this ticket if you don't have `make` installed on your machine. In any case, you should be able to compare what you already have translated with the new/missing strings to be translated (I've zipped it up since GitHub won't let me upload .pot files).

[messages.zip](https://github.com/mu-editor/mu/files/1791238/messages.zip)

I hope this makes sense..!

Many thanks for your support for Mu. It's most appreciated. :-)

yonghuming, 2018-03-08:
in some normal computer have not installed python before will not get this error

[image]
i unintalled all python version(python anaconda, pyhton 3.x x32 python 32 x64) and mu and reinstall mu editor, everything works.
i guess it mayb because python path or something.

yonghuming, 2018-03-08:
@carlosperate @ntoll thank you very much

carlosperate, 2018-03-08:
Oh! that's a really good point.

It's possible that PyQt looks for dlls in directories in a different order than what the pynsis installation expects, this is something that I think I've seen around the PyInstaller issues as well (I think there is a config file you can add somewhere in your python or PtQt directory where it might force PyQt to look first in the directories listed there).

Say a user has an old version of PyQt5 installed on a Windows computer, PyQt in Mu could be going through it's list of directories to find specific dlls, finds the global PyQt installation and gets confused when it doesn't find everything that it expects.

We should try to replicate this one, with anaconda and normal python. If that is what is happening then perhaps there should be a special rule/hook to be implemented in pynsist.

ntoll, 2018-03-08:
Hmmm... so this *is* interesting.

I wonder how PyQt does this sort of thing. As I understand it, the version of Python packaged with Mu is *only* allowed to look in certain directories for "stuff", thus making it run in a sort of isolated manner. I wonder if PyQt is adding a bunch of things to the PYTHON_PATH or there's an envar that Python is picking up (in which case, I believe it's possible to tell Python to run in "isolated" mode and ignore system settings).

The question is, do we want to do this..?

carlosperate, 2018-03-08:
Well, we can't really add "uninstall PyQt from your system" as a requirement/installation step for Mu, so something needs to be done if that is indeed a problem.

About Python "isolated" mode, I don't know if that would be enough for PyQt to stop looking for dlls in it's own list of directories, as a lot of that might be done by the compile Qt libs/dlls instead of Python.

I think first we should see if this is indeed a problem, and look into the qt.conf file, which might solve this for us anyway: http://doc.qt.io/qt-5/qt-conf.html

yonghuming, 2018-03-08:
mybe we should install virtualenv and run mu in virtualenv just like run on raspberrypi.....

ntoll, 2018-03-23:
Since I've seen @yonghuming using more recent versions of the installer based Mu, I'm assuming this has been fixed, so will close the issue.

closed 2018-03-23 by ntoll

carlosperate, 2018-03-23:
The issue has been fixed because the other python environments (including Anaconda) were uninstalled from the user system. This is possibly an issue with the current PyQt packaging, so if people that are running Mu with other Python installations, this could still be a problem.

While this is something that might not be fixed at moment, we should keep this in mind that this could still be a problem for other users. Having anaconda installed (which includes PyQt) is very common for Windows users looking for an easy way to use Python.

ntoll, 2018-03-24:
Sure... and the easy solution is to chat with the Anaconda folks so Mu can be installed via their mechanism. ;-)