mu-editor/mu #375: bug of plotter

yonghuming, 2018-03-08:
[image]
i want to track random data.
but the figure seems wrong, i use random()  to generate data, and the chart does not show the data in right form

ntoll, 2018-03-08:
You're right! Well done, you've found a bug. I think the numbers along the vertical axis should work as floating point values. It's obvious that the middle "0" is the real "0", but the "0"s above and below it should read 0.5 and -0.5. I'll see what I can do. :-)

Also, it's *so* good to see you using the Chinese Mu!

I have a question about "Mu" which I hope you can answer. Could "Mu" be represented in Chinese like the following example: https://en.wikipedia.org/wiki/Mu_(negative)?

ntoll, 2018-03-08:
Fixed by 981c909c3ca33f71a983fa4d1a50cbaace4fc36e

commit 981c909c (2018-03-08): Ensure the y-range on the plotter axis is meaningful and sensible. Fixes #375.
closed 2018-03-08 by ntoll