mu-editor/mu #376: Mu does not support Chinse cha

yonghuming, 2018-03-08:
If you are reporting a bug, we would like to know:

i want to print chinese char,
[image]
```
# -*- coding:utf-8 -*-
print("hello world!你好")
```
but i got an error
```
  File "c:\users\langxianmeng\mu_code\zh_cn_demo.py", line 2
SyntaxError: (unicode error) 'utf-8' codec can't decode byte 0xc4 in position 12: invalid continuation byte
>>> 
```

yonghuming, 2018-03-08:
it seems that mu saved file as ANSI format so do not suppot chinese even if i declare # -*- coding:utf-8 -*-

tjguk, 2018-03-08:
Thanks, @yonghuming this is being addressed in PR #370, which is in progress. Please feel free to comment over there. (And I'm very glad to have a non-Western tester as well!)

tjguk, 2018-03-08:
Duplicate of #370

yonghuming, 2018-03-08:
@tjguk think you i think we need to change code dealing with read/write stuffs

carlosperate, 2018-03-08:
Out of curiosity @yonghuming, could you create the same python file with notepad and attach it to this issue? (maybe the file will have to have a `.txt` extension https://help.github.com/articles/file-attachments-on-issues-and-pull-requests/).

ntoll, 2018-03-21:
This was fixed by #399

closed 2018-03-21 by ntoll