mu-editor/mu #377: Uninstalling Mu on Windows leaves files user files behind

carlosperate, 2018-03-08:
Steps to reproduce:
* Download https://s3-eu-west-2.amazonaws.com/mu-builds/windows/mu_2018-03-08_13_52_master_981c909_32bit.exe
* Install for all users (goes to Program Files)
* Uninstall
* Look into the user AppData local files, the log and settings files are still there

Does Pynsist support any way to add additional directories to the uninstall process?

ntoll, 2018-03-08:
Nope. :-(

Pynsist is a *simple* frontend to nsis. If we need more "advanced" features, we'll need to use nsis instead, which isn't something that fills me with joy. These files don't do any harm... right..?

carlosperate, 2018-03-17:
They don't do any harm, but if Mu creates these files then it should be responsible to delete them as well. That's just the nice thing to do. Any chance we can submit this as a feature request to Pynsist?

ntoll, 2018-03-20:
@carlosperate Sure, @takluyver is very responsive to contributions but is also anxious pynsist *doesn't* grow into a monster since it's supposed to be *simple* (if you want more control, use nsis [which pynsist wraps] instead).

Knock yourself out! ;-)

takluyver, 2018-03-20:
I generally reckon that uninstall shouldn't delete user data, so that you can uninstall and reinstall a program without losing data you have created. It's also tricky when the program itself is installed systemwide - you can't clear up user data for other user accounts which may have run it.

The good news is that it should be easy to do with a custom template - just extend the `uninstall_files` block:

https://github.com/takluyver/pynsist/blob/c772cfb829e2af1137a3c70efda68fa72631403c/nsist/pyapp.nsi#L164-L173

I'm not ruling it out, though - I'd consider it as a Pynsist feature.

ntoll, 2018-03-20:
Heh... well, I didn't want to say, but given that I'm forever installing and uninstalling versions of Mu for the purposes of testing, I quite like the fact that my Mu sessions and so on remain intact between installs. ;-) I also feel nervous about deleting user data (i.e. in contrast with the installed files).

carlosperate, 2018-03-21:
Fair enough.
As a user I still greatly dislike leaving files behind (not user created data, of course we shoudn't touch that, but autogenerated configuration files), if I uninstall an application I'd like all useless files to be removed, and if I reinstall the application I actually would prefer to have the defaults loaded rather than an old config. Some uninstallers do include a tick box to delete user data, but unfortunately not enough of them do.

That's only a personal point of view though, and to be fair this point from @takluyver does convince me to leave it as it is:
> It's also tricky when the program itself is installed systemwide - you can't clear up user data for other user accounts which may have run it.