mu-editor/mu #378: could not run Mu on mac

yonghuming, 2018-03-09:
i want to test Mu on mac,
i downloaded Mu, and run Mu installer and it exit immediately
i hope Mu will install an run success, 
but it could not run
Mac sierra

yonghuming, 2018-03-09:
```
Last login: Tue Mar  6 15:59:46 on ttys000
jusifangdeMacBook-Air:~ yeyimilk$ /Users/yeyimilk/Downloads/mu\ 4.app/Contents/MacOS/mu ; exit;
Logging to /Users/yeyimilk/Library/Logs/mu/mu.log
Traceback (most recent call last):
  File "run.py", line 6, in <module>
  File "mu/app.py", line 121, in run
  File "mu/logic.py", line 390, in setup
  File "shutil.py", line 235, in copy
  File "shutil.py", line 114, in copyfile
FileNotFoundError: [Errno 2] No such file or directory: '/var/folders/44/2c2jb6s91zz9fjpdkffzwzpw0000gn/T/_MEIkGHbRz/mu/resources/pygamezero/alien.png'
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
Deleting expired sessions...1 completed.

```
[image]

yonghuming, 2018-03-09:
could not install pgzero
```
jusifangdeMacBook-Air:Downloads yeyimilk$ pip install pgzero
Collecting pgzero
  Downloading pgzero-1.2.tar.gz (2.2MB)
    100% |████████████████████████████████| 2.2MB 166kB/s 
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/44/2c2jb6s91zz9fjpdkffzwzpw0000gn/T/pip-build-IaPMmb/pgzero/setup.py", line 7, in <module>
        with open(path, encoding='utf8') as f:
    TypeError: 'encoding' is an invalid keyword argument for this function
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/44/2c2jb6s91zz9fjpdkffzwzpw0000gn/T/pip-build-IaPMmb/pgzero/
jusifangdeMacBook-Air:Downloads yeyimilk$ 
```

yonghuming, 2018-03-09:
install mu on raspberry pi just use ```sudo apt install mu```on raspberry pi,
but i spend about 16 hours on it, what a ...... so sad...

tjguk, 2018-03-09:
re the pgzero install: Your pip is bound to python3. There might be a "pip3" command, or you could use "python3 -mpip" to install pygame zero. A recent Twitter thread clarified that pygame only has wheels (binary downloads) for python 3.6, so you might even need to upgrade Python, depending on what's installed.

yonghuming, 2018-03-09:
@tjguk i used pip3 but filed

carlosperate, 2018-03-09:
Hi @yonghuming, how did you download Mu? was it from http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=osx/ ? Right now the builds have not been updated to work on macOS, so the only guaranteed working app package would be for version 0.9.13 from https://codewith.mu/ or https://github.com/mu-editor/mu/releases

There is issue https://github.com/mu-editor/mu/issues/367 to track development for the new macOS builds.

tjguk, 2018-03-09:
@yonghuming for the pygame zero install, can I suggest you raise an issue over there --> https://github.com/lordmauve/pgzero

carlosperate, 2018-03-09:
pgzero is python3 only, I assume you are using the default python installation in macOS?
@yonghuming could you run `pip --version` and copy the result here?

If you install python 3.6 (important to be 3.6 in this case) you can install Mu from PyPi. Ideally using a virtualenvironment you can do:
```
pip3 install mu-editor
```
However, it's possible something might failed due to some dependencies (I've been switching a lot between different versions of Python3 and OS, so can't remember exactly which combination is broken). In that case you can download the Mu source code and install using `pip install .`, this way it should pull the latest versions of all dependencies, if that one doesn't work then that'd definitely a bug that we should fix.

yonghuming, 2018-03-11:
@carlosperate yes i downlaod from latest version. 
ok, i will try. Eager to install latest version on Mac
the releases for mac seems do not support python?

willingc, 2018-04-03:
You can install using `python3 -m pip install mu-editor` if using Python 3.6 or higher.

fmorton, 2018-04-05:
A complete guess on my part, but I experienced the same thing on the mac. It turned out that the "mount" command was not accessible, but instead had to be "/sbin/mount". I have submitted a pull request that fixes this, but to try this theory, add "/sbin" to your $PATH and see if all is ok. You could also test this theory by opening a terminal session and typing the "mount" command. If it is not found, this is the problem.

ntoll, 2018-06-04:
Falling back to  `/sbin/mount` has been implemented. Closing this issue.

closed 2018-06-04 by ntoll