mu-editor/mu #379: gettext support only implicitly imported by logic.py

tjguk, 2018-03-09:
As it stands, it is impossible to import logic.py on its own from mu, but it relies implicitly on app.py having been imported beforehand which calls gettext.install() which injects the _ into Python's globals.

I'm not actually sure how the tests were working, but I rather assume that one or more modules in the pytest stack do themselves import and install gettext.

I propose to move the gettext support into __init__.py within the mu package. In that way, it's present whenever a package module is imported but will only be called once (because of the module cache).

I've implemented this change as part of PR #371

ntoll, 2018-03-09:
Aha... I found the docs about gettext very confusing and so it's likely I've got things wrong. +1 to this change for the reasons you state.

As for the tests, the configuration that deals with this is found here: https://github.com/mu-editor/mu/blob/master/conftest.py

tjguk, 2018-03-09:
FWIW I don't think it's wrong as such: it's clearly worked so far! It's more that it's a bit of a code / structure smell, to have one module depend on another to the extent that you have to import in a particular order. I initially had a third module i18n, but then thought that the __init__.py solution was a neater solution.

tjguk, 2018-03-09:
BTW is there a reason why we're not using relative imports (eg from . import logic)?

ntoll, 2018-03-09:
Yes to fighting code smells. 

I'm trying to remember why the lack of relative imports. I think it was because when I started Mu I wanted it to be beginner friendly code and I liked the explicitness of `mu.logic` rather than `.` since it showed *where* the thing was being imported. That was just over two years ago and now that I think about it, while I still want Mu to be a beginner friendly code base, I also believe we should do "the right thing" and act as an example of "Pythonic" good practice. Thoughts..? (I'm happy to change.)

tjguk, 2018-03-09:
re relative imports; the usual problems are from trying to run something directly inside the package which uses relative imports. I might raise a separate issue and try it on a different branch to see how it plays out.

tjguk, 2018-03-17:
Fixed by #389

closed 2018-03-17 by tjguk