mu-editor/mu #380: Mu has no standard line-ending for programmatic use

tjguk, 2018-03-09:
Mu is designed to run on any platform which supports Python / Qt. This includes
Windows and any *nix variant, including OS/X. Windows traditionally uses \r\n
(ASCII 13 + ASCII 10) for line-endings while *nix usually recognises a single \n
(ASCII 10). Although many editors now detect and adapt to either convention,
it's common enough for beginners to use, eg, Windows notepad which only honours
and only generates the \r\n convention.

When reading / writing files, Python offers several forms of line-ending
manipulation via the newline= parameter in the built-in open() function.
Mu originally used Universal newlines (newline=None; the default), but then switched
to retaining newlines (newline="") in PR #133 

The effect of this last change is to retain whatever convention or mix of
conventions is present in the source file. In effect it is overriding any newline
manipulation to present to the editor control the characters originally present
in the file. When the file is saved, the same characters are written out.

However this creates a quandary when programmatically manipulating the editor
text: do we use the most widespread \n as a line-ending; or do we use the
platform convention `os.linesep`; or do we use the convention used in the
file itself, which may or may not follow the platform convention?

My proposal here is that Mu operate its own line-ending manipulation.

When reading the file, sniff the majority line-ending convention but
convert wholly to \n. When writing the file, use the convention noted
on the way in. 

This way the line-endings are honoured so that, eg, a file can be read/written
in Notepad without problems. And the Mu code can be sure of using
\n as a line-ending convention when manipulating the text.

In terms of the current implementation, the convention from the incoming
file could presumably be stored on the tab object.

An approach to this has been implemented as part of #371

ntoll, 2018-03-09:
Heh... this is wonderful to see and I completely agree with the outlook / modus operandi presented. :-)

carlosperate, 2018-03-12:
👍 +1 on this as well.

Just need to make sure that when new files are created on Windows we follow the local `\r\n` format on save as well.

tjguk, 2018-03-12:
@carlosperate that should happen with the current code (although I'll obviously add a test for this) since the default -- in the absence of anything else -- is os.linesep

carlosperate, 2018-03-12:
I think (and I might be wrong here, it's been a while since I looked at the code) that it currently happens because Qscintilla uses the OS default and the python `open` simply doesn't touch it. To manage line endings locally means turning this off to force Qscintilla to always use `\n`, so at that point I guess it would depend on whatever the tab object property default for line endings is, no?

tjguk, 2018-03-17:
Some work done under #390

ntoll, 2018-03-21:
This was fixed in #399

closed 2018-03-21 by ntoll