mu-editor/mu #381: Add make translate / translateall to make.py

tjguk, 2018-03-09:
More of a note to self.

Add the recent translate / translateall make targets to make.py. (The former relies on a posix command pipeline which we'll have to recreate in Python).

tjguk, 2018-03-16:
So I've got a branch to bring the make.py in line with the Makefile. However... could @ntoll or anyone tell me where their operating system or venv places pygettext? IOW what does `where pygettext` return? On Windows, it's in the slightly obscure "C:\Program Files\Python36\Tools\i18n\pygettext.py" which isn't on any PATH or in the venv at all. I can copy it there or attempt to detect its presence etc. but I wondered where it was for anyone else?

tjguk, 2018-03-16:
My bash-fu is not great. Would someone mind decoding the pipeline in the translate Make target, please, so I can translate it into the equivalent Python?

I *think* it's saying: excluding any file called _build or var plus anything in the ./docs or ./mu/contrib or./utils or ./mu/modes/api paths -- run pygettext across all the .py files.

find . \( -name _build -o -name var -o -path ./docs -o -path ./mu/contrib -o -path ./utils -o -path ./mu/modes/api \) -type d -prune -o -name '*.py' -print0 | $(XARGS) pygettext

tjguk, 2018-03-17:
cf #392

ntoll, 2018-03-17:
Yes, your reading for `translate` is correct. The intention being to only target those strings in Mu itself (so nothing from the third party code in `contrib` or utility code in `utils` or the auto-generated [and rather large] API docs in `mu.modes.api` namespace). If people want to translate the API docs, they can use `translate_all` for that.

My intention is for `translate` to be the *minimum* for localization of Mu, and `translate_all` is a pick-and-mix "nice to have" (i.e. I imagine people maybe translating the micro;bit based APIs but not the strings associated with the full Python standard library).

Viz: https://mu.readthedocs.io/en/latest/translations.html#create-a-new-translation

tjguk, 2018-03-17:
Fixed by #392

closed 2018-03-17 by tjguk