mu-editor/mu #383: windows mu-editor pip installation need a shortcut and should run like exe installer

yonghuming, 2018-03-11:
[image]

pip installation have not shortcut for users, and will be better if have one

after install with pip, we have to run mu with a command windows behind, feel not good.
any solutions when run mu.exe, and command teminal will exit and remain mu run just like windows installer installation works.
i try code below but not work
```
D:\coding\pgzero-cn\doc>start /b mu.exe

D:\coding\pgzero-cn\doc>Logging to C:\Users\langxianmeng\AppData\Local\python\mu\Logs\mu.log

D:\coding\pgzero-cn\doc>

D:\coding\pgzero-cn\doc>cd ../../mu

D:\coding\mu>python run.py
日志 TO C:\Users\langxianmeng\AppData\Local\python\mu\Logs\mu.log

D:\coding\mu>start /b python run.py

D:\coding\mu>日志 TO C:\Users\langxianmeng\AppData\Local\python\mu\Logs\mu.log
```
i sees this is related to logging code, so maybe should not logging to terminal.

ZanderBrown, 2018-03-11:
Seem to remember setuptools.py fails miserably with Windows shortcuts, definitely something to be looked into but I believe once it's available MSI will be the recommended install method for end users making this less of a problem

martinohanlon, 2018-03-12:
`pip` support for creating 'shortcuts' is virtually non-existent, you can mess around with post install scripts and all sorts of stuff, but these solutions wont work when packaged and distributed via pypi.

I have created a utility called `shortcut` which creates x platform shortcuts. http://shortcut.readthedocs.io/en/latest/

To create desktop and menu shortcuts to Mu, you would:

```
pip3 install shortcut
shortcut mu
```

There is no way (certainly that I have found) of getting this to run as part of the pip install process. You could include `shortcut` as an install_requires simplifying the process to:

```
pip3 install mu-editor
shortcut mu
```

ntoll, 2018-03-12:
Awesome work @martinohanlon.

NOTE TO SELF: this should be mentioned in the Mu docs.

carlosperate, 2018-03-12:
> There is no way (certainly that I have found) of getting this to run as part of the pip install process. 

To be honest, I am very happy about that. There a lot of packages in PyPi that do a lot of horrible things when building/installing a package (@bennuttall surely has some good stories from piwheels), and anything touching a directory oustide of the python or virtualenv environment should either be part of a proper installer or an extra step after pip, not part of a `pip install` process.

Having it on the docs seems like a good compromise to me.

tjguk, 2018-03-12:
What @carlosperate said

bennuttall, 2018-03-12:
> There a lot of packages in PyPi that do a lot of horrible things when building/installing a package (@bennuttall surely has some good stories from piwheels)

yuuup

+1 to this - pip install shouldn't try to do this, but that's not a problem because pip won't be the recommended way for regular users to install Mu, there'll be an exe or something.

ZanderBrown, 2018-03-12:
@martinohanlon does your magic support translations?

yonghuming, 2018-03-14:
thank you all of you

martinohanlon, 2018-03-14:
@zanderbrown what do you mean by translations?

ZanderBrown, 2018-03-14:
`.desktop` and I assume Window & Mac support translations for things like application comment, described in ["Localized values for keys"](https://specifications.freedesktop.org/desktop-entry-spec/latest/ar01s05.html). I was wondering if your tool could communicate with gettext and include that information?

Not a problem if no.

ntoll, 2018-03-16:
Fixed in cb01ae82f4336624c2983386b8923601da9f9888.

Thanks @martinohanlon for `shortcut`... ;-)

closed 2018-03-16 by ntoll