mu-editor/mu #384: maybe the APIs also need i18n

yonghuming, 2018-03-11:
[image]

Since APIs in ```mu/api``` dir also could be translated maybe need a machinic to i18n APIs
it will be more easy to use mu

ZanderBrown, 2018-03-11:
I'm confused what do you want translated?

yonghuming, 2018-03-14:
@ntoll yes it help

yonghuming, 2018-03-14:
@ntoll i tried make translateall on windows 10 under mu root dir, but it seems have no make translateall command
```
D:\coding\mu>python make.py translateall
Traceback (most recent call last):
  File "make.py", line 196, in main
    function = _exported[command]
KeyError: 'translateall'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "make.py", line 204, in <module>
    sys.exit(main(*sys.argv[1:]))
  File "make.py", line 198, in main
    raise RuntimeError("No such command: %s" % command)
RuntimeError: No such command: translateall

```

tjguk, 2018-03-14:
Sorry; it's on my to-do list. cf #381

tjguk, 2018-03-14:
Of course, anyone else is very welcome to submit a PR; I'm happy to review

ntoll, 2018-06-04:
Since translating the APIs would be require around 15751 new translation strings, and my guess is this is beyond the scope of a volunteer led project, I'm closing this quickly before anyone actually tries to do it and bite off more than they can chew..! :-)

Thank you for the suggestion.

closed 2018-06-04 by ntoll