mu-editor/mu #387: Check messages first character

carlosperate, 2018-03-13:
Not sure if this is a bug, but it looks like the first character on the check messages is not displayed correctly:
[image]

Tested on macOS 10.13 High Sierra, running from latest master.

tjguk, 2018-03-13:
Current Windows looks ok (assuming that the symbol is supposed to be a right-angled arrow)

[image]

ntoll, 2018-03-13:
Ah yes... I was given feedback that it wasn't obvious how the lines in the feedback related to particular lines of code, hence the use of the arrow pointing up. I guess we should do the easy thing and use a UTF-8 code-point that'll be in a font on *every* computer... ;-)

Open to suggestions! (Basically, we need some visual indication that the feedback relates to the line immediately above the box.)

ntoll, 2018-03-13:
Good point... (no pun intended)... :-D

carlosperate, 2018-03-13:
A caret would be a safe bet, but I must admit I'm not a fan. Works well in the terminal, but on a GUI is not very clear that it's mean to be pointing up:

[image]

I think a normal arrows is still a safe bet in mosts systems (↑), and looks much better:

[image]

ntoll, 2018-03-13:
+1 on the up arrow. Care to make the change..?

carlosperate, 2018-03-13:
Sure thing, I'll create the PR later tonight :)

closed 2018-03-20 by ntoll