mu-editor/mu #388: it will be intresting use microbit to control pygame zero

yonghuming, 2018-03-14:
maybe pygame zero mode should be a sublclass of microbit mode class.
or we need to create a new mode so we can use microbit to contorl pygame
i am trying.

ntoll, 2018-03-14:
Heh... so this is definitely possible. The thing to do would be to see how PyGame consumes USB serial messages (i.e. from devices like the micro:bit). I know for certain that PyGame supports various game controller / USB devices. It's something I've thought about too, and may use April's PyWeek (see: https://pyweek.org/) as an excuse to try to build something. In the worst case, PySerial could be an option to read and write data to the device.

tjguk, 2018-03-14:
Pygame's use of USB controllers is via the joystick module. I assume it would be easier to insert your own serial-handler into the pygame loop than to try to make the microbit mimic the joystick data

ukBaz, 2018-03-14:
FYI: There is work going on over on the micro:bit DAL to get the micro:bit to look like a Bluetooth Hardware Interface Device (HID).
https://github.com/lancaster-university/microbit-dal/pull/249

I know Bluetooth cannot be used by micropython on the micro:bit but to use the micro:bit as a wireless controller in Pygame this would probably make it easier to deal with.

ntoll, 2018-03-16:
@illume you're a :star: Thank you for posting this!

illume, 2018-03-23:
I hereby grant all, all permissions on those bytes to do as they wish. *ceremonial bow and handwave*

ntoll, 2018-03-24:
Stonking stuff... *ceremonial bow and handwave* back at'cha. ;-)

ntoll, 2018-06-04:
Closing since this is more to do with PyGame than Mu. :-) Thanks for all the input everybody.

closed 2018-06-04 by ntoll