mu-editor/mu #396: Cannot load Python files with a capitalised extension

carlosperate, 2018-03-17:
Tested on Windows 10, Python 3.5.2 32-bits, latest Mu master (6121350e5261df15cd7542fa938c2e0aab6b4fc2).

To reproduce:
* Create in the `mu_code` folder a file named `a.PY` with some text content
* Click the `Load` button
* Select and open the `a.PY` file
* The filed is not opened, no error message given, just a new empty tab opened:

[image]

Log:
```
Starting Mu 1.0.0.beta.15
2018-03-17 21:38:00,613 - root:95(run) INFO: uname_result(system='Windows', node='DESKTOP-KD60UUU', release='10', version='10.0.16299', machine='AMD64', processor='Intel64 Family 6 Model 60 Stepping 3, GenuineIntel')
2018-03-17 21:38:00,613 - root:96(run) INFO: Python path: ['C:\\workspace\\mu-temp\\mu', 'C:\\Users\\carlo\\Envs\\py3mu\\Scripts\\python35.zip', 'C:\\Users\\carlo\\Envs\\py3mu\\DLLs', 'C:\\Users\\carlo\\Envs\\py3mu\\lib', 'C:\\Users\\carlo\\Envs\\py3mu\\Scripts', 'C:\\Python35\\Lib', 'C:\\Python35\\DLLs', 'C:\\Users\\carlo\\Envs\\py3mu', 'C:\\Users\\carlo\\Envs\\py3mu\\lib\\site-packages', 'C:\\Users\\carlo\\Envs\\py3mu\\lib\\site-packages\\win32', 'C:\\Users\\carlo\\Envs\\py3mu\\lib\\site-packages\\win32\\lib', 'C:\\Users\\carlo\\Envs\\py3mu\\lib\\site-packages\\Pythonwin', 'C:\\Users\\carlo\\Envs\\py3mu\\lib\\site-packages\\IPython\\extensions']
2018-03-17 21:38:00,613 - mu.logic:499(__init__) INFO: Setting up editor.
2018-03-17 21:38:00,613 - mu.logic:510(__init__) INFO: Settings path: C:\Users\carlo\AppData\Local\python\mu\settings.json
2018-03-17 21:38:00,613 - mu.logic:511(__init__) INFO: Session path: C:\Users\carlo\AppData\Local\python\mu\session.json
2018-03-17 21:38:00,613 - mu.logic:512(__init__) INFO: Log directory: C:\Users\carlo\AppData\Local\python\mu\Logs
2018-03-17 21:38:00,613 - mu.logic:513(__init__) INFO: Data directory: C:\Users\carlo\AppData\Local\python\mu
2018-03-17 21:38:00,666 - mu.logic:521(setup) INFO: Available modes: debugger, python, microbit, pygamezero, adafruit
2018-03-17 21:38:00,713 - mu.logic:856(change_mode) INFO: Workspace directory: C:\Users\carlo\mu_code
2018-03-17 21:38:00,713 - mu.logic:560(restore_session) INFO: Restoring session from: C:\Users\carlo\AppData\Local\python\mu\session.json
2018-03-17 21:38:00,713 - mu.logic:561(restore_session) DEBUG: {'mode': 'python', 'paths': [], 'theme': 'day'}
2018-03-17 21:38:00,713 - mu.logic:580(restore_session) INFO: Loaded files.
2018-03-17 21:38:00,745 - mu.logic:589(restore_session) INFO: Starting with blank file.
2018-03-17 21:38:00,751 - mu.logic:856(change_mode) INFO: Workspace directory: C:\Users\carlo\mu_code
2018-03-17 21:38:08,624 - mu.interface.main:233(get_load_path) DEBUG: Getting load path: C:/Users/carlo/mu_code/a.PY
2018-03-17 21:38:08,624 - mu.logic:615(_load) INFO: Loading script from: C:/Users/carlo/mu_code/a.PY
2018-03-17 21:38:08,624 - mu.logic:643(_load) DEBUG: 
2018-03-17 21:38:10,644 - mu.logic:811(show_logs) INFO: Showing logs from C:\Users\carlo\AppData\Local\python\mu\Logs\mu.log

```

tjguk, 2018-03-17:
I'll pick this up for now, since I'm working in the load/save space for issue #370. At least, please can others leave this area alone for the moment?

carlosperate, 2018-03-17:
Great, thanks Tim!

ZanderBrown, 2018-03-18:
From my work on #74 (a side project to #385) i've found that Mu tries to load anything other than `.py` as a hex file.

https://github.com/mu-editor/mu/blob/6121350e5261df15cd7542fa938c2e0aab6b4fc2/mu/logic.py#L627-L639

When a hex file is opened the python source is extracted and pasted into a new untitled tab but clearly `.PY` isn't hex so it's failing with an empty document

**Side note:** *a non micropython hex will crash Mu*

So some verification that the file is actually a hex maybe a good idea, and case-insensitive matching of `.py`

carlosperate, 2018-03-18:
> Side note: a non micropython hex will crash Mu

If that the case we should probably open a new issue to track that one.

ZanderBrown, 2018-03-18:
Technically it would be a uFlash problem, in fact you already reported it https://github.com/ntoll/uflash/issues/29

carlosperate, 2018-03-19:
If uFlash is throwing an exception (intentionally or unintentionally) and Mu is crashing instead of catching it, we should still categorise that as a bug in Mu.

ntoll, 2018-03-20:
@ZanderBrown this should be a very simple fix (try/except/report problem, if encountered). I'm happy to fix this myself, but don't want to tread on @tjguk's toes as he does the encoding / load / save refactoring.

Good catch and +1 on @carlosperate's assessment (Mu should handle the problem, probably as per my description above).

ntoll, 2018-03-21:
The issues mentioned in this ticket have been fixed and the changes have landed in master.

closed 2018-03-21 by ntoll