mu-editor/mu #397: macOS tab close button has dissapeared?

carlosperate, 2018-03-17:
Not sure how this has happened, or if maybe this is only on my current environment. So hoping other macOS users can replicate.

Basically this is the way Mu currently looks for me. The tab doesn't show the close button any more:

[image]

There is a clear space on the left side of the tab name, and if you click there it will close the tab, but that's it.

Tested on the latest master at the time of writting: 6121350e5261df15cd7542fa938c2e0aab6b4fc2

carlosperate, 2018-03-17:
Looking at older screenshots from https://github.com/mu-editor/mu/issues/387 this was already an issue for a while:

[image]

ZanderBrown, 2018-03-18:
So clearly I didn't break it :smile:!

Looking at the screenshot it's trying to put the close button on the left, which would make some sense, is there any hover / click sensitivity there at all?

ZanderBrown, 2018-03-18:
[An upstream bug](https://bugreports.qt.io/browse/QTBUG-61092) Are you running against Qt 5.9.0 perhaps?

carlosperate, 2018-03-18:
Does that mean that changing the tab mode to "documet mode" would be a work around?
To be honest, after reading the Qt documentation I don't quite understand what that does.

ZanderBrown, 2018-03-18:
I'm not sure that it'll help with the close button (probably need newer/older Qt5 for that) but it should improve the behaviour of tabs, currently, they are working more like pages in a dialogue that documents on macos

From my non-mac owning or using perspective I believe that property will change the tabs alignment and populate the Window menu

ntoll, 2018-03-20:
Hmmm... so the problem persists in Qt5.10 on OSX (I just checked). I'll report upstream since they believe it fixed.

ntoll, 2018-03-20:
FYI: https://bugreports.qt.io/browse/QTBUG-61092?focusedCommentId=396123&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel#comment-396123

ntoll, 2018-03-20:
@ZanderBrown WRT `setDocumentMode(True)`, this does, indeed, make the tabs behave more like on Windows and Linux.

I've messed about with CSS to see we've broken something, but alas, it appears to be a regression in Qt5.10. :-(

ntoll, 2018-03-20:
OK... this is definitely a bug in Qt5.10. I just left a comment in their bug tracker which says this appears only when you modify things (not even the QTabBar) with custom CSS. If I delete the contents of the day theme css file, the icon magically appears for me on OSX with QT5.10.

I guess we need to wait for upstream to fix this. :-/

carlosperate, 2018-03-20:
I can confirm I was using PyQT 5.10.1 when I opened this issue, and PyQt 5.9.2 doesn't work either.

So https://bugreports.qt.io/browse/QTBUG-66213 is also the bug causing issues. 

If the next release takes long we can set PyQt to version 5.8.2 (and Qscintilla to v2.10) until the issue is resolved upstream. I can confirm this version does work:

[image]

ntoll, 2018-03-20:
Does PyQtChart still work in this configuration..?

carlosperate, 2018-03-20:
The option is not shown as a button, so I guess the answer is no.

I guess it needs Qt 5.10. From the REPL:
```
>>> from PyQt5 import QtChart
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: dlopen(/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6/site-packages/PyQt5/QtChart.so, 2): Library not loaded: @rpath/QtWidgets.framework/Versions/5/QtWidgets
  Referenced from: /Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6/site-packages/PyQt5/QtChart.so
  Reason: Incompatible library version: QtChart.so requires version 5.10.0 or later, but QtWidgets provides version 5.8.0

```

Looks like https://github.com/mu-editor/mu/blob/981c909c3ca33f71a983fa4d1a50cbaace4fc36e/mu/interface/panes.py#L40-L45 takes care processing that incompatibility, might be a good idea to log it as well, as I couldn't find anything in the generated log:

```
Starting Mu 1.0.0.beta.15
2018-03-20 14:54:41,751 - root:95(run) INFO: uname_result(system='Darwin', node='Carlos-MBP.local', release='17.4.0', version='Darwin Kernel Version 17.4.0: Sun Dec 17 09:19:54 PST 2017; root:xnu-4570.41.2~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2018-03-20 14:54:41,751 - root:96(run) INFO: Python path: ['/Users/microbit-carlos/workspace/mine/mu-upstream', '/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python36.zip', '/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6', '/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6/lib-dynload', '/usr/local/Cellar/python3/3.6.4_2/Frameworks/Python.framework/Versions/3.6/lib/python3.6', '/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6/site-packages', '/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6/site-packages/IPython/extensions']
2018-03-20 14:54:41,961 - mu.logic:499(__init__) INFO: Setting up editor.
2018-03-20 14:54:41,961 - mu.logic:510(__init__) INFO: Settings path: /Users/microbit-carlos/Library/Application Support/mu/settings.json
2018-03-20 14:54:41,961 - mu.logic:511(__init__) INFO: Session path: /Users/microbit-carlos/Library/Application Support/mu/session.json
2018-03-20 14:54:41,962 - mu.logic:512(__init__) INFO: Log directory: /Users/microbit-carlos/Library/Logs/mu
2018-03-20 14:54:41,962 - mu.logic:513(__init__) INFO: Data directory: /Users/microbit-carlos/Library/Application Support/mu
2018-03-20 14:54:41,990 - mu.logic:521(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-03-20 14:54:42,599 - mu.logic:856(change_mode) INFO: Workspace directory: /Users/microbit-carlos/mu_code
2018-03-20 14:54:42,599 - mu.logic:560(restore_session) INFO: Restoring session from: /Users/microbit-carlos/Library/Application Support/mu/session.json
2018-03-20 14:54:42,599 - mu.logic:561(restore_session) DEBUG: {'theme': 'day', 'mode': 'python', 'paths': []}
2018-03-20 14:54:42,599 - mu.logic:580(restore_session) INFO: Loaded files.
2018-03-20 14:54:42,643 - mu.logic:589(restore_session) INFO: Starting with blank file.
2018-03-20 14:54:42,653 - mu.logic:856(change_mode) INFO: Workspace directory: /Users/microbit-carlos/mu_code
2018-03-20 14:54:43,219 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-20 14:54:43,896 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-20 14:54:44,881 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-20 14:54:45,883 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-20 14:54:46,883 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-20 14:54:47,882 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-20 14:54:48,884 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-20 14:54:49,690 - mu.logic:822(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
2018-03-20 14:54:50,122 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-20 14:54:50,882 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-20 14:54:51,885 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-20 14:54:51,903 - mu.logic:856(change_mode) INFO: Workspace directory: /Users/microbit-carlos/mu_code
2018-03-20 14:54:52,882 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-20 14:54:53,056 - mu.logic:811(show_logs) INFO: Showing logs from /Users/microbit-carlos/Library/Logs/mu/mu.log
```

ntoll, 2018-03-20:
Good catch WRT logging the missing QCharts. I'll address that now.

WRT the missing close button... I guess we need to wait for upstream Qt to fix the bug.

carlosperate, 2018-03-20:
PyQtChart 5.8 does work though. So it would need PyQt 5.8.2,  PyQtChart 5.8, and Qscintilla to v2.10.

ntoll, 2018-03-20:
Aha... OK... perhaps these could be the basis of new minimum requirements for things like Debian and setup.py. Nice find! :-)

willingc, 2018-04-01:
FYI I replicated the issue. While doing so, I realize that the functionality of close still works in the area that the close element would be. The close element is not displayed, but you can close each tab.
On Mac 10.13.3.

[image]

ZanderBrown, 2018-04-01:
Do you know what Qt version your running against?
(Out of interest what happens with `setDocumentMode(True)`?)

ntoll, 2018-04-05:
This confirms the behaviour I was observing with latest Qt.

It appears to be a problem in Qt itself and there's not anything we can do about it until upstream fix this (see the links to related tickets in Qt in the comments above).

@ZanderBrown I've also been experimenting with `setDocumentMode(True)` on Mac and while this makes the tabs on OSX look the same as in Windows and Linux, it changes some of the CSS attributes (specifically the borders) and ends up looking worse (to my untrained eye).

I guess this is a case of "wait and see" for upstream.

ZanderBrown, 2018-04-05:
Unfortunately I don't really have access to a mac but I may investigate that further *if* I get a chance

Had a slightly left of field idea about a workaround for the close icon though