mu-editor/mu #400: codewith.mu out of date

martinohanlon, 2018-03-19:
codewith.mu installers are out of date.

I was going to make some changes (update links, add instructions, auto download javascript thingy) and push a PR but I couldn't find the repo.

ntoll, 2018-03-19:
Work on the new website is happening here: https://github.com/mu-editor/mu-editor.github.io/tree/v1 (note the v1 branch). I aim to fill in as much content as I can this week.

yonghuming, 2018-03-22:
i will help do some translation

ntoll, 2018-03-22:
@yonghuming please don't do anything yet! :-) This is a work in progress and I'll make an announcement when things are stable enough for translation efforts to start!

Thank you, as always, for your support! :-)

ntoll, 2018-03-23:
I'm going to close this since efforts on the website are ongoing and will land very soon.

@yonghuming I'll cc/ you in the appropriate issue for the website when it comes to translations etc... (and thank you for the offer). :-)

closed 2018-03-23 by ntoll