mu-editor/mu #401: print Chinese char in jupyter repl work well and could work when run a py with run button

yonghuming, 2018-03-22:
[image]

ntoll, 2018-03-22:
Are you using the latest version? I fixed this yesterday! :-)

ntoll, 2018-03-22:
See latest builds here: http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=windows/

tjguk, 2018-03-22:
Is it possible that, because we don't write the encoding cookie until we save (and then reload) that Jupyter's complaining? If so, we could (a) add to the new file; or (b) reload when saving so we get it in place

yonghuming, 2018-03-22:
@ntoll i use windows 10 64bit and when i print chinese it will exit immediately

ntoll, 2018-03-22:
Please confirm that you are using the *latest* version of Mu: https://s3-eu-west-2.amazonaws.com/mu-builds/windows/mu_2018-03-21_18_47_master_5ef9518_64bit.exe

yonghuming, 2018-03-22:
yes i am sure,

yonghuming, 2018-03-22:
# windows 10

ntoll, 2018-03-22:
OK... so I've managed to re-create a problem. The latest build of Mu fixes your original problem @yonghuming, however, I now see a new problem when running this on Windows (NOTE: it works fine on Linux and OSX -- I've just checked). See the attached screenie... @tjguk it looks like a Windows specific problem relating to `charmap`.  Any ideas..?

[image]

NB: I'm using the latest build (referenced in previous comment) on Windows 10 64bit, just like @yonghuming. The code I was running:

```
print("你好")
```

ntoll, 2018-03-22:
To recreate, copy the `print("你好")` code into a file and run it. Works fine on Linux/OSX.

yonghuming, 2018-03-22:
bot newest 32 and 64bit mu could not run

yonghuming, 2018-03-22:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
i have tried but notwork

    
        a#ntes-pcmail-signature-default:hover {
            text-decoration: underline;
            color: #199cff;
            cursor: pointer;
        }
        a#ntes-pcmail-signature-default:active {
            text-decoration: underline;
            color: #246fce;
            cursor: pointer;
        }
    

                
                    发自网易邮箱大师

tjguk, 2018-03-22:
I'll try to take a look later today unless someone gets there first.

ntoll, 2018-03-22:
I won't be able to look at this, so please knock yourself out @tjguk :-)

ntoll, 2018-04-13:
@tjguk any luck with this..?

tjguk, 2018-05-20:
just starting to look at this again. To help me and anyone else trying to reproduce, this code snippet will trigger the error:

`print("\u4f60\u597d")`

tjguk, 2018-06-06:
@yonghuming I believe that this was fixed when @ntoll merged #471. Could you try again with a dev build / git HEAD please?

ntoll, 2018-06-13:
I've checked this on my Windows10 machine and it appears to work.

closed 2018-06-13 by ntoll