mu-editor/mu #403: Adding a breakpoint in the debugger causes an exception

carlosperate, 2018-03-22:
Steps to reproduce:
* Select Python project
* Open file with a few lines (or create a new one)
* Click on "Debug"
* Add a breakpoint
* Mu crashes

```
Traceback (most recent call last):
  File "/Users/microbit-carlos/workspace/mine/mu-upstream/mu/logic.py", line 952, in debug_toggle_breakpoint
    self.modes['debugger'].toggle_breakpoint(line, tab)
  File "/Users/microbit-carlos/workspace/mine/mu-upstream/mu/modes/debugger.py", line 199, in toggle_breakpoint
    bps = self.debugger.breakpoints(tab.path)
  File "/Users/microbit-carlos/workspace/mine/mu-upstream/mu/debugger/client.py", line 244, in breakpoints
    return self.bp_index.get(filename, {})
AttributeError: 'Debugger' object has no attribute 'bp_index'
```

Last log:
```
Starting Mu 1.0.0.beta.15
2018-03-22 11:55:57,388 - root:95(run) INFO: uname_result(system='Darwin', node='Carlos-MBP.local', release='17.4.0', version='Darwin Kernel Version 17.4.0: Sun Dec 17 09:19:54 PST 2017; root:xnu-4570.41.2~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2018-03-22 11:55:57,388 - root:96(run) INFO: Python path: ['/Users/microbit-carlos/workspace/mine/mu-upstream', '/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python36.zip', '/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6', '/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6/lib-dynload', '/usr/local/Cellar/python3/3.6.4_2/Frameworks/Python.framework/Versions/3.6/lib/python3.6', '/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6/site-packages', '/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6/site-packages/IPython/extensions']
2018-03-22 11:55:57,666 - mu.logic:499(__init__) INFO: Setting up editor.
2018-03-22 11:55:57,667 - mu.logic:510(__init__) INFO: Settings path: /Users/microbit-carlos/Library/Application Support/mu/settings.json
2018-03-22 11:55:57,667 - mu.logic:511(__init__) INFO: Session path: /Users/microbit-carlos/Library/Application Support/mu/session.json
2018-03-22 11:55:57,667 - mu.logic:512(__init__) INFO: Log directory: /Users/microbit-carlos/Library/Logs/mu
2018-03-22 11:55:57,667 - mu.logic:513(__init__) INFO: Data directory: /Users/microbit-carlos/Library/Application Support/mu
2018-03-22 11:55:57,708 - mu.logic:521(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-03-22 11:55:58,408 - mu.logic:893(change_mode) INFO: Workspace directory: /Users/microbit-carlos/mu_code
2018-03-22 11:55:58,408 - mu.logic:560(restore_session) INFO: Restoring session from: /Users/microbit-carlos/Library/Application Support/mu/session.json
2018-03-22 11:55:58,408 - mu.logic:561(restore_session) DEBUG: {'theme': 'day', 'mode': 'python', 'paths': ['/Users/microbit-carlos/mu_code/test.py']}
2018-03-22 11:55:58,408 - mu.logic:615(_load) INFO: Loading script from: /Users/microbit-carlos/mu_code/test.py
2018-03-22 11:55:58,409 - mu.logic:677(_load) DEBUG: # -*- coding: utf-8-*-# Encoding cookie added by Mu Editor
# Write your code here :-)
print("line 2")
print("line 3")
print("line 4"")
print("line 5")
2018-03-22 11:55:58,465 - mu.logic:580(restore_session) INFO: Loaded files.
2018-03-22 11:55:58,480 - mu.logic:893(change_mode) INFO: Workspace directory: /Users/microbit-carlos/mu_code
2018-03-22 11:55:59,065 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:55:59,665 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:00,666 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:01,665 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:02,073 - mu.modes.python3:185(debug) INFO: Starting debug mode.
2018-03-22 11:56:02,090 - mu.logic:893(change_mode) INFO: Workspace directory: /Users/microbit-carlos/mu_code
2018-03-22 11:56:02,090 - mu.modes.debugger:112(start) DEBUG: # -*- coding: utf-8-*-# Encoding cookie added by Mu Editor
# Write your code here :-)
print("line 2")
print("line 3")
print("line 4"")
print("line 5")
2018-03-22 11:56:02,095 - mu.interface.panes:604(start_process) INFO: Running script: /Users/microbit-carlos/mu_code/test.py
2018-03-22 11:56:02,095 - mu.interface.panes:609(start_process) INFO: Command args: []
2018-03-22 11:56:02,096 - mu.interface.panes:615(start_process) INFO: Working directory: /Users/microbit-carlos/mu_code
2018-03-22 11:56:02,096 - mu.interface.panes:620(start_process) INFO: Python path: ['/Users/microbit-carlos/workspace/mine/mu-upstream', '/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python36.zip', '/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6', '/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6/lib-dynload', '/usr/local/Cellar/python3/3.6.4_2/Frameworks/Python.framework/Versions/3.6/lib/python3.6', '/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6/site-packages', '/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6/site-packages/IPython/extensions']
2018-03-22 11:56:02,669 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:03,665 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:04,665 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:05,665 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:06,665 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:07,666 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:08,668 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:09,668 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:10,665 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:11,747 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:12,341 - mu.debugger.client:195(on_fail) ERROR: Connection timed out. Is your machine slow or busy? Free up some of the machine's resources and try again.
2018-03-22 11:56:12,666 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:13,666 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
[... 7 lines cut ...]
2018-03-22 11:56:21,665 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:22,665 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:23,672 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:24,665 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:25,664 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:26,665 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:27,665 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:28,750 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:29,840 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:30,666 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:31,666 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:32,665 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:33,665 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:34,666 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:35,664 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
2018-03-22 11:56:35,830 - root:83(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Users/microbit-carlos/workspace/mine/mu-upstream/mu/logic.py", line 952, in debug_toggle_breakpoint
    self.modes['debugger'].toggle_breakpoint(line, tab)
  File "/Users/microbit-carlos/workspace/mine/mu-upstream/mu/modes/debugger.py", line 199, in toggle_breakpoint
    bps = self.debugger.breakpoints(tab.path)
  File "/Users/microbit-carlos/workspace/mine/mu-upstream/mu/debugger/client.py", line 244, in breakpoints
    return self.bp_index.get(filename, {})
AttributeError: 'Debugger' object has no attribute 'bp_index'
```

ntoll, 2018-03-22:
Yes... there had been some refactoring of the debugger in the early autumn of last year and this caused some stability issues. It's on my radar as the next thing to do once I've done the env var things for @bennuttall.

carlosperate, 2018-03-22:
Cool, feel free to close if this is a duplicate, I though this was a relatively new issue.

ntoll, 2018-06-19:
This is fixed in 07b160bedb9609a1f4d37a0022e3f283243d6dc8

closed 2018-06-19 by ntoll