mu-editor/mu #404: Mu is constantly "finding" connected devices

carlosperate, 2018-03-22:
The log get's completely flooded with the following entry, even when the python mode is preselected on start up:

```
2018-03-22 12:11:39,674 - mu.modes.base:127(find_device) INFO: Found device on port: cu.usbmodem14632
```

Looks like setting focus on the window (so cliking a different window, and then Mu) will always trigger the device, in this case a micro:bit, to be "found" again.

Tested on macOS, Python 3.6, latest master at the time of writting: 5ef9518c089a0c2038d5c8acb920d25c96a6eac7

yonghuming, 2018-03-22:
i met this problem

ntoll, 2018-03-22:
OK... that's a quick fix that'll land in master soon. If `find_device` is called as part of Mu's occasional polling for USB connections then logging is switched off. However, if the user attempts to use anything that requires a connection to the device (the REPL or plotter) then the call to `find_device` has logging enabled. In this way, we only log the information about the discovery of connected devices when the end user explicitly needs it. IYSWIM. :-)

ntoll, 2018-03-22:
Fixed in df8d20aa6ce011374497f5203d63473facac8e68

closed 2018-03-22 by ntoll

carlosperate, 2018-03-22:
That fixes the logging flood, but should Mu be constainly looking for devices? Specially if it's in a mode that doesn't use any of the devices?

ntoll, 2018-03-22:
Yes, this was a request of people who use micro:bit and CircuitPython. If a device is detected, Mu will tell you which mode to change to in order to use it (in the message area at bottom left).

ntoll, 2018-03-22:
Also, polling appears to be the only way to do this. :-( C'est la vie...

carlosperate, 2018-03-22:
Oh! 😮 I didn't realised that was a feature, pretty cool!