mu-editor/mu #407: Remove mouse over background colour change in the status bar?

carlosperate, 2018-03-23:
This is a very minor thing.
Mousing over the "mode" text in the status bar changes it background, making it look like a clickable UI element, this could be a bit confusing when it doesn't do anything.
[image]
The tooltip does help clarify what the text mean, which is definitely a good addition to keep there.

ZanderBrown, 2018-03-23:
When I added this they where clickable (hence I added them)

#406 removes them again as a side effect

carlosperate, 2018-03-23:
Do the tooltips still work with #406?

carlosperate, 2018-03-23:
Already fixed on master, thanks @ZanderBrown and @ntoll!

closed 2018-03-23 by carlosperate