mu-editor/mu #408: On macOS the new tab style has got white text on a lighter background

carlosperate, 2018-03-24:
On macOS using the latest master 9d82c9b201a43d3a2c9bbe41e7e730e6753154b4.

The new styling on the tabs makes it a bit more difficult to read the tab title.

Old style, dark backgroudn dark text (the active tab is the "untitled" on the right):
[image]

New style, light background light text (the active tab is the "untitled" on the right):
[image]

Same applies to the log/env tabs:
[image]

carlosperate, 2018-03-24:
Thanks @ntoll!

ntoll, 2018-03-24:
NP... It's CSS, so of course it never works as you quite expect on all platforms... ;-)

[image]