mu-editor/mu #409: Specifying a file on the command-line can duplicate if it's also in the restored session

tjguk, 2018-03-26:
If you specify a file on the command line (eg mu myfile.py) and that file is also stored in the session settings, it will be loaded twice.

ntoll, 2018-03-26:
good catch

ntoll, 2018-03-26:
Hmm... I've just tested this and everything works as expected on Linux (I even get told that `my_file.py` is already open). Could this be Windows specific..?

ZanderBrown, 2018-03-26:
Perhaps it's a problem of relative vs absolute paths?
https://github.com/mu-editor/mu/blob/40fbd75c46f5ae10c458fb478bf5b0af98d1416c/mu/logic.py#L647
As far as I can tell we never 'normalise' path to be absolute, if that's the case it would explain oddities like this

ntoll, 2018-03-26:
I thought of that... the `session.json` file stores the paths as absolute. I was starting Mu with a relative path. Having said that, I think you're right and I've just not found the right edge case. Given that logic.py's `restore_session` is dealing with absolute paths from the session.json file then if we normalise the `passed_filename` this problem shouldn't happen. This should be a quick fix in app.py.

ntoll, 2018-03-26:
@tjguk doe 297669a9cbebdeb8991232995d524e6b80cf47e2 fix your problem?

tjguk, 2018-03-26:
Haven't tried yet, but since I was able to reproduce without even having a file on the command line, here's the contents of my session file:

```
{
  "theme": "day",
  "mode": "python",
  "paths": [
    "C:/Users/goldent/mu_code/t.py",
    "C:\\Users\\goldent\\mu_code\\t.py"
  ]
}
```

ntoll, 2018-03-26:
hah... it looks like Window paths strike again. What would be really useful is to understand how each of the paths was created. I'm assuming the second path is the absolute path of the file you specified on the command line? In which case, something (QWidget) is storing paths in a way that's more like a URL rather than an OS specific filesystem path. We should choose one (and my vote would be for the URL like convention -- in which case, how do I replace abspath with the right thing?).

tjguk, 2018-03-26:
os.abspath (which calls os.normpath) on Windows will produce a path of the form c:\users\goldent\... from either of the two forms above.

tjguk, 2018-03-26:
With your patch applied, I see the following behaviour:

* settings.json contains "C:/Users/goldent/mu_code/t.py"
* python -mmu t.py
* t.py is opened successfully in the editor pane, but an alert appears "The file t.py is already open"

ntoll, 2018-03-26:
Aha... so that's the expected behaviour. Is that good enough for you to say it's fixed..?

tjguk, 2018-03-26:
I don't feel strongly about this but I would find this behaviour surprising, principally because the session file and its mechanisms are invisible to me. From my (the user's) perspective, I specified a file to open. And it opened. But then it told me it was already open!

That this is an artefact of an automatic background mechanism is invisible to me and --effectively -- outside my control.

ntoll, 2018-03-26:
Hmm... I'll look into this further. I think it's perhaps a bit clunky for the reasons you state. A quick look suggests that it's QScintilla's `.path` that's giving us paths like "C:/Users/goldent/mu_code/t.py" on Windows (which, to my mind should look like "C:\\Users\\goldent\\mu_code\\t.py"). I don't have a Windows machine available right now (I'm in NY with my Linux laptop) to fix this.

ntoll, 2018-03-26:
OK... I have a VirtualBox! :-) I'm on it.

ntoll, 2018-03-26:
Aha... @tjguk f1b1d6615226766d14ab945e63e24fa798da7b11 should fix this for you. AFAICT from experiments in the Python REPL on a Windows 7 VirtualBox `os.path.abspath` should fix things as per that commit. Please confirm that the alert no longer appears (and if so, please feel free to close this issue).

ZanderBrown, 2018-03-26:
So from what @tjguk is saying we should only show the 'already open' dialogue when opening from the file picker, otherwise OS/CLI/Drop should just switch to the relevant tab when it's already loaded

ZanderBrown, 2018-03-27:
Just pulled the latest commit and `It's Dead, Jim`
```bash
$ ./run.py 
Logging to /home/zbrown/.cache/mu/log/mu.log
Traceback (most recent call last):
  File "./run.py", line 6, in <module>
    run()
  File "/home/zbrown/Projects/mu/mu/app.py", line 109, in run
    editor.restore_session(os.path.abspath(passed_filename))
  File "/usr/lib64/python3.6/posixpath.py", line 369, in abspath
    path = os.fspath(path)
TypeError: expected str, bytes or os.PathLike object, not NoneType
```

tjguk, 2018-03-27:
Still fails when the (same) paths differ by case only. We really need a "normalise_path" function somewhere which:

* abspath
* normcase

 and anything else we need. The present solution treats case-different paths as different (which they aren't, on Windows). Sorry to have opened such a can of worms here.

The good news is that it works fine for chinese-character filenames (even when they don't display correctly in the Window console!). FWIW the most likely use-case for this isn't the command-line but drag-drop files onto the Mu icon itself.

ZanderBrown, 2018-03-27:
@tjguk just to clarify is that with @ntoll 's master or my branch?

tjguk, 2018-03-27:
@ZanderBrown sorry; current master. I haven't had the bandwidth to look at your branch yet; sorry.

ntoll, 2018-03-28:
Fixed by #411

closed 2018-03-28 by ntoll