mu-editor/mu #410: Can't double-click on mode

tjguk, 2018-03-26:
When presented with the list of modes, one expects to be able to double-click in lieu of click-and-ok

ZanderBrown, 2018-03-26:
Looks like it's the `itemDoubleClicked` signal we want, away from my machine for a while but it looks like we want to connect to that around line 66 and handle it by emiting `self.accept`

https://github.com/mu-editor/mu/blob/40fbd75c46f5ae10c458fb478bf5b0af98d1416c/mu/interface/dialogs.py#L46-L91

i.e.

```python
self.setLayout(widget_layout)
self.mode_list = QListWidget()

@self.mode_list.itemDoubleClicked.connect
def select_and_accept():
    self.accept.emit()

widget_layout.addWidget(self.mode_list)
self.mode_list.setIconSize(QSize(48, 48))
```

carlosperate, 2018-03-26:
I agree with this one, I keep double clicking the mode every time I need to change it, so this additional would definitely be welcomed.

ntoll, 2018-03-26:
OK... that didn't take long... ;-)

commit 4ffe4b67 (2018-03-26): Allow mode selection to respond to double-clicks. Fixes #410.
closed 2018-03-26 by ntoll