mu-editor/mu #414: Application id?

ZanderBrown, 2018-03-28:
I discovered the other day we've been [packaged for Fedora](https://apps.fedoraproject.org/packages/mu) and of course, we have Debian packaging stuff in the repo already

I know we don't currently need one as such and we decided against putting any effort into #318 but it could be good if we define an application id and name our .desktop file accordingly (and macOS equivalents)

Personally I think `mu.codewith.editor` seems an obvious choice but what do people think

(@ntoll , @carlosperate , @bennuttall  raw thoughts?)

ntoll, 2018-03-28:
Just about to step onto a train, but is the naming convention is the reverse of a URL, then that looks fine. ;-)

ZanderBrown, 2018-03-28:
Yes reverse DNS is the normal system

bennuttall, 2018-03-29:
SGTM :+1:

ZanderBrown, 2018-04-13:
I have a branch `base/desktop-integration` where I'd started to work with this

The changes being:
* Rename `.desktop`
* Rename icon installed for `.desktop`
* Use setDesktopFileName on the QApplication object

My branch also started some appdata niceties for Linux packaging

ntoll, 2018-05-03:
Fixed by #440. Thanks @ZanderBrown :-)

closed 2018-05-03 by ntoll