mu-editor/mu #415: beta 15 - no plotter…

Pskalou, 2018-03-29:
Hello, I want to test the new plotter issue, but I haven't the plotter icon on my menus. 
Also when i click on the "Acces the file system on the micro:bit", mu stop and exit immediately.

Here is a copy of my log file 

> 
> 2018-03-29 08:38:42,872 - root:95(run) INFO: 
> 
> -----------------
> 
> Starting Mu 1.0.0.beta.15
> 2018-03-29 08:38:42,872 - root:96(run) INFO: uname_result(system='Linux', node='xxxxxxxxxx', release='4.15.0-2-amd64', version='#1 SMP Debian 4.15.11-1 (2018-03-20)', machine='x86_64', processor='')
> 2018-03-29 08:38:42,872 - root:97(run) INFO: Python path: ['/home/xxxxxxxxxx/Logiciels/mu', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/home/xxxxxxxxxx/.local/lib/python3.6/site-packages', '/usr/local/lib/python3.6/dist-packages', '/usr/lib/python3/dist-packages', '/usr/local/lib/python3.6/dist-packages/IPython/extensions']
> 2018-03-29 08:38:42,945 - mu.logic:512(__init__) INFO: Setting up editor.
> 2018-03-29 08:38:42,945 - mu.logic:524(__init__) INFO: Settings path: /home/xxxxxxxxxx/.local/share/mu/settings.json
> 2018-03-29 08:38:42,945 - mu.logic:525(__init__) INFO: Session path: /home/xxxxxxxxxx/.local/share/mu/session.json
> 2018-03-29 08:38:42,945 - mu.logic:526(__init__) INFO: Log directory: /home/xxxxxxxxxx/.cache/mu/log
> 2018-03-29 08:38:42,945 - mu.logic:527(__init__) INFO: Data directory: /home/xxxxxxxxxx/.local/share/mu
> 2018-03-29 08:38:42,963 - mu.logic:540(setup) INFO: Available modes: python, adafruit, microbit, debugger
> 2018-03-29 08:38:43,025 - mu.logic:967(change_mode) INFO: Workspace directory: /home/xxxxxxxxxx/mu_code
> 2018-03-29 08:38:43,026 - mu.logic:581(restore_session) INFO: Restoring session from: /home/xxxxxxxxxx/.local/share/mu/session.json
> 2018-03-29 08:38:43,026 - mu.logic:582(restore_session) DEBUG: {'theme': 'day', 'mode': 'microbit', 'paths': ['/home/xxxxxxxxxx/mu_code/test.py'], 'envars': []}
> 2018-03-29 08:38:43,026 - mu.logic:650(_load) INFO: Loading script from: /home/xxxxxxxxxx/mu_code/test.py
> 2018-03-29 08:38:43,027 - mu.logic:712(_load) DEBUG: # -*- coding: utf-8-*-# Encoding cookie added by Mu Editor
> Write your code here :-)
> from microbit import *
> 
> flag = True
> 
> while True:
>     
>     if button_a.was_pressed():
>         flag = not flag
>         
>     if flag:
>         val = accelerometer.get_values()
>         print(val)
>     sleep(100)
> 2018-03-29 08:38:43,132 - mu.logic:603(restore_session) INFO: Loaded files.
> 2018-03-29 08:38:43,132 - mu.logic:607(restore_session) INFO: User defined environment variables: []
> 2018-03-29 08:38:43,153 - mu.logic:967(change_mode) INFO: Workspace directory: /home/xxxxxxxxxx/mu_code
> 2018-03-29 08:38:48,804 - mu.modes.base:128(find_device) INFO: Found device on port: ttyACM0
> 2018-03-29 08:38:48,808 - mu.modes.base:177(add_repl) INFO: Started REPL on port: /dev/ttyACM0
> 2018-03-29 08:38:48,808 - mu.modes.base:159(toggle_repl) INFO: Toggle REPL on.
> 2018-03-29 08:38:58,021 - mu.modes.base:156(toggle_repl) INFO: Toggle REPL off.
> 2018-03-29 08:38:59,159 - mu.modes.microbit:376(toggle_files) INFO: Toggle filesystem on.
> 2018-03-29 08:38:59,159 - root:84(excepthook) ERROR: Unrecoverable error
> Traceback (most recent call last):
>   File "/home/xxxxxxxxxx/Logiciels/mu/mu/modes/microbit.py", line 378, in toggle_files
>     self.view.button_bar.slots['plotter'].setEnabled(False)
> KeyError: 'plotter'

thanks for your work. I hope it will be very usefull in classrooms
PsKalou

ntoll, 2018-03-29:
Hi @Pskalou,

Sorry you've been having problems. However, well done you, you've found a bug! ;-)

Out of interest, what version of PyQt do you have installed..? Do you also have PyQCharts installed too..?

The best way to install Mu is by creating a virtualenv and pip installing it in there. Python should download all the necessary things you need. If Mu can't detect the presence of QtCharts it won't display the button for the plotter (thus explaining the first part of your report), and when you click on the "Files" menu, it attempts to turn off the (non-existent) plotter button, thus causing the crash (and hence the kudos to you for finding the bug!).

Fixing this will be very easy and it'll definitely be in the next release of Mu. In the meantime, if you use a virtualenv and pip (`pip install mu-editor`) then you should get all the required libraries needed for the plotter.

Hope this helps,

N.

Pskalou, 2018-03-29:
Hello @ntoll. 

> pip install mu-editor

solved immediately the problem. 
Your command installed on my computer these libraries : 
mu-editor, pgzero==1.1, matplotlib==2.1.2, pyqt5==5.10, PyQtChart==5.10, qscintilla==2.10.2, pycodestyle==2.3.1 and pygame>=1.9.

Well done to you, and thanks a lot,
PsKalou

ntoll, 2018-03-29:
I'll leave this issue open until I've fixed the bug with the "files" button. :-) It's a good way to track what's going on. Thanks!

ntoll, 2018-03-30:
Fixed by  ddaddce

closed 2018-03-30 by ntoll