mu-editor/mu #416: On Windows os.path.abspath doesn't fail on None

tjguk, 2018-03-29:
The newly-added test test_abspath_fail is relying on a value of None to cause a failure. Presumably on Posix platforms it does. But on Windows, it returns the current directory! I believe that Windows will successfully fail when given a non-prefixed path containing a wildcard character (*, ?) so I'll add those to the list so at least something fails!

ZanderBrown, 2018-03-29:
Looking at `/Lib/posixpath.py` and  `/Lib/ntpath.py` the Windows implementation seems far more... *complex*

I'm inclined to say `ntpath` mapping `None` to cwd is bug (clearly it's inconsistent with `posixpath` and I assume `macpath`) but obviously that doesn't help us all the way here downstream

ZanderBrown, 2018-03-29:
To help your debugging this is what happens with `posixpath`
```python
Python 3.6.4 (default, Mar 13 2018, 18:18:20) 
[GCC 7.3.1 20180303 (Red Hat 7.3.1-5)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import os.path
>>> os.path.abspath(None)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib64/python3.6/posixpath.py", line 369, in abspath
    path = os.fspath(path)
TypeError: expected str, bytes or os.PathLike object, not NoneType
>>> 
```

ZanderBrown, 2018-03-29:
So ntpath does in fact [have a fallback](https://github.com/python/cpython/blob/master/Lib/ntpath.py#L517-L530) to cwd that [doesn't exist](https://github.com/python/cpython/blob/master/Lib/posixpath.py#L369-L378) in [any other](https://github.com/python/cpython/blob/master/Lib/macpath.py#L186-L194) path implementation (at least > 3.6)
In fact it's documented as:
> Return a normalized absolutized version of the pathname path. On most platforms, this is equivalent to calling the function normpath() as follows: `normpath(join(os.getcwd(), path))`.

But calling `os.path.join` with `None` is itself an error so this behaviour would seem to contradict the documentation

This, of course, brings us to the question of does any of us have experience reporting CPython bugs/undocumented behaviour

tjguk, 2018-03-30:
Thanks for looking into this @ZanderBrown . Leaving aside for now the question of Python's internal inconsistency, I can easily create the same effect by calling os.path.abspath(5), for example. But thinking that through, I suggest that we shouldn't in fact be catching Exception but only OSError. If Editor._abspath ever received something which isn't a string, it should be raising an error -- or letting it be raised.

So... this bring us to: what errors did you expect Editor._abspath to log and ignore? os.path.abspath specifically doesn't check whether the filename exists or is valid. So what *is* it expecting to ignore?

ZanderBrown, 2018-03-30:
The catch in `_abspath` is intended to ignore nonsensical paths (such as `None`) on the assumption they only occur when Mu has been called with nonsense as arguments because the OS won't let you open a file that can't exist so to speak (If it's a bad path it can't be there to double click)

Your right a bad path should be `OSError` and a non-"path like" argument should be `TypeError`, specifying `Exception` is largly just laziness on my part but should be a catch all

ntoll, 2018-03-30:
I'll try looking into this / fixing on my flight back from the US later today! ;-)

ntoll, 2018-03-30:
OK... so I'm at the airport ready to fly home so spending some time thinking about this.

What we definitely don't want is for Mu to crash because of an uncaught exception. We do want the error to be visible in some way to the user and we want Mu to attempt to process as many of the paths it can. 

To this end, the code in this function won't cause Mu to crash, if an error is encountered it is both logged along with the exception AND the failure of a single path doesn't stop Mu from successfully processing any other paths.

So, in terms of the implementation, I'm happy that it works, despite the generous use of `Exception` to catch all.

I guess what remains is to make the test work with Windows, in which case I'll use @ZanderBrown's inspirationally bonkers `lambda: None`. ;-)

Awesome work on the investigation @tjguk..!

ntoll, 2018-03-30:
See  2c5b70a - please confirm this works on Windows.

closed 2018-03-30 by ntoll

tjguk, 2018-03-31:
For info, this is an existing (unclosed) report of the abspath/None difference already on bugs.python.org:

https://bugs.python.org/issue22587

It was left unresolved (4 years ago). If you thought it worth pursuing, we could revive it, but any patch wouldn't land until 3.8 as 3.7 has recently entered feature-freeze.