mu-editor/mu #420: same code work in vscode but not mu

yonghuming, 2018-04-02:
i wrote a simple program using pygame:
```
import pygame

pygame.init()

screen = pygame.display.set_mode([800, 600])

keep_going = True
pic = pygame.image.load("baoer.jpg")
posx = 0
posy = 0

timer = pygame.time.Clock()
speedx = 4
speedy = 4

while keep_going:
    for evt in pygame.event.get():
        if evt.type == pygame.QUIT:
            keep_going = False
            print(evt)
    # screen.fill((0, 0, 0))
    screen.blit(pic, (posx, posy))
    posx += speedx / 2
    posy += speedy / 3

    if pic.get_width() + posx > 800 or posx < 0:
        speedx = -speedx
    if pic.get_height() + posy > 600 or posy < 0:
        speedy = -speedy

    
    pygame.display.update()
    timer.tick(60)

pygame.quit()
```
it loads a image called ```baoer.jpg``` in the same dir, works well in vscode, but could not work in mu and it told me pygame could not load images ```baoer.jpg```, 

why?

tjguk, 2018-04-02:
Can you stick a os.getcwd() call just before the pygame.image.load, please? That should show us which directory is current at the time.

ZanderBrown, 2018-04-02:
Yes I suspect @tjguk is right and its cwd that's catching you out, something else worth noting is that the vscode python extension has an awful habit of defaulting to Python2

ntoll, 2018-04-05:
Hi Folks,

What @tjguk says. PyGameZero does the "right thing" (tm) when importing images, but that's based on PyGameZero conventions. With PyGame you'll need to ensure you are running the code with the correct / expected current working directory.

yonghuming, 2018-04-13:
oh sorry i have pased wrong code

i send same code below to a lot of student on different places, the use windows 10 or window 7, but they both occoured problems that do nothing after press play button

```
2018-04-13 09:54:08,911 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.0.beta.15
2018-04-13 09:54:08,915 - root:113(run) INFO: uname_result(system='Windows', node='LAPTOP-SSO2P9NF', release='10', version='10.0.16299', machine='AMD64', processor='Intel64 Family 6 Model 78 Stepping 3, GenuineIntel')
2018-04-13 09:54:08,915 - root:114(run) INFO: Python path: ['C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\Python', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2018-04-13 09:54:09,043 - mu.logic:356(__init__) INFO: Setting up editor.
2018-04-13 09:54:09,043 - mu.logic:367(__init__) INFO: Settings path: C:\Users\yelh\AppData\Local\python\mu\settings.json
2018-04-13 09:54:09,044 - mu.logic:368(__init__) INFO: Log directory: C:\Users\yelh\AppData\Local\python\mu\Logs
2018-04-13 09:54:09,044 - mu.logic:369(__init__) INFO: Data directory: C:\Users\yelh\AppData\Local\python\mu
2018-04-13 09:54:09,088 - mu.logic:377(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-04-13 09:54:10,281 - mu.logic:713(change_mode) INFO: Workspace directory: C:\Users\yelh\mu_code
2018-04-13 09:54:10,282 - mu.logic:416(restore_session) INFO: Restoring session from: C:\Users\yelh\AppData\Local\python\mu\settings.json
2018-04-13 09:54:10,282 - mu.logic:417(restore_session) DEBUG: {'theme': 'day', 'mode': 'pygamezero', 'paths': ['C:/Users/yelh/mu_code/colorConfusion.py'], 'workspace': 'C:\\Users\\yelh\\mu_code', 'microbit_runtime_hex': None}
2018-04-13 09:54:10,282 - mu.logic:471(_load) INFO: Loading script from: C:/Users/yelh/mu_code/colorConfusion.py
2018-04-13 09:54:10,282 - mu.logic:499(_load) DEBUG: # -*- coding: utf-8-*-# Encoding cookie added by Mu Editor

from random import choice

colors = ['red', 'blue', 'green']

meaning_text = choice(colors)

meaning_color = choice(colors)

circle_color = choice(colors)

def reset_color():

    global colors, meaning_color, meaning_text, circle_color

    meaning_text = choice(colors)

    meaning_color = choice(colors)

    circle_color = choice(colors)

score = 0

WIDTH = 300

HEIGHT = 490

def draw():

    global meaning_color, meaning_text, circle_color, score

    screen.clear()

    screen.blit('bg', (0, 0))

    screen.draw.text('Score:', (100, 100), color = 'black', fontsize = 32)

    screen.draw.text(str(score), (170, 100), color = 'black', fontsize = 32)

    screen.draw.text(meaning_text, (120, 160), color = meaning_color, fontsize = 32)

    

    screen.draw.filled_circle((150, 260), 16, circle_color)

    

def on_mouse_down(pos):
[... 116 lines cut ...]

        if x > 150:

            score = score + 1

        else:

            score = score - 1

    #===========================================================

    reset_color()

    

    

2018-04-13 09:54:18,969 - mu.interface.panes:602(start_process) INFO: Running script: c:\users\yelh\mu_code\colorconfusion.py
2018-04-13 09:54:18,969 - mu.interface.panes:607(start_process) INFO: Command args: []
2018-04-13 09:54:18,970 - mu.interface.panes:613(start_process) INFO: Working directory: C:\Users\yelh\mu_code
2018-04-13 09:54:18,970 - mu.interface.panes:618(start_process) INFO: Python path: ['C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\Python', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2018-04-13 09:56:11,278 - mu.logic:680(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
2018-04-13 09:56:18,946 - mu.logic:669(show_logs) INFO: Showing logs from C:\Users\yelh\AppData\Local\python\mu\Logs\mu.log

```

yonghuming, 2018-04-13:
after i add ```os.getcwd()```
the log
```
  

2018-04-13 10:00:57,710 - mu.interface.panes:602(start_process) INFO: Running script: c:\users\yelh\mu_code\colorconfusion.py
2018-04-13 10:00:57,710 - mu.interface.panes:607(start_process) INFO: Command args: []
2018-04-13 10:00:57,711 - mu.interface.panes:613(start_process) INFO: Working directory: C:\Users\yelh\mu_code
2018-04-13 10:00:57,711 - mu.interface.panes:618(start_process) INFO: Python path: ['C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\Python', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2018-04-13 10:01:03,187 - mu.modes.pygamezero:129(stop_game) DEBUG: Stopping script.
2018-04-13 10:01:05,045 - mu.modes.pygamezero:118(run_game) DEBUG: # -*- coding: utf-8-*-# Encoding cookie added by Mu Editor

from random import choice

import os

print(os.getcwd())

colors = ['red', 'blue', 'green']

meaning_text = choice(colors)

meaning_color = choice(colors)

circle_color = choice(colors)

def reset_color():

    global colors, meaning_color, meaning_text, circle_color

    meaning_text = choice(colors)

    meaning_color = choice(colors)

    circle_color = choice(colors)

score = 0

WIDTH = 300

HEIGHT = 490

def draw():

    global meaning_color, meaning_text, circle_color, score

    screen.clear()

    screen.blit('bg', (0, 0))

    screen.draw.text('Score:', (100, 100), color = 'black', fontsize = 32)

    screen.draw.text(str(score), (170, 100), color = 'black', fontsize = 32)

    screen.draw.text(meaning_text, (120, 160), color = meaning_color, fontsize = 32)

    

    screen.draw.filled_circle((150, 260), 16, circle_color)

    

def on_mouse_down(pos):

    print(pos)

    x, y = pos

    global meaning_text, circle_color, score

    

    #Type your code below to complete the color confusion game!

    #==========================================================

    if meaning_text == circle_color:

        

        if x < 150:

            score = score - 1

        else:

            score = score + 1

    else:

        if x > 150:

            score = score + 1

        else:

            score = score - 1

    #===========================================================

    reset_color()

    

    

2018-04-13 10:01:05,048 - mu.interface.panes:602(start_process) INFO: Running script: c:\users\yelh\mu_code\colorconfusion.py
2018-04-13 10:01:05,048 - mu.interface.panes:607(start_process) INFO: Command args: []
2018-04-13 10:01:05,048 - mu.interface.panes:613(start_process) INFO: Working directory: C:\Users\yelh\mu_code
2018-04-13 10:01:05,049 - mu.interface.panes:618(start_process) INFO: Python path: ['C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\Python', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2018-04-13 10:01:06,311 - mu.modes.pygamezero:129(stop_game) DEBUG: Stopping script.
2018-04-13 10:01:08,272 - mu.logic:669(show_logs) INFO: Showing logs from C:\Users\yelh\AppData\Local\python\mu\Logs\mu.log
```
it seems pgrun do not work and pgzero do not installed properly

yonghuming, 2018-04-13:
i reinstalled the latested Mu
```
2018-04-13 09:54:08,911 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.0.beta.15
2018-04-13 09:54:08,915 - root:113(run) INFO: uname_result(system='Windows', node='LAPTOP-SSO2P9NF', release='10', version='10.0.16299', machine='AMD64', processor='Intel64 Family 6 Model 78 Stepping 3, GenuineIntel')
2018-04-13 09:54:08,915 - root:114(run) INFO: Python path: ['C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\Python', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2018-04-13 09:54:09,043 - mu.logic:356(__init__) INFO: Setting up editor.
2018-04-13 09:54:09,043 - mu.logic:367(__init__) INFO: Settings path: C:\Users\yelh\AppData\Local\python\mu\settings.json
2018-04-13 09:54:09,044 - mu.logic:368(__init__) INFO: Log directory: C:\Users\yelh\AppData\Local\python\mu\Logs
2018-04-13 09:54:09,044 - mu.logic:369(__init__) INFO: Data directory: C:\Users\yelh\AppData\Local\python\mu
2018-04-13 09:54:09,088 - mu.logic:377(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-04-13 09:54:10,281 - mu.logic:713(change_mode) INFO: Workspace directory: C:\Users\yelh\mu_code
2018-04-13 09:54:10,282 - mu.logic:416(restore_session) INFO: Restoring session from: C:\Users\yelh\AppData\Local\python\mu\settings.json
2018-04-13 09:54:10,282 - mu.logic:417(restore_session) DEBUG: {'theme': 'day', 'mode': 'pygamezero', 'paths': ['C:/Users/yelh/mu_code/colorConfusion.py'], 'workspace': 'C:\\Users\\yelh\\mu_code', 'microbit_runtime_hex': None}
2018-04-13 09:54:10,282 - mu.logic:471(_load) INFO: Loading script from: C:/Users/yelh/mu_code/colorConfusion.py
2018-04-13 09:54:10,282 - mu.logic:499(_load) DEBUG: # -*- coding: utf-8-*-# Encoding cookie added by Mu Editor

from random import choice

colors = ['red', 'blue', 'green']

meaning_text = choice(colors)

meaning_color = choice(colors)

circle_color = choice(colors)

def reset_color():

    global colors, meaning_color, meaning_text, circle_color

    meaning_text = choice(colors)

    meaning_color = choice(colors)

    circle_color = choice(colors)

score = 0

WIDTH = 300

HEIGHT = 490

def draw():

    global meaning_color, meaning_text, circle_color, score

    screen.clear()

    screen.blit('bg', (0, 0))

    screen.draw.text('Score:', (100, 100), color = 'black', fontsize = 32)

    screen.draw.text(str(score), (170, 100), color = 'black', fontsize = 32)

    screen.draw.text(meaning_text, (120, 160), color = meaning_color, fontsize = 32)

    

    screen.draw.filled_circle((150, 260), 16, circle_color)

    

def on_mouse_down(pos):

    print(pos)

[... 1562 lines cut ...]
        else:
            score = score + 1
    else:
        if x > 150:
            score = score + 1
        else:
            score = score - 1
    #===========================================================
    reset_color()
    
    

2018-04-13 10:39:54,683 - mu.interface.panes:608(start_process) INFO: Running script: c:\users\yelh\mu_code\colorconfusion.py
2018-04-13 10:39:54,684 - mu.interface.panes:613(start_process) INFO: Command args: []
2018-04-13 10:39:54,684 - mu.interface.panes:624(start_process) INFO: Working directory: C:\Users\yelh\mu_code
2018-04-13 10:39:54,684 - mu.interface.panes:629(start_process) INFO: Python path: ['C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\Python', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\yelh\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2018-04-13 10:39:58,203 - mu.logic:1030(autosave) INFO: Autosave has detected changes.
2018-04-13 10:40:03,197 - mu.logic:1030(autosave) INFO: Autosave has detected changes.
2018-04-13 10:40:08,201 - mu.logic:1030(autosave) INFO: Autosave has detected changes.
2018-04-13 10:40:13,207 - mu.logic:1030(autosave) INFO: Autosave has detected changes.
2018-04-13 10:40:18,197 - mu.logic:1030(autosave) INFO: Autosave has detected changes.
2018-04-13 10:40:23,201 - mu.logic:1030(autosave) INFO: Autosave has detected changes.
2018-04-13 10:40:28,202 - mu.logic:1030(autosave) INFO: Autosave has detected changes.
2018-04-13 10:40:29,139 - mu.logic:962(show_admin) INFO: Showing logs from C:\Users\yelh\AppData\Local\python\mu\Logs\mu.log
```

yonghuming, 2018-04-13:
i reinstalled python and mu ,everything work well and then i uninstall python, mu could not work well, so seems mu run pgzero related to python on localhost, not seprately

yonghuming, 2018-04-13:
and i reinstalled python, mu will work with pgzero well

ntoll, 2018-05-15:
Closing this since it appears to be a configuration-of-host-machine problem rather than anything to do with Mu.

closed 2018-05-15 by ntoll