mu-editor/mu #429: Splash screen is shown after mode is selected

carlosperate, 2018-04-07:
Just wondering if this is the desired behaviour.

* Install Mu
* Open Mu
* Window opens asking the user to select mode -> select any mode
* The mode window closes and then the splash screen opens

I would probably expect the splash screen to be shown before any action is taken on any of the Mu windows.

Tested from source on latest master at the time of writing (56fc1a3a83822ce30585d6adde1a646f96545c5d) on Raspbian Stretch.

ntoll, 2018-04-07:
Stylesheets will be because RPi is underpowered.

As for the splash / mode select dialog on first start. I think we should just NOT display the splash on first start (only on subsequent starts). Does this make sense?

ZanderBrown, 2018-04-07:
Possibly but on my main machine you'll notice that on 'first-run' the rows in the mode selector have been styled as expected but the toolbar and content area are not
[image]
Dismiss the selector in whatever way and the styling is applied

ntoll, 2018-04-07:
OK... so first start-up needs some thought / potential re-jigging. I suspect the modal nature of the mode selector is blocking application of the theme in some sense. :-/

ZanderBrown, 2018-04-07:
The dialog is shown here and is styled because ModeSelector loads styles internally
https://github.com/mu-editor/mu/blob/c93ef493dcd528123bd2085808a8097989a6b593/mu/logic.py#L594

But the main window isn't styled until here
https://github.com/mu-editor/mu/blob/c93ef493dcd528123bd2085808a8097989a6b593/mu/logic.py#L617

All of which is happening when restore_session is called here
https://github.com/mu-editor/mu/blob/03197d6c4629930abb458e5be5a7f0485cf5f858/mu/app.py#L128

Which is before the splash screen is shown
https://github.com/mu-editor/mu/blob/03197d6c4629930abb458e5be5a7f0485cf5f858/mu/app.py#L136

In fact the fact the splash screen is shown so late it acts more as a delay than an indication things are loading

dybber, 2020-09-29:
I've added a PR that should solve this #1116, all that is left is to figure out how to write tests for it, which ended up being a bit complex, and I haven't quite figured out how to do it best.

dybber, 2020-09-30:
I've added the test, so the PR is ready for review.

commit 7337246e (2020-10-06): Merge pull request #1116 from dybber/show_splash_before_dialog

Show splash screen before mode selector, and fix theming on first run (fixes #429)
closed 2020-10-06 by dybber