mu-editor/mu #432: micro:bit files panel doesn't always exit gracefully MicroPython raw mode

carlosperate, 2018-04-08:
When closing the files mode sometimes it doesn't exit the "raw mode" properly. It doesn't always happen and it is only noticeable if the REPL is entered after the files.

This gif shows an example where, after a few attempts, the REPL gives you the raw mode, and then an empty list (likely the available files returned by `os.listdir()`). The gif shows a soft reboot triggered by pressing "Ctrl+D" still keeps you in the raw mode until "Ctrl+B" is sent.

[image]

ntoll, 2018-04-09:
OK... so don't do that! ;-)

My guess is that you're toggling the files "off" before the `ls` command is able to finish and clean up state. Once the REPL is connected the remaining bytes in the buffer are read and you find yourself in the weird state you outline above.

My original point still stands ("so don't do that!"). You may think it important to handle such a case but then I'd ask, "how?"

We'd essentially have to start building the equivalents of "guard rails" around all the different bits of functionality related to microfs exposed via Mu. This would complicate both the code and the usability of the application (I imagine the "Files" button having to be disabled while such functionality was "in flight"). To my eyes, this would look weird and be exposing side effects of the implementation via the UI, which may create confusion (why is the "Files" button flickering on and off?).

I have an alternative proposal: wabi-sabi (https://en.wikipedia.org/wiki/Wabi-sabi). Mu isn't perfect *and* by exposing such imperfections (despite this being a very "edge" edge case) it's an opportunity for learning in a context where the problem hasn't caused any permanent problem). If this is addressed in Mu's documentation, then (and I've just made sure it has been -- https://github.com/mu-editor/mu-editor.github.io/blob/v1/en/howto/copy_files_microbit.md) then I think we've probably got ourselves covered for this situation.

Is this helpful? Does it make sense? Happy to be talked around, otherwise. :-)

carlosperate, 2018-04-09:
> My guess is that you're toggling the files "off" before the ls command is able to finish and clean up state. Once the REPL is connected the remaining bytes in the buffer are read and you find yourself in the weird state you outline above.

Yep, I agree that it's very likely what is happening.

> My original point still stands ("so don't do that!"). You may think it important to handle such a case but then I'd ask, "how?"

It's not like I was just toggling the buttons to break things (well, I was for this gif :), but this has happened more than a couple of times with just normal use.
In fact when I opened this issue it had just happened while I was doing a screen recording using Mu, so I was being very concious about the steps I was following to show normal usage and I got into this state any.

I think it might be important to handle this case, because when it happens it looks like Mu is simply not executing the REPL. If you close and reopen the REPL it shows a blank screen that does not respond to any of the user input. You have to either restart the micro:bit or Mu to get it back to work:

[image]

> We'd essentially have to start building the equivalents of "guard rails" around all the different bits of functionality related to microfs exposed via Mu. This would complicate both the code and the [...]

Not sure I follow you here. I think all we have to do is hook a listener for the filesystem thread "on exit" to send a "ctrl+c" command to stop any currently running operation and a "ctrl+b" (or `microfs.raw_off()`) to exit the raw input mode.
If an operation was running when the panel was removed then that should exit gracefully, if nothing was running then sending those commands woudn't really have much of an effect.

Of course it's never that easy with threads, so I'm more than happy to have a look at and create a PR if the solution is viable. Perhaps that way I'll get a better view of what you mean and realise it is a lot more fiddly than I expected (which is most often the case).

> I have an alternative proposal: wabi-sabi (https://en.wikipedia.org/wiki/Wabi-sabi). Mu isn't perfect and by exposing such imperfections (despite this being a very "edge" edge case) it's an [...]

I agree that learning oportunities can be benefitial, but I do not think this is one of them.
When the user experiences a problem with no indications of any wrongdoing, what the problem is or what might have caused it, then it's most likely going to be viewed as a frustating "bug" rather than oportunity to learn. Taking  that in consideration, if the users see this as a bug I very much doubt they will go to the documentation either. 

> Is this helpful? Does it make sense? Happy to be talked around, otherwise. :-)

Always helpful, after all, having differences of opinion and these conversations is how we all better explore other perspectives!

closed 2018-04-12 by ntoll

carlosperate, 2018-04-13:
Thanks @ntoll, that is certainly a lot simpler than what I was trying!

It's much better now. It can still get into a situation where either Mu send some code or the micro:bit replies and it prints either of those into the REPL panel, but it does exit the raw mode and let's you use the REPL without trouble.

So sometimes it shows something like:

```
import os >>> 
```
or:
```
OK>>>
```

But that's fine, you can still use the REPL without issues, and the note in the documentation cover the reason. 

Thanks!!

commit e164c5c3 (2018-04-12): Fix #432.