mu-editor/mu #433: mu does not install pygame zero on some windows 7

yonghuming, 2018-04-08:
we have hold some online coding course for kids
some kids install mu on windows 7 and seems it do not install pgzero on windows 7
but no problems with windows 10.
a python script using pygame zero could run properly on my most sutdents's computer but not a little sutdent.
when run pygame zero on windows 7 it shows nothing

yonghuming, 2018-04-08:
there Python\Python36-32\Scripts is empty

yonghuming, 2018-04-08:
[image]
mybe bug of python

ntoll, 2018-04-08:
Are you using Mu as installed with the Windows installer..? If so, it doesn't use the system Python but you'll find the packages in somewhere under the "Program Files\Mu\ " directory.

yonghuming, 2018-04-09:
ok i'll try to fix the problem

yonghuming, 2018-04-09:
i checked user's install dir, and no Program Files\Mu\ " directory. seems the pip do not install correctly on windows 7

yonghuming, 2018-04-09:
though mu use seperate python env, but after i install pgzero on windows 7 use pip through command, pygame zero  mode will work properly

yonghuming, 2018-04-09:
[image]

ntoll, 2018-04-09:
Ahh... so you're using `pip` and *NOT* the Windows exe installer..? OK... then the directory I mention won't exist.

I guess there's a pip related problem (are you using the correct version of Python/pip..?). PyGame Zero is *definitely* in the requirements for pip to install... so I'm guessing some sort of configuration problem on your PCs.

ntoll, 2018-04-09:
Hmm... perhaps the version of Mu on PyPI is older and doesn't include the PyGame Zero requirement. In any case, this will be fixed in the next release. :-)

yonghuming, 2018-04-09:
seems pgzero not install but i got no error when run pygame games script

yonghuming, 2018-04-09:
some of our students use windows met this problem

ntoll, 2018-04-09:
OK... this is fixed in current master, and when the next beta release lands in PyPI then it'll work properly for you. The version in PyPI is relatively old at this moment in time.

closed 2018-04-09 by ntoll

yonghuming, 2018-04-09:
when will next beta be released?

ntoll, 2018-04-09:
Soon (perhaps in about a week). :-)

If you want the latest version, use Git to clone the repository and, in a virtualenv, `pip install .`

yonghuming, 2018-04-09:
ok
thank you