mu-editor/mu #435: Setting microbit_runtime_hex in settings.json has no effect

WayneKeenan, 2018-04-11:
I'm trying to get `mu' to use a custom firmware by setting `microbit_runtime_hex ` in `settings.json` , but it has no effect, examples:

```
{
  "theme": "day",
  "workspace": "/Users/wayne/mu_code",
  "paths": [],
  "microbit_runtime_hex": "microbit-blue-pairing-not-required.hex"
}
```

and

```
{
  "theme": "day",
  "workspace": "/Users/wayne/mu_code",
  "paths": [],
  "microbit_runtime_hex": "/Users/wayne/Library/Application Support/mu/microbit-blue-pairing-not-required.hex"
}
```

ntoll, 2018-04-11:
This is duplicate of #434

ntoll, 2018-04-11:
Sorry... my mistake. This needs investigating.

ntoll, 2018-04-13:
This has been fixed by  bf8bb75 which just landed in master. It'll be in beta.16

closed 2018-04-13 by ntoll