mu-editor/mu #436: Flash modal popup behaviour change request.

WayneKeenan, 2018-04-11:
Please can the way the info in the modal pop-up that appears after pressing flash be presented in a way such that it doesn't have to be manually dismissed, e.g. a status line or a perhaps an auto closing notification bubble in the top corner.

Being able to prevent it from appearing at all via config would be an alternative.

Thanks
Wayne

ntoll, 2018-04-12:
Hi @WayneKeenan,

Thanks for the feedback! However, I think you're on an old version of Mu, since this was changed several months ago in master and works in exactly the way you suggest. Please *DO NOT* use the version of Mu currently available via the main website (http://codewith.mu/) since this is the old version. I hope to update things VERY SOON so the latest version of the beta is available for download via the website. However, until such time, please use the version you get from using `pip` to install the editor (which is tracking the beta releases). You'll see we've already made the change you suggest! ;-)

closed 2018-04-12 by ntoll

WayneKeenan, 2018-04-12:
ah, that's better. ooo, lots of new toys too!  thanks @ntoll

ntoll, 2018-04-12:
You're welcome @WayneKeenan -- all feedback most welcome! :-)