mu-editor/mu #437: Debugger breakpoint markers cannot be removed

carlosperate, 2018-04-12:
When a debugger marker is placed on the marging and then the text is modified, the marker moves with the changes. When this happens then it's not possible to remove the markers anymore.

Gif showing the issue:

[image]

ntoll, 2018-04-12:
Yeah... tell me about it. ;-)

Next week I've set aside some (significant) time to fix the outstanding debugger related issues.

carlosperate, 2018-06-13:
@ntoll to save us from future pain, would it be a good idea to only allow to add breakpoints when in debug mode, and have them all removed when the debug mode is closed?

ntoll, 2018-06-13:
Nope... you need to be able to add breakpoints *before* debugging so the debugger knows when to stop (otherwise it just runs the script, but slowly). I have some refactoring that'll fix this. Sorting out the debugger is my next thing to fix (probably tomorrow). :-)

carlosperate, 2018-06-13:
Okay.

My reservation comes from having had issues with breakpoints moving around, in pretty much the mayority of IDEs I've used (open source and commercial ones, except maybe JetBrains). So unless we remove breakpoints when the text is edited, trying to folow changes could be one of those problematic areas with a lot of issues.

ntoll, 2018-06-13:
I hear ya!

When in debug mode, you can't edit the text anyway (for that very reason). Let me work on my refactoring and you'll be the first to know. ;-)

As always, feedback would be most welcome.

carlosperate, 2018-06-14:
Cool, I'll give it a spin when it's ready :)

ntoll, 2018-06-14:
Well, I've already got it working this afternoon by changing my approach to how breakpoints are tracked. I've a bunch of debugger based changes taking place on this branch (they're all related/overlap in a sense) so when all the work/refactoring is done I'll definitely let people know. :-)

ntoll, 2018-06-19:
Fixed in af47bc1ba247856af88d5f504a448c8360ea3314

closed 2018-06-19 by ntoll