mu-editor/mu #441: Raspberry Pi doesn't have PyQtChart

carlosperate, 2018-04-14:
There is no ARM linux compatible package for PyQtChar in PyPI or piwheel:

```
$ pip install PyQtChart --verbose
Created temporary directory: /tmp/pip-ephem-wheel-cache-ajv09ya3
Created temporary directory: /tmp/pip-install-cisywfag
Looking in indexes: https://pypi.python.org/simple, https://www.piwheels.org/simple
Collecting PyQtChart
  2 location(s) to search for versions of PyQtChart:
  * https://pypi.python.org/simple/pyqtchart/
  * https://www.piwheels.org/simple/pyqtchart/
  Getting page https://pypi.python.org/simple/pyqtchart/
  Looking up "https://pypi.python.org/simple/pyqtchart/" in the cache
  Current age based on date: 159
  Freshness lifetime from max-age: 600
  Freshness lifetime from request max-age: 600
  The response is "fresh", returning cached response
  600 > 159
  Analyzing links from page https://pypi.python.org/simple/pyqtchart/
    Skipping link https://pypi.python.org/packages/84/92/732e11d0ccf603abcc899b3df136e032d8d8930d5a39d825c110a0100731/PyQtChart-5.10-5.10.0-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl#md5=207d19bb1170afe9dc8fe2f97cacd6fa (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/9a/9a/574f998b7cea21c1fbc34bb997090f4eaef226ebfb3201e2b93bd19d0126/PyQtChart-5.10-5.10.0-cp35.cp36.cp37-abi3-manylinux1_x86_64.whl#md5=4b69108afcd11fa11d2f417ea30bd5dd (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/db/e6/ed16b1ecd2427e86ce427ed3ed9cc4ff20865fc365bc08f0b20a57874458/PyQtChart-5.10-5.10.0-cp35.cp36.cp37-none-win32.whl#md5=f4e2037b6f194c2ceefbe837e560a66e (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/90/53/33b2b433131daf818a5c7bc22a5f34d0e4d23d353236431ae8c0b9621099/PyQtChart-5.10-5.10.0-cp35.cp36.cp37-none-win_amd64.whl#md5=2227b68ad1ed06206bc98df3210c7030 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/14/f2/c59413bf7291137ed8270a09aadfd3f6b8ef48fa8995ddd72911fabb0279/PyQtChart-5.10.1-5.10.1-cp35.cp36.cp37.cp38-abi3-macosx_10_6_intel.whl#md5=00c1b6c75e768e538e1f333331aa4ec2 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/53/2e/6754d1656d1ecf2ac57b10a5f1090274d78f30e39d2c4cf8430312b788f1/PyQtChart-5.10.1-5.10.1-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl#md5=15a27f997b54e87411c770affd0d50fb (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/bb/ca/4a6bf4ffd3098c2a199aaea9dcfc2dcfb251c7300aaf514903fc3ad2d0a9/PyQtChart-5.10.1-5.10.1-cp35.cp36.cp37.cp38-none-win32.whl#md5=4da2dcf44572eef57077a66dc8ce7523 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/29/3b/dcc97d7a57830ac448087aefda19dd4bf478895fc4c3a65b2097e4a94734/PyQtChart-5.10.1-5.10.1-cp35.cp36.cp37.cp38-none-win_amd64.whl#md5=5adb55512d334cf5dd4b2105aa99a591 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/8b/73/357511cf1d231d7cbc0464ec283ad9c5a9cfc6e4045b0e8f10ddfec33974/PyQtChart-5.7-cp35-cp35m-macosx_10_6_intel.whl#md5=4148d2192a12ca675cd16684acf4f93e (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/a0/25/64a49f1c023c3ebddc1e8f3305b062f463e5674b6a1bdaab67dde7b4552d/PyQtChart-5.7-cp35-cp35m-manylinux1_x86_64.whl#md5=9a6e195bccebfdd84e7e715bf49a57c1 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/73/0b/7fc7aeffce6fb264374f8a345cbbe2967841eceac018376cafe1c1f77d7e/PyQtChart-5.7-cp35-none-win32.whl#md5=f3803a8513bbcfc834c1f7945efbcb30 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/d9/40/9fcf332dae9f8c72b8355c54eceb0af6661179c53bb9f80604651b8b708b/PyQtChart-5.7-cp35-none-win_amd64.whl#md5=d07773c83b3e3875801978c653f7ba1d (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/e5/d2/56927d8760592617c49b28c54ad80ed98321041fa050a104398d7afd8bd6/PyQtChart-5.7.1-5.7.1-cp34.cp35.cp36-abi3-macosx_10_6_intel.whl#md5=2d27b56fbd5c27387cc62cfd4acd627b (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/3c/bf/15a6bf4dd7bd91696ad67b7281a92fa3ac0efbc69a579f944757a933b45e/PyQtChart-5.7.1-5.7.1-cp34.cp35.cp36-abi3-manylinux1_x86_64.whl#md5=beb7101208ae9e697789902741224d59 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/2c/ad/44ce426ec312b6dec67abebe59d029b5cc196471aee16b3cd2b9ea6a972f/PyQtChart-5.7.1-5.7.1-cp34.cp35.cp36-none-win32.whl#md5=d5784699fea08d8ef489d6e4e90cb376 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/8d/41/263846474f7ac01ac744fa5e176c8ae70fd725cb424c0f08bbc2826228d4/PyQtChart-5.7.1-5.7.1-cp34.cp35.cp36-none-win_amd64.whl#md5=5628f53fea79119b0b0ba0cf1a07d969 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/5a/40/5a942f5e29e43c6844951aac92087c9fd65440e5f192464f923f6ceb4f0c/PyQtChart-5.8-5.8.0-cp34.cp35.cp36.cp37-abi3-macosx_10_6_intel.whl#md5=b6e08ec1579b5d6e9418c7a79eb6e5e5 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/dc/8b/4e537924cc8c5c52222827b46135fdc9984066bf20d595a62c88bac1fbdd/PyQtChart-5.8-5.8.0-cp34.cp35.cp36.cp37-abi3-manylinux1_x86_64.whl#md5=303fbc50b60ba79857d6e8e623f21f9f (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/a6/7d/140b6f5bd30fbf77582455e9160f200e3c30e190c8f5acbfd5908274f3d6/PyQtChart-5.8-5.8.0-cp34.cp35.cp36.cp37-none-win32.whl#md5=224abcb0c654a3888383397b6929faa9 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/e7/04/62951e3a875c0e22d08e83bc73a2ea1f9b3995c63da09693f42b7dedbeb7/PyQtChart-5.8-5.8.0-cp34.cp35.cp36.cp37-none-win_amd64.whl#md5=64bc8fa8c708ac3a3d12fc595008e556 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/8d/8a/cd24a9576fb3a837f53d85d92f4f5e28f1f1095bcd077f6df91baea4c3a7/PyQtChart-5.9-5.9.1-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl#md5=d97c669987dd2b3a504d6ae684ef7098 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/63/ea/0b79f9d05cb8f9d1fcaba1b1cd9316576161e80980c13c18d1b77c72f1ed/PyQtChart-5.9-5.9.1-cp35.cp36.cp37-abi3-manylinux1_x86_64.whl#md5=b8089a5e5670c78be9725d390fc40717 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/c5/ca/c9e7148c4900f01b904cff1920d24b0b16dd9329197d0fcb606e6d0e58cf/PyQtChart-5.9-5.9.1-cp35.cp36.cp37-none-win32.whl#md5=854e28a7a2674e9112fa37a9c43c11cf (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/71/76/b4fbad6a306f7d9742755de5f846c026c6467e8f2c3a7576c24fa20f88ab/PyQtChart-5.9-5.9.1-cp35.cp36.cp37-none-win_amd64.whl#md5=d70fe2e501a7911bfa3f4bf5294b5915 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
    Skipping link https://pypi.python.org/packages/0a/ee/a812c1d82db37d99fa57f6e323afcd7994c49a3c572b09b9dd453b0ffc92/PyQtChart-5.9.1-5.9.2-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl#md5=a0905fb7e027f095a0dc8f0e274ae530 (from https://pypi.python.org/simple/pyqtchart/); it is not compatible with this Python
[... cut ...]
python3-spyder - Python IDE for scientists (Python 3 modules)
spyder-common - Python IDE for scientists (common files)
spyder-doc - Python IDE for scientists (Documentation)
```

So there is no plotter functionality in Raspberry Pi :(

Getting PyQt things to built on a raspberry pi has proven difficult, so that's a can or worms I'd rather not open, but maybe we could build a wheel for piwheels.

I wonder if ARM debian has a package for PyQtChart that we could adopt in raspbian, not sure if they already take all, or how that hierarchy works.

ZanderBrown, 2018-04-14:
Originally Raspbian was merely a rebuild of Debian (that said there are a lot of packages :slightly_smiling_face: ) so 99.9% of Debian was in Raspbian (the .1 being packages they couldn't build) but over the years things seem to have become a little more complex

`python3-pyqtgraph` looks like the right package in Debian

ZanderBrown, 2018-04-14:
Nope that's not it, doesn't seem to be in upstream Debian repos either

ntoll, 2018-04-14:
See: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=892198

I submitted this several weeks ago but have not had any acknowledgement or seen any movement. I guess we'd need to go to Debian with some sort of solution.

Perhaps we should reach out to Riverbank and see if they might help..?

carlosperate, 2018-04-15:
I might be wrong, but I think Ben might have contacted Riverbank about ARM/Raspberry pi wheels in the past. @bennuttall did you email/talk to them? If so what was their general response?  

Btw, why targetting Qt 5.9? Is that the version Debian strech is packaging? 
Raspbian stretch only provides PyQt 5.7 (and to make things even more complicated, the Qt dev packages like `qtdeclarative5-dev` and `qt5-default` are at 5.7.1 ):

```
$ apt-get install python3-pyqt5
apt-cache policy python3-pyqt5
python3-pyqt5:
  Installed: 5.7+dfsg-5
  Candidate: 5.7+dfsg-5
  Version table:
 *** 5.7+dfsg-5 500
        500 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status
```

ntoll, 2018-04-15:
Yup, @bennuttall did get in touch with Riverbank... but that was about source distributions and making binary wheels for ARM (i.e. Raspberry Pi) based devices. Their response was "too complicated and not worth the effort given the user base". They may respond more positively to a query about QCharts.

As for 5.9 -- that's the version of Qt currently in Debian testing.

bennuttall, 2018-04-16:
@waveform80 did some digging recently:

> Found out why QtCharts is missing: it was only introduced in Qt5.7 (previously it was a commercial only module), and Ubuntu Xenial is on Qt5.5
> 
> Apparently it'll be in Artful onwards
> 
> Related bug: https://bugs.launchpad.net/ubuntu/+source/qtcharts-opensource-src/+bug/1682643 …
> 
> Just checked Raspbian and although Stretch is on Qt5.7 I'm guessing it didn't quite make it in there; guessing Buster might include it

Stretch was released in June 2017 and I'd expect Buster around mid-2020.

I thought that the Plotter might work with gpiozero on our x86 distro, but Riverbank don't even provide x86 wheels (just x86_64).

bennuttall, 2018-04-16:
> Would packages in Debian Testing end up in the current Debian Stable Stretch repos?

Testing is Buster, until Buster is released. So no, Testing is about getting stuff ready for the next Debian release. Occasionally things are available in backports, so there's a possibility we could pull something from there, as long as it didn't require the Buster versions of all the things.

bennuttall, 2018-04-16:
> Although if it came down to us having to do the packaging, then a "one off" wheel on piwheels would be the path of least resistance.

If we could build it ourselves, we would, but it's undocumented and doesn't use standard tools. If Riverbank built it on a Pi (and we had the right version of qt for it) and supplied us with a wheel, we could stick it on piwheels and that would allow people to install it to unlock the plotter. But I can't see that happening.

carlosperate, 2018-04-16:
Well that was "fun"...

After many failed attempts at getting the right dependencies with the right versions I managed to get PyQtChart compiled on the Pi.

In the end I had to compile from source SIP (the package `python3-sip-dev` is at version 4.18, and it complained that I needed to use at least v4.19) and QtCharts, to be get a successfull build of PyQtChart 5.7.
It would have been a bit easier if the `libqt5charts5` package was available at version 5.7, but at least having this available at 5.9 should simplify this in the future:
http://raspbian.raspberrypi.org/raspbian/pool/main/q/qtcharts-opensource-src/

However, once everything was built and installed it turns out that I didn't use the right version of SIP (the error said to use at least version 4.19, so I used exactly 4.19), so there was a mismatch of APIs between different PyQt modules.

```
$ python3
Python 3.5.3 (default, Jan 19 2017, 14:11:04) 
[GCC 6.3.0 20170124] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from PyQt5.QtChart import QtChart
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: the sip module implements API v11.0 to v11.3 but the on[QColor, Qt.GlobalColor, QGradient]) module requires API v12.0
```

I'll continue tomorrow, next step is to figure out what SIP version has API v12, and then recompile PyQtChart with that version instead.

***

Note for future self: These are the steps that got me to this point. It's possible there might be some dependencies or steps missing, I tried a couple things that led me to the wrong path, and I excluded those, but I the stuff below should be enough.

```
$ apt-get install python3-pyqt5
$ apt-cache policy python3-pyqt5
python3-pyqt5:
  Installed: 5.7+dfsg-5
  Candidate: 5.7+dfsg-5
  Version table:
 *** 5.7+dfsg-5 500
        500 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status

$ apt-get install qtdeclarative5-dev
$ apt-cache policy qtdeclarative5-dev
qtdeclarative5-dev:
  Installed: 5.7.1-2+rpi1
  Candidate: 5.7.1-2+rpi1
  Version table:
 *** 5.7.1-2+rpi1 500
        500 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status

$ sudo apt-get install qt5-default
$ apt-cache policy qt5-default
qt5-default:
  Installed: 5.7.1+dfsg-3+rpi1
  Candidate: 5.7.1+dfsg-3+rpi1
  Version table:
 *** 5.7.1+dfsg-3+rpi1 500
        500 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status

$ apt-get install pyqt5-dev
$ apt-cache policy pyqt5-dev
pyqt5-dev:
  Installed: 5.7+dfsg-5
  Candidate: 5.7+dfsg-5
  Version table:
 *** 5.7+dfsg-5 500
        500 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status

$ apt-get install pyqt5-dev-tools
$ apt-cache policy pyqt5-dev-tools
pyqt5-dev-tools:
  Installed: 5.7+dfsg-5
  Candidate: 5.7+dfsg-5
  Version table:
 *** 5.7+dfsg-5 500
        500 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status

$ apt-get install pyqt5.qsci-dev
$ apt-cache policy pyqt5.qsci-dev
pyqt5.qsci-dev:
  Installed: 2.9.3+dfsg-4
  Candidate: 2.9.3+dfsg-4
  Version table:
 *** 2.9.3+dfsg-4 500
        500 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status

$ apt-cache policy qt5-qmake
qt5-qmake:
  Installed: 5.7.1+dfsg-3+rpi1
  Candidate: 5.7.1+dfsg-3+rpi1
  Version table:
 *** 5.7.1+dfsg-3+rpi1 500
        500 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status
```

SIP 4.19:

```
$ wget https://netcologne.dl.sourceforge.net/project/pyqt/sip/sip-4.19/sip-4.19.tar.gz
$ tar zxvf sip-4.19.tar.gz
$ cd sip-4.19
$ python3 configure.py
$ make -j 3
$ sudo make install
```

QtCharts 5.7:

```
$ git clone git://code.qt.io/qt/qtcharts.git -b 5.7
$ cd qtcharts
$ qmake -r
$ make
$ sudo make install
```

PyQtChart 5.7:

```
$ wget https://datapacket.dl.sourceforge.net/project/pyqt/PyQtChart/PyQtChart-5.7/PyQtChart_gpl-5.7.tar.gz
$ tar zxvf PyQtChart_gpl-5.7.tar.gz
$ cd PyQtChart_gpl-5.7
$ python3 configure.py --qtchart-version=2.0.1 --verbose
$ make
$ sudo make install
```

carlosperate, 2018-04-17:
So it looks like a newer SIP was the opposite direction of what we needed. With 4.19 I had:

```
RuntimeError: the sip module implements API v11.0 to v11.3 but the on[QColor, Qt.GlobalColor, QGradient]) module requires API v12.0
```
And with 4.19.8:

```
RuntimeError: the sip module implements API v11.0 to v11.3 but the on[QColor, Qt.GlobalColor, QGradient]) module requires API v12.4
```

So I tried 4.18.1 and voila!
[image]

Now the difficult part might be packaging, does anybody know of any good tutorials or articles explaning the intricacies of building wheels for complex stuff like this?

ntoll, 2018-04-17:
@carlosperate, I am in awe of your stonkingly amazing work on this. It's heroic. It's immensely valuable. It always makes me grin like a Cheshire Cat. Kids and teachers are going to love this!

So @bennuttall is *the world expert* on making binary wheel packages for Raspbian from source through his exceptional work on PiWheels, and you're both on this thread already..!

'nuff said. :-D

XECDesign, 2018-06-13:
@carlosperate I'm on the raspberrypi.org end, not raspbian.org.

When the package makes it into Debian stable, raspbian.org should automatically pick it up, but I'm guessing that's not going to happen any time soon.

We may be able to backport the package before it hits stable and add it to stretch through the raspberrypi.org repo before then.

carlosperate, 2018-06-13:
Thanks for the quick reply @XECDesign!

> @carlosperate I'm on the raspberrypi.org end, not raspbian.org.

Right, I see, sorry, didn't realised that one was managed by a different organisation.

> When the package makes it into Debian stable, raspbian.org should automatically pick it up, but I'm guessing that's not going to happen any time soon.

Yeah, this was my concern as well, might be quite a while until that happens.

> We may be able to backport the package before it hits stable and add it to stretch through the raspberrypi.org repo before then.

Sounds like the best path forward. Who could help us do that? And what can we do to help with the process?

XECDesign, 2018-06-14:
> Sounds like the best path forward. Who could help us do that? And what can we do to help with the process?

That's something I can do as long as there are no show-stopping issues. It may be possible that pyqtchart depends on a version of Qt that's newer than what's available in Stretch.

I'll wait for the package to make it into unstable or testing and stabilise a bit before looking into it.

carlosperate, 2018-06-14:
Thanks @XECDesign, I'll keep an eye as well and we'll have definite info at that point.

From the looks of it, the package from https://salsa.debian.org/skitt/pyqt5chart might be for Qt/PyQt/QtCharts v5.10.1 (the Qt version released for the latest PyQt5), and when I last installed PyQt5 using `apt-get install python3-pyqt5' in Raspbian Stretch (16th of April) it installed version 5.7.

Debian Stretch stable has 5.7 as well: https://packages.debian.org/stretch/python3-pyqt5
However, Debian Stretch testing has 5.10.1 already: https://packages.debian.org/testing/python3-pyqt5

I assume Raspian only packages from debian stable?
This being the case, these are the three options I see, but I would appreciate it if you could correct me or let me know if my assumptions here are wrong, as this is uncharted territory for me. We could:
1) Backport all PyQt5 related packages from Debian testing to Raspbian (all at Qt 5.10.1)
    - This sounds the most painful or risky
2) We could fork the package and modify it to use Qt/PyQt/QtCharts at v5.7
    - Could we just do that? Having a similar packaging hosted in GitHub and use it only for creating a package to be included in Raspbian repo?
   - This sounds like perhaps the path of least resistance? Although I'd be a one-off as future Debian/Raspbian releases will include this package by default
3) We could talk with `@skitt` to see if there is any option to update the debian repo (or create a new one) for Qt v5.7?
    - Not sure how debian packaging works, but if all Qt packages are at 5.10.1 in Debian testing and debian unstable, I doubt they could create a package in unstable for Qt 5.7.

XECDesign, 2018-06-14:
1) We'd need a very good reason to do that, so it's very unlikely.

2) This is probably the most realistic approach. If there is a version of qtcharts that works with pyqt 5.7 (either an older or patched version) then that would be a starting point for me.

3) That doesn't really fit into Debian's development cycle.

skitt, 2018-06-14:
Right, we can’t upload a 5.7 build into Debian since unstable has 5.10.1 already and all packages must build in unstable first (apart from security updates etc.). I *do* have a 5.7 build though so I could make that available somewhere... I’m still figuring out a build issue with 5.10.1, I’ll deal with 5.7 once that’s done.

carlosperate, 2018-06-14:
Thank you so much @XECDesign and @skitt for all your help and guidance!

@skitt, when you get all the details sorted, if you could make a 5.7 build in a different repo so that @XECDesign could create the Raspbian package it would be absolutely fantastic. Thanks again for all your efford moving this forward, your help here has been truly invaluable!

faisolp, 2019-05-03:
> So it looks like a newer SIP was the opposite direction of what we needed. With 4.19 I had:
>
> ```
> RuntimeError: the sip module implements API v11.0 to v11.3 but the on[QColor, Qt.GlobalColor, [...]

Hi carlosperate
I is beginner developer 
can you help me show sample code for this your graph

carlosperate, 2022-07-16:
The `python3-pyqt5.qtchart` was added in Debian/Raspbian Buster, so we can close this one 🎉

closed 2022-07-16 by carlosperate