mu-editor/mu #442: mu get different result same function to IDLE

yonghuming, 2018-04-15:
i wrote i function to cal the sum two inputed num

[image]

but it get wrong answer,and i get the right answer in IDLE
[image]

yonghuming, 2018-04-15:
[image]
but right answer if print b and c

yonghuming, 2018-04-15:
[image]

ntoll, 2018-04-15:
That's weird... looking into it right now. All Mu is doing, is showing you the output from a standard Python process. I wonder what's going on.

ntoll, 2018-04-15:
I can reproduce this bug... I'll see what's going on.

yonghuming, 2018-04-15:
[image]

yonghuming, 2018-04-15:
fill really strange

yonghuming, 2018-04-15:
i think it's a bug of mu for input() works well in IDLE

ntoll, 2018-04-15:
I believe it's a bug in the Python runner pane that's handling input/output from the Python process. It shouldn't take long to prove (and if so, fix).

yonghuming, 2018-04-15:
加油

ntoll, 2018-04-15:
Heh... (thanks to Google translate)

I've found the problem. It's as I expected, a problem in the input/output pane for Python processes. I'll fix it now.

yonghuming, 2018-04-15:
how efficient of you

yonghuming, 2018-04-15:
and also it seems mu could not run pygame in some computer, i need reinstall python 3.x and mu to fix this problem ,and this drove me crazy......the script is no error, but show nothing after press play button

ntoll, 2018-04-15:
Found the problem... just need to write some tests to prove it works and doesn't break anything. But in the meantime, here's picture proof:

[image]

yonghuming, 2018-04-15:
wanderful,

ntoll, 2018-04-15:
OK... all pushed to `master`. Thanks for reporting this.

The function that I had to change is probably one of the most complicated in Mu, and I wouldn't be surprised if there are more bugs. In any case, thanks to the tests, it was very easy to find, diagnose and fix. :-)

謝謝 :-)

commit b86ce645 (2018-04-15): Set start_of_current_line on newline (and associated test adjustment). Fixes #442
closed 2018-04-15 by ntoll