mu-editor/mu #443: pygame bloat 1.0.0.beta.15

Sembazuru, 2018-04-16:
I'm using Mu 1.0.0.beta.15 for Adafruit Circuit Python on Win10x64.

I find that when I have my board plugged into my USB port (volume name CIRCUITPY) and start Mu, if the two image files and the sound file for pygame aren't on the board the only thing that happens is those files get created on the board. After that happens (or if the files already exist) when I start Mu the editor then opens in my pre-selected Adafruit mode as expected.

The issue is two fold.
1. I'm not planning on using pygame right now, so the auto installed asset files are bloatware clutter.
2. When using a limited storage space board like the Gemma M0 the file system is all internal to the SAMD21 chip, no external flash storage like the Adafruit Express boards, There is only 59KiB for file system, so the pygame asset files take up about 1/3 of the available file system space.

I have a work-around where I have replaced the asset files with empty files of the same name. This solves the waste of file system space, but doesn't solve the clutter issue.

Maybe the check for the pygame asset files should be done after Mu figures out that the pygame mode is the selected mode.

carlosperate, 2018-04-16:
Thanks for the report, this was a bug that has been fixed in master but hasn't been released yet: https://github.com/mu-editor/mu/issues/358

Did you install Mu using pip? If you don't mind running from source, then the master branch should not be creating those files in your circuit playground anymore.
Instructions on how to run from source can be found in the developer docs: https://mu.readthedocs.io/en/latest/setup.html

ntoll, 2018-04-16:
Hi @Sembazuru 

Thanks for the bug report. As @carlosperate says, this issue has already been fixed in `master` and will be in the next release. Yay for beta testing... (and apologies for the inconvenience) :-)

N.

Sembazuru, 2018-04-16:
Thank you for the very prompt reply. Since you say that this was fixed already, I'll go ahead and close this issue.

I actually installed Mu using the windows installer "mu_2018-02-21_15_12_master_2a33437_32bit.exe" downloaded using the link on Adafruit's "Welcome to CircuitPython!" learning page: https://learn.adafruit.com/welcome-to-circuitpython/installing-mu-editor
For now I'll keep my empty file placeholders and watch for a final release of 1.x, hopefully there the build that I have is stable enough until then.

I finally found the build page at: http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=windows/ so I can see that there are many "nightly" builds available since the version that I downloaded. I usually try to avoid nightly builds (at least until RC status)... I'll decide later if I'll tempt fate with beta versions on my own...

closed 2018-04-16 by Sembazuru

carlosperate, 2018-04-16:
> I actually installed Mu using the windows installer "mu_2018-02-21_15_12_master_2a33437_32bit.exe" downloaded using the link on Adafruit's "Welcome to CircuitPython!" learning page: https://learn.adafruit.com/welcome-to-circuitpython/installing-mu-editor

Ahh, that explains the increase in S3 traffic from `mu-builds` :)

It's hard to tell which version that is just from the name, looking at the commit hash (2a33437666483ec6e24f2bf588a13608c07db082) it matches with the version uploaded to PyPI as beta15, so they linked to the right version there. 

@ntoll for the beta16 we could tag it as pre-release in GitHub and host the executable there?

ntoll, 2018-04-16:
@carlosperate you bet'cha... ;-)