mu-editor/mu #444: iPython REPL crashes Mu

carlosperate, 2018-04-17:
To replicate:

* Open Mu
* Select Python mode
* Open REPL
* Execute `dir()`
* Exception closes Mu:
```
Traceback (most recent call last):
  File "/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6/site-packages/qtconsole/base_frontend_mixin.py", line 163, in _dispatch
    handler(msg)
  File "/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6/site-packages/qtconsole/rich_jupyter_widget.py", line 151, in _handle_execute_result
    return super(RichJupyterWidget, self)._handle_execute_result(msg)
  File "/Users/microbit-carlos/.local/share/virtualenvs/mu-upstream-umoz5JlW/lib/python3.6/site-packages/qtconsole/jupyter_widget.py", line 247, in _handle_execute_result
    self._append_plain_text(self.output_sep, True)
TypeError: _append_plain_text() takes 2 positional arguments but 3 were given
```

Log: [mu.log](https://github.com/mu-editor/mu/files/1918525/mu.log)
macOS High Sierra, Python 3.6, qtconsole 4.3.1

ntoll, 2018-04-17:
Good spot. That's my fault (I had to override `_append_text` in the widget we use so it emits a signal the plotter can handle. I've called the `super` version of the function with the wrong args. Fixed now on master. :-) Sorry 'bout that.

closed 2018-04-17 by ntoll

carlosperate, 2018-04-17:
Thanks for the quick fix!

commit 1f6ce525 (2018-04-17): Fix missing *args in _append_plain_text. Fixes #444.