mu-editor/mu #445: Feature suggestion: Offer user to change mode when device is connected

microbit-carlos, 2018-04-18:
When the beta 15 hit the Raspbian repos we had a few reports of users struggling to flash the micro:bit because they were in a different mode without realising it (e.g., they clicked "run" and got a `ModuleNotFoundError` errors).

Beta 15 was released before the "modes" button was added, so having this button in the next version should help users realise they have different modes for different purposes.

However because some users might upgrade Mu without knowing that it has evolved from a "micro:bit editor" to a multipurpose Python editor, we were wondering if there was anything we could do to help with the transition.

Mu currently scans USB devices for anything new connected and prints that info in the log and status bar. What are your thoughts about updating it so that if a micro:bit or CircuitPython device is detected it asks the user if they want to change modes? 

[image]

Something similar could be done as well when loading hex files:

[image]

If a user plugs a new device, or opens a micro:bit hex file, is very likely that they will want to use Mu in that mode, so even though the pop-up message is a bit more intrusive I think it offers useful information and action that the user likely want to take. What do you think?

Of course, we'd be happy to provide a PR for this or whatever that might come out of this discussion.

ZanderBrown, 2018-04-18:
I quite like both these proposals but the "we noticed a device" should be fairly easy as we are already polling for devices

microbit-carlos, 2018-04-18:
Thanks @ZanderBrown, and yes the changes should be relatively simple (proof-of-concept code in https://github.com/microbit-carlos/mu/commits/detect_device fork branch).

ntoll, 2018-04-18:
+1 LGTM... care to rustle up a PR..? ;-)

To be clear:

* If a new micro:bit is detected and Mu's not in micro:bit mode, it should ask.
* If a new Adafruit board is detected and Mu's not in Adafruit mode, it should ask.
* If both an Adafruit and micro:bit are plugged in, Mu does nothing and assumes the user knows what they're doing.
* Tests!

ntoll, 2018-05-03:
Fixed by #452

closed 2018-05-03 by ntoll