mu-editor/mu #446: settings.json on osx?

brjdk, 2018-04-19:
hi mu developers

new here, new to osx, new to mu and new to micro:bit - so quite new :-)

i have some questions, but i'm not sure where to ask them - please feel free to move them or guide me.
- where do i find the settings.json on osx? i have searched but not found.
- is it possible to disable the warning popup that pops up every time i press flash?
- is it there any keyboard shortcuts for eg. flash?

thanks
Jorgensen

ZanderBrown, 2018-04-19:
Hello, new people is what we are all about!

Do you know what version of Mu you are using? We've been working on a new version for a couple years now so some things have been implemented just aren't in the codewith.mu download yet..

The folder mu uses for settings.json is inside a hidden directory so your unlikely to find it  by searching

If your version of Mu has the cog icon in the bottom right you can click it to see the logs, near the top you should see the settings path listed

brjdk, 2018-04-19:
Hi Zander

I downloaded my version (can't figure out how to check the version) from codewith.mu, so i guess its an old version?

but no cog icon.

where do i find the lastest version for osx? 
i can't quite figure this page out 
[https://github.com/mu-editor/mu]

jorgensen

brjdk, 2018-04-19:
I've just tried download the latest version (v0.9.13) from here [https://github.com/mu-editor/mu/releases](https://github.com/mu-editor/mu/releases), but i don't see a cog in the buttom right corner.

Jorgensen

brjdk, 2018-04-19:
Thanks Zander - do you know how i download the latest version for osx?

Jorgensen

ZanderBrown, 2018-04-19:
macOS test packages are available here: http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=osx/ but you may have to try a few to find one that works

Alternatively if you have pip installed you could use `pip3 install mu-editor` in a terminal, there are [tutorials online](http://itsevans.com/install-pip-osx/) for how to install pip

brjdk, 2018-04-19:
hi zander

i went ahead and tried what you suggested - installed python3 and pip (not really knowing what i was doing). 

then i added install mu-editor, and a lot of packages got downloaded and installed, and then it ended like this 

Successfully built pgzero tornado simplegeneric backcall
Installing collected packages: sip, pyqt5, qscintilla, pyflakes, six, cycler, python-dateutil, pytz, numpy, pyparsing, matplotlib, pygame, pgzero, PyQtChart, decorator, ipython-genutils, traitlets, jupyter-core, tornado, pyzmq, jupyter-client, simplegeneric, appnope, parso, jedi, wcwidth, prompt-toolkit, pickleshare, backcall, pygments, ptyprocess, pexpect, ipython, ipykernel, qtconsole, pycodestyle, pyserial, mu-editor
Successfully installed PyQtChart-5.10 appnope-0.1.0 backcall-0.1.0 cycler-0.10.0 decorator-4.3.0 ipykernel-4.8.2 ipython-6.3.1 ipython-genutils-0.2.0 jedi-0.12.0 jupyter-client-5.2.3 jupyter-core-4.4.0 matplotlib-2.1.2 mu-editor-1.0.0b15 numpy-1.14.2 parso-0.2.0 pexpect-4.5.0 pgzero-1.1 pickleshare-0.7.4 prompt-toolkit-1.0.15 ptyprocess-0.5.2 pycodestyle-2.3.1 pyflakes-1.6.0 pygame-1.9.3 pygments-2.2.0 pyparsing-2.2.0 pyqt5-5.10 pyserial-3.4 python-dateutil-2.7.2 pytz-2018.4 pyzmq-17.0.0 qscintilla-2.10.2 qtconsole-4.3.1 simplegeneric-0.8.1 sip-4.19.8 six-1.11.0 tornado-5.0.2 traitlets-4.3.2 wcwidth-0.1.7

but where to i find the mu-editor now?

thanks
jorgensen

brjdk, 2018-04-19:
i have downloaded the lastest version from here:
[https://github.com/mu-editor/mu](https://github.com/mu-editor/mu)

and looked here
[http://mu.readthedocs.io/en/latest/setup.html](http://mu.readthedocs.io/en/latest/setup.html)

and started the terminal, changed to the folder with the files from github and entered python3 run.py but got the error 

Traceback (most recent call last):
  File "run.py", line 2, in <module>
    from mu.app import run
  File "/Users/brj/Desktop/tmp/mu-master/mu/__init__.py", line 9, in <module>
    current_locale, encoding = locale.getdefaultlocale()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/locale.py", line 562, in getdefaultlocale
    return _parse_localename(localename)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/locale.py", line 490, in _parse_localename
    raise ValueError('unknown locale: %s' % localename)
ValueError: unknown locale: UTF-8

Jorgensen

ZanderBrown, 2018-04-19:
When mu is installed though pip you can type `mu-editor` to run it

As for the traceback you seem to have found a bug

brjdk, 2018-04-20:
hi zander

sorry for being such a noob - but where do i write "mu-editor", and how?

if i write python3 mu-editor in the folder where the downloaded files are i get this:

Bos-MacBook-Pro:mu-master brj$ python3 mu-editor
/Library/Frameworks/Python.framework/Versions/3.6/Resources/Python.app/Contents/MacOS/Python: can't open file 'mu-editor': [Errno 2] No such file or directory
Bos-MacBook-Pro:mu-master brj$ python
Python 2.7.10 (default, Oct  6 2017, 22:29:07) 
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.31)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> 
>>> quit()
Bos-MacBook-Pro:mu-master brj$ python3
Python 3.6.5 (v3.6.5:f59c0932b4, Mar 28 2018, 03:03:55) 
[GCC 4.2.1 (Apple Inc. build 5666) (dot 3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> mu-editor
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'mu' is not defined

Jorgensen

carlosperate, 2018-04-20:
Hi @brjdk,

First of all good call installing Mu via pip, as the app bundles in http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=osx/  are currently broken (we store test files in there, so it's more for internal development and testing).

About your second error (the one in your last message), the command you would need to run is `python3 -m mu` or `python3 run.py` to execute Mu from the source code you downloaded. When the `mu-editor` command is run, that executes the version of Mu installed when you did `pip install mu-editor`.

About your first error (the one you got when running `python3 run.py`), it looks like you've found an interesting bug, thanks for reporting it! It will help us fix it and make Mu better.

@ntoll @tjguk @ZanderBrown I've been meaning to look into this one for little while, since it depends on environmental variables, and to be honest, I can't say I fully understand the description on the docs. So I'm not all that confidence that it will always return predictable results in different OSes (of course I could be wrong, that was just the completely unsubstantiated "feeling" I got from the docs):
https://github.com/mu-editor/mu/blob/bc4fc942921089fccbc5464399b3a87a932cafda/mu/__init__.py#L9
https://docs.python.org/3.5/library/locale.html#locale.getdefaultlocale
The reason this was on my radar, is that when using PyCharm it returns `None`, so in my dev copy I've been wrapping it in a "try except" and giving it a default value.

@brjdk Could you give us a bit more info about your OS? Like version and language?
Could you also try executing this line of code in the terminal?
```
python3 -c "import locale;print(locale.getdefaultlocale())"
```

When I run it I get:
```
$ python3 -c "import locale;print(locale.getdefaultlocale())"
('en_GB', 'UTF-8')
```

carlosperate, 2018-04-20:
@ukbaz has found this related ticket in the Python issue tracker: https://bugs.python.org/issue18378

tjguk, 2018-04-20:
Heh :) I was just looking at that! 

ISTM that we're only using the locale info to determine the correct translation. So a fairly simple option is to try/except and drop through to our default language, which we have to do anyway for unknown languages.

ntoll, 2018-04-20:
Hello folks,

This is interesting.

@brjdk once you've pip installed Mu, you should just be able to type `mu-editor` in the terminal and "it should just work" (tm). :-)

@carlosperate / @tjguk that's an interesting bug which simply doesn't appear on my OS (Linux). I concur with @tjguk as to the probable best way forward to fix it: `try` it and `except` by setting default language.

I'm up to my eyes in PyWeek today and have a v.busy weekend ahead so will be off radar until Monday (when I'll re-focus on Mu for a week).

brjdk, 2018-04-20:
Hi ntoll

i'm really not a programmer, just an architect that have done some PHP / javascript coding, and wanted to play around with an micro:bit - and use mu for that. so i'm actually just looking for newest version of mu for osx. 

should i be able to an version of mu that i can run just as any app on osx?

as i wrote, i don't really understand what i'm doing - just following instructions and guessing abbreviations :-)

could be cool to compile my own version of mu editor - though :-)

thanks
jorgensen

i hope this says more to you than to me

Bos-MacBook-Pro:~ brj$ pip3 install mu-editor
Requirement already satisfied: mu-editor in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (1.0.0b15)
Requirement already satisfied: qtconsole==4.3.1 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from mu-editor) (4.3.1)
Requirement already satisfied: pyflakes==1.6.0 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from mu-editor) (1.6.0)
Requirement already satisfied: PyQtChart==5.10 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from mu-editor) (5.10)
Requirement already satisfied: matplotlib==2.1.2 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from mu-editor) (2.1.2)
Requirement already satisfied: pycodestyle==2.3.1 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from mu-editor) (2.3.1)
Requirement already satisfied: qscintilla==2.10.2 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from mu-editor) (2.10.2)
Requirement already satisfied: pgzero==1.1 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from mu-editor) (1.1)
Requirement already satisfied: pyserial==3.4 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from mu-editor) (3.4)
Requirement already satisfied: pyqt5==5.10 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from mu-editor) (5.10)
Requirement already satisfied: ipykernel>=4.1 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from qtconsole==4.3.1->mu-editor) (4.8.2)
Requirement already satisfied: pygments in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from qtconsole==4.3.1->mu-editor) (2.2.0)
Requirement already satisfied: traitlets in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from qtconsole==4.3.1->mu-editor) (4.3.2)
Requirement already satisfied: ipython-genutils in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from qtconsole==4.3.1->mu-editor) (0.2.0)
Requirement already satisfied: jupyter-client>=4.1 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from qtconsole==4.3.1->mu-editor) (5.2.3)
Requirement already satisfied: jupyter-core in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from qtconsole==4.3.1->mu-editor) (4.4.0)
Requirement already satisfied: pytz in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from matplotlib==2.1.2->mu-editor) (2018.4)
Requirement already satisfied: numpy>=1.7.1 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from matplotlib==2.1.2->mu-editor) (1.14.2)
Requirement already satisfied: six>=1.10 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from matplotlib==2.1.2->mu-editor) (1.11.0)
Requirement already satisfied: python-dateutil>=2.1 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from matplotlib==2.1.2->mu-editor) (2.7.2)
Requirement already satisfied: cycler>=0.10 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from matplotlib==2.1.2->mu-editor) (0.10.0)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from matplotlib==2.1.2->mu-editor) (2.2.0)
Requirement already satisfied: pygame>=1.9 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from pgzero==1.1->mu-editor) (1.9.3)
Requirement already satisfied: sip<4.20,>=4.19.4 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from pyqt5==5.10->mu-editor) (4.19.8)
Requirement already satisfied: tornado>=4.0 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from ipykernel>=4.1->qtconsole==4.3.1->mu-editor) (5.0.2)
Requirement already satisfied: ipython>=4.0.0 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from ipykernel>=4.1->qtconsole==4.3.1->mu-editor) (6.3.1)
Requirement already satisfied: decorator in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from traitlets->qtconsole==4.3.1->mu-editor) (4.3.0)
Requirement already satisfied: pyzmq>=13 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from jupyter-client>=4.1->qtconsole==4.3.1->mu-editor) (17.0.0)
Requirement already satisfied: prompt-toolkit<2.0.0,>=1.0.15 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from ipython>=4.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor) (1.0.15)
[... cut ...]
Requirement already satisfied: parso>=0.2.0 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from jedi>=0.10->ipython>=4.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor) (0.2.0)
Requirement already satisfied: ptyprocess>=0.5 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from pexpect; sys_platform != "win32"->ipython>=4.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor) (0.5.2)
Bos-MacBook-Pro:~ brj$ 
Bos-MacBook-Pro:~ brj$ mu-editor
-bash: mu-editor: command not found

brjdk, 2018-04-20:
sure 

Bos-MacBook-Pro:~ brj$ echo $PATH
/Library/Frameworks/Python.framework/Versions/3.6/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin
Bos-MacBook-Pro:~ brj$ pip3 --version
pip 10.0.1 from /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pip (python 3.6)
Bos-MacBook-Pro:~ brj$

carlosperate, 2018-04-20:
Odd, your python bin directory is in the path, and without installation errors the mu-editor should have been there.

Just to double chec, could you do execute the following as well?
```
ls /Library/Frameworks/Python.framework/Versions/3.6/bin
```
Normally there should be a `mu-editor` file in there, but I guess it's missing.

brjdk, 2018-04-20:
hi carlos

here is what i get
Bos-MacBook-Pro:~ brj$ ls /Library/Frameworks/Python.framework/Versions/3.6/bin
2to3			idle3.6			jupyter-kernel		miniterm.py		pip3			pygmentize		python3.6		pyvenv
2to3-3.6		iptest			jupyter-kernelspec	miniterm.pyc		pip3.6			pylupdate5		python3.6-32		pyvenv-3.6
__pycache__		iptest3			jupyter-migrate		mu			pycodestyle		pyrcc5			python3.6-config	wheel
easy_install-3.6	ipython			jupyter-qtconsole	mu-debug		pydoc3			python3			python3.6m
f2py			ipython3		jupyter-run		pgzrun			pydoc3.6		python3-32		python3.6m-config
idle3			jupyter			jupyter-troubleshoot	pip			pyflakes		python3-config		pyuic5
Bos-MacBook-Pro:~ brj$

carlosperate, 2018-04-20:
How funny, you've got `mu-debug` available but no `mu-editor`.

Maybe try reinstalling:
```
pip3 uninstall mu-editor
pip3 install mu-editor
```

Or this one might be quicker instead:
```
pip3 install --upgrade --no-deps --force-reinstall mu-editor
```

brjdk, 2018-04-20:
it says that it uses a cached version - is that a problem?
Bos-MacBook-Pro:~ brj$ pip3 install --upgrade --no-deps --force-reinstall mu-editor
Collecting mu-editor
  Using cached https://files.pythonhosted.org/packages/a6/8b/991ba5365f19538a9164ce45e7fa844714d76bd00b93cd24216c7f57c9fc/mu_editor-1.0.0b15-py3-none-any.whl
Installing collected packages: mu-editor
  Found existing installation: mu-editor 1.0.0b15
    Uninstalling mu-editor-1.0.0b15:
      Successfully uninstalled mu-editor-1.0.0b15
Successfully installed mu-editor-1.0.0b15
Bos-MacBook-Pro:~ brj$ mu-editor
-bash: mu-editor: command not found
Bos-MacBook-Pro:~ brj$

carlosperate, 2018-04-20:
Ups, completely missed the fact that the command is `mu`, not `mu-editor`, and that file was already present in your `/Library/Frameworks/Python.framework/Versions/3.6/bin` before, so it should be there again after the force reinstalled.

Could you try opening Mu by just using the `mu` command in the termina?

brjdk, 2018-04-20:
hi carlos

just tried but got this message

Bos-MacBook-Pro:~ brj$ mu
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/bin/mu", line 7, in <module>
    from mu.app import run
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/mu/app.py", line 31, in <module>
    current_locale, encoding = locale.getdefaultlocale()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/locale.py", line 562, in getdefaultlocale
    return _parse_localename(localename)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/locale.py", line 490, in _parse_localename
    raise ValueError('unknown locale: %s' % localename)
ValueError: unknown locale: UTF-8
Bos-MacBook-Pro:~ brj$

carlosperate, 2018-04-20:
Yes, that's the same error you were getting when running from source with the command `python3 run.py`, so it means that the `mu` command is installed and working "as expected", but that there is a bug in Mu we need to fix.

Btw if you try to run the following line I assume you get the same error printed on the terminal?
```
python3 -c "import locale;print(locale.getdefaultlocale())"
```

brjdk, 2018-04-20:
i get this error
Bos-MacBook-Pro:~ brj$ python3 -c "import locale;print(locale.getdefaultlocale())"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/locale.py", line 562, in getdefaultlocale
    return _parse_localename(localename)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/locale.py", line 490, in _parse_localename
    raise ValueError('unknown locale: %s' % localename)
ValueError: unknown locale: UTF-8
Bos-MacBook-Pro:~ brj$

brjdk, 2018-04-22:
should i wait for an other build or what should i do?

thanks
jorgensen

brjdk, 2018-04-22:
hi carlos 
i have the deepest respect for the work you people do - especially as volunteers. 
i just didn't know what to do :-)

i will try your workaround.

thanks
jorgensen

brjdk, 2018-04-24:
it worked - i entered
export LC_ALL=en_GB.UTF-8
python3 mu

:-)

ntoll, 2018-04-25:
Hurrah! :-)

brjdk, 2018-04-25:
btw - this version is so cool - really like 
- the plotter
- shortcuts
- and no popup when pressing the flash button
this should be the 'official' version, downloaded from codewith.mu!

so people could enjoy mu even more :-)

brjdk, 2018-04-25:
stupid question, but anyone know if i can add this 
export LC_ALL=en_GB.UTF-8
python3 mu

to somekind of shortcut / script on the desktop or elsewhere so i don't need to start the terminal and go to
Library / Frameworks / Python.framework / Versions / 3.6 / bin / to start Mu? 

Thanks :-)
Jorgensen

ukBaz, 2018-04-26:
@brjdk does this answer your question?  
https://www.hastac.org/blogs/joe-cutajar/2015/04/21/how-make-simple-bash-script-mac

brjdk, 2018-04-26:
@ukBaz thanks - that did it :-)

closed 2018-05-14 by ntoll