mu-editor/mu #448: SHIFT-Tab has 2 behaviours

WayneKeenan, 2018-04-21:
I'm only raising this because it surprised me and I'm not sure if it a consious design decision, I could see why it might be, but just thought I'd say:

1. If the cursor is at the start of a line in the whitespace then SHIFT-TAB will un-indent the current line
2. If the cursor is in the text then SHIFT-TAB moves the cursor back 4 spaces.

My preference would be for it to only have one behaviour and that behaviour to be option 1, similar to many other text, IDE and word processors I've used. 

Says me, myself, and I ;)

ntoll, 2018-04-21:
@WayneKeenan heh... so this is the default behaviour of the underlying Scintilla widget. The API is extraordinarily comprehensive but should also come with a "here be dragons" notice. I'm not even sure where I'd start if I wanted to modify this sort of behaviour and, alas, it looks like quite a rabbit hole to fall into in terms of time and effort to use in order to modify behaviour those familiar with Scintilla would be expecting.

WayneKeenan, 2018-04-21:
fair do's

closed 2018-04-21 by WayneKeenan