mu-editor/mu #456: Could pygame-zero mode bundle sample assets?

bennuttall, 2018-04-29:
Just an idea: would it be possible to bundle a handful of sample game assets in the `images` and `sounds` directories which are created by Mu? I'm not talking about an extensive suite of artwork, just a few sample assets to help someone get started, like the alien ones in the [tutorial](http://pygame-zero.readthedocs.io/en/stable/introduction.html#draw-a-sprite):

[image] [image]

This would mean a user could follow the tutorial without having to work out how to download these images and save them to the right location. A few more would be nice, but I don't know where these ones came from.

ping @lordmauve

ptorrone, 2018-04-29:
if ya'll want, we can "donate" blinka the circuitpython snake art and/or circuitplayground character art for this (adabot, minerva, etc).

[image]

[image]

ntoll, 2018-04-29:
@bennuttall that already happens with the version of Mu in master (next beta.16). @ptorrone AWESOME... now we just need a PGZero game with those assets. :-P @lordmauve..?

bennuttall, 2018-04-29:
Ha! I just realised Mu already did that, came back to say so, and you beat me to it. Great idea by the way! :+1:

lordmauve, 2018-04-29:
The graphics in the example come from [kenney.nl](http://kenney.nl), which is [noted in the documentation since 1.2](http://pygame-zero.readthedocs.io/en/stable/introduction.html#draw-a-sprite).

ntoll, 2018-05-03:
OK... I'll close this issue since I've just used my Tardis to go back to fix this a few weeks ago so @bennuttall gets to see that it already works that way...! ;-)

closed 2018-05-03 by ntoll